m-labs/artiq #1061: artiq build timeouts

jordens, 2018-06-07:
`conda build ... --output` regularly takes 5 and sometimes more than 20 minutes.
http://buildbot.m-labs.hk/builders/artiq-board/builds/1627/steps/conda_build_output/logs/stdio

sbourdeauducq, 2018-06-26:
I've disabled the overclocking in the BIOS on ``build.lab.m-labs.hk`` so it should be stable now. We should move the builds to that faster machine.

jordens, 2018-06-26:
I suspect this particular issue is from a large number of packages (in our conda channels, previous builds on disk, or in conda-forge). Might not even be CPU bound.

sbourdeauducq, 2018-06-26:
Yes, we should look at that too. But the new machine also has SSD.

sbourdeauducq, 2018-07-17:
That was due to the 26GB of trash in ``/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld``. I deleted it and it now "conda build output" takes a more reasonable time. We should clean up that folder regularly.

closed 2018-07-17 by sbourdeauducq