m-labs/artiq #1063: tools should not emit a traceback when exiting cleanly on KeyboardInterrupt

jordens, 2018-06-08:
The current behavior is unnecessarily verbose and irritating.

```
artiq_ctlmgr
^CTraceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_ctlmgr", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_ctlmgr')()
  File "/home/rj/src/artiq/artiq/frontend/artiq_ctlmgr.py", line 84, in main
    loop.run_until_complete(rpc_server.wait_terminate())
  File "/home/rj/src/conda/envs/py35/lib/python3.5/asyncio/base_events.py", line 454, in run_until_complete
    self.run_forever()
  File "/home/rj/src/conda/envs/py35/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/home/rj/src/conda/envs/py35/lib/python3.5/asyncio/base_events.py", line 1388,in _run_once
    event_list = self._selector.select(timeout)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

```
artiq_master
^CTraceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_master", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_master')()
  File "/home/rj/src/artiq/artiq/frontend/artiq_master.py", line 153, in main
    loop.run_forever()
  File "/home/rj/src/conda/envs/py35/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/home/rj/src/conda/envs/py35/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

sbourdeauducq, 2018-06-08:
How to exit cleanly on KeyboardInterrupt? Python asynchronous exceptions are broken by design (there is a race condition).
Should we catch Ctrl-C in the main loop instead?

jordens, 2018-06-08:
Maybe we don't always exit cleanly on `KeyboardInterrupt`. But even in that case I feel that the harm done by not printing the traceback would be smaller than the annoyance caused by printing it (all the time).
What's the main loop? If you mean adding a signal handler I don't think that's better than suppressing the traceback.

sbourdeauducq, 2018-06-09:
Yes, regular signal handlers have the same unsolvable race condition bug.
On Linux this can be solved with signalfd (and then watch the fd in the asyncio loop), but I don't know how to do it on Windows.

jordens, 2018-11-27:
Since at it's core it is about exiting cleanly when requested, we can circumvent issues with signals and windows/unix differences.
This approach would do what I want. Opinions before I expand this to the other tools?
 
```patch
diff --git a/artiq/frontend/artiq_master.py b/artiq/frontend/artiq_master.py
index 4583ed6d..0e21d022 100755
--- a/artiq/frontend/artiq_master.py
+++ b/artiq/frontend/artiq_master.py
@@ -5,6 +5,7 @@ import argparse
 import atexit
 import os
 import logging
+import threading

 from artiq.tools import (simple_network_args, atexit_register_coroutine,
                          bind_address_from_args)
@@ -150,7 +151,13 @@ def main():
     atexit_register_coroutine(server_logging.stop)

     print("ARTIQ master is now ready.")
+
+    def stop_loop():
+        input('Press <enter> to stop.')
+        loop.call_soon_threadsafe(loop.stop)
+    threading.Thread(target=stop_loop).start()
     loop.run_forever()

+
 if __name__ == "__main__":
     main()
```

sbourdeauducq, 2018-11-27:
What about exiting from the client/dashboard instead? Then it's just using the existing sockets.
Threads in python are not so well implemented and it is best to avoid them when possible.

jordens, 2018-11-27:
That's a different feature. This one applies to other tools than master as well. And as long as you can't start the master from the dashboard I'm not that interested in being able to shut it down. If a tool is started from the command line it should be possible to end it cleanly right there as well.
What's wrong with that?

sbourdeauducq, 2018-11-27:
Threads, and also it's easy to inadvertently press Enter.

Have you tried this on Windows:
https://docs.python.org/3/library/signal.html#signal.set_wakeup_fd
https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.add_signal_handler (doc says it supports only Unix, despite Victor's comment below)
https://bugs.python.org/issue12304#msg238411

sbourdeauducq, 2022-05-25:
```
Python 3.9.11 (main, Mar 18 2022, 16:54:01)  [GCC 11.2.0 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
...
>>> loop.add_signal_handler(signal.SIGINT, lambda: print("hello"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "Z:\nix\store\jksyp1bxc3vabfwwlvdpg49l3ad4ld6c-msys2-env\mingw64\lib\python3.9\asyncio\events.py", line 539, in add_signal_handler
    raise NotImplementedError
NotImplementedError
```

sbourdeauducq, 2022-05-27:
This works on Linux and Windows:
```python
import signal
import socket
import asyncio

rsock, wsock = socket.socketpair()
loop = asyncio.get_event_loop()

wsock.setblocking(0)
signal.set_wakeup_fd(wsock.fileno())
signal.signal(signal.SIGINT, lambda sig, frame: None)

async def terminator():
    data = await loop.sock_recv(rsock, 1)
    print(data)
    loop.stop()

loop.create_task(terminator())

loop.run_forever()
```

sbourdeauducq, 2022-05-27:
done

closed 2022-05-27 by sbourdeauducq

sbourdeauducq, 2022-05-27:
[image]