m-labs/artiq #1064: Sayma: UART silence

hartytp, 2018-06-10:
- Some builds give no output at all on the UART
- When I've seen this, I've found that repeatedly building/flashing the same version of the code reproducibly gives no output on the UART. I can switch back and forth between working and non-working versions of the code reproducibly and always find that some versions work while others don't
- The code changes that trigger this issue have no connection to booting (e.g changes to serwb)

Needs investigation:
- next time someone sees this, can they post the build here?
- haven't checked if this is reproducible across boards, or whether the same binaries work on some boards but not others
- haven't checked if this is reproducible with the RTM disconnected
- reproducible with/without SAWG?

jbqubit, 2018-06-10:
I haven't seen this yet. I'll post if I do.

hartytp, 2018-06-11:
Reproduced with https://github.com/hartytp/artiq/commits/master 10b87ac. Files here [when upload finishes].

Conda build environment 
[broken_conda.txt](https://github.com/m-labs/artiq/files/2091778/broken_conda.txt)

It would be really helpful if others with access to Sayma could try this out on their AMC/RTM.

hartytp, 2018-06-11:
Resetting artiq to current master (0b86225) and rebuilding gateware and firmware fixed this. Rebuilding just the firmware (with `--no-comiple-gateware`) did not seem to (although I will double check those observations tomorrow.

Since none of the changes I made touch the gateware, I'm at a loss to explain this.

hartytp, 2018-06-12:
Okay, this was some conda issue...

Is the buildbot working for Sayma now?

sbourdeauducq, 2018-11-28:
Check I/O timing numbers (reported by Vivado) at the flash pins.

marmeladapk, 2018-11-28:
AFAIK in case of @bradbqc it's the same bitstream that worked before and didn't work after flash.

In my case AFAIR when I had this bug and after clearing flash the same bitstream worked. I had this problem both on Kasli and on Sayma, @bradbqc has this problem on Sayma.

bradbqc, 2018-11-29:
@marmeladapk Yes, except the flash failed, citing "Error: Cannot enable write to flash. Status=0x00000000." I'll try erasing and re-flashing and report back

bradbqc, 2018-11-29:
I've discovered that part (if not all) of the reason I'm unable to flash is that the mmc is restarting every minute (pretty much exactly 60s, as far as I can tell) for some reason. I noticed a very brief dip in the current sayma was drawing and so I looked at the mmc serial output. It is repeatedly printing 
`AMC Hotswap handle pressed!
RTM Hotswap handle pressed!`
for 60 seconds and then restarting (output on restart: https://paste.ofcode.org/YyckiatAgDf36Gjw9rJMAY). Is this something that could possibly be solved by successfully flashing artiq or is it an issue with the mmc firmware? If it's the former, would it be possible for me to split the flashing into smaller steps that take < 1 minute? That worked for me when dumping and erasing flash. Also, I've attached the flash contents prior to erasing (had to zip them because github only allows certain file types to be attached).

[flash_dump_spi.zip](https://github.com/m-labs/artiq/files/2630254/flash_dump_spi.zip)

gkasprow, 2018-11-29:
@bradbqc do you use one or two power units at the time ?
Can you please make short circuit to the transistors descried [here.](https://github.com/sinara-hw/sinara/issues/571#issuecomment-441328494) 
IT's quite possible that the problem is related with management power overload.

sbourdeauducq, 2018-11-29:
@bradbqc @gkasprow MMC and µTCA bugs are a different issue.

bradbqc, 2018-11-30:
@gkasprow We only have one, unfortunately. I've tried this with and without the shorts and it exhibits the same behavior for both setups. @sbourdeauducq In addition to trying with/without the shorts, I've also tried running it on the benchtop with an external power supply (that's where I've been doing most of my work) - the only difference is that when I'm in the crate, it only prints "AMC Hotswap handle pressed," whereas when I'm on the benchtop, it prints that for both AMC and RTM.

I do have a theory as to the cause of the restarting, although it isn't perfect. In the mmc firmware there's a watchdog timer (mmc-firmware/modules/watchdog.c) that is set for 60 seconds, and if it doesn't get reset before time runs out it restarts everything. I'm thinking that the incessant interrupts from the hotswap handle are preventing the timer from being reset. As for the cause of the hotswap interrupts, I see in mmc-firmware/modules/sensors/hotswap.c (line 169) that the message I'm seeing only appears when new_state_amc != old_state_amc, so for some reason old_state_amc is not getting set to the new value the first time (or any subsequent time) this method runs. It looks like this variable should be getting set in line 150, inside an if-block that depends on the return value of hotswap_send_event. I'm guessing the condition in that if statement must be returning false for some reason. I tried tracing hotswap_send_event back to the core code, but after 3 or 4 layers of function calls I decided to give up.

All that being said, this is the first time I've looked at the mmc firmware source, so there's a good chance I'm misunderstanding/overlooking quite a few things (such as the intricacies of the mmc's scheduler, which may come into play since the watchdog timer and hotswap interrupts have different priorities).

Unfortunately, I don't have time to look into it further as I've been instructed to put sayma on the back burner and focus on other things.

bradbqc, 2018-11-30:
Yeah, that's the one question my theory (or anything purely software related) doesn't account for, especially since I think I'm using the same exact mmc firmware as @marmeladapk. And it has never worked in the crate, but we only have 1 power supply so I don't know if it works with 2.

gkasprow, 2018-11-30:
It's quite possible that board got damaged. It happens that one of the RC component goes off and gives some strange symptoms ...

bradbqc, 2018-11-30:
That very well could be the case. Thankfully I don't have to worry about that (for now, anyway) since I won't be working with sayma for the foreseeable future.

gkasprow, 2018-12-01:
I think the only solution is to send it to Warsaw.

sbourdeauducq, 2019-04-01:
Has anyone seen this with recent Vivado (2018.3)? I haven't.

sbourdeauducq, 2019-12-09:
Seems fixed AFAICT.

closed 2019-12-09 by sbourdeauducq