m-labs/artiq #1066: Sayma: 0x40023e78

jbqubit, 2018-06-10:
```
[     0.003882s]  INFO(runtime): software version 4.0.dev+1133.g0b086225
[     0.010234s]  INFO(runtime): gateware version 4.0.dev+1133.g0b086225
```
While running script using sawg....

```
[   168.348834s]  INFO(runtime::session): no connection, starting idle kernel
[   168.354543s]  INFO(runtime::session): no idle kernel found
[   252.177085s]  INFO(runtime::session): new connection from 192.168.1.68:47124
[   252.225018s]  INFO(runtime::kern_hwreq): resetting RTIO
@ 0x40023e78

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1133.g0b086225
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011101001001001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111001001001001001001001001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101111001110010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001110100010010010010010000000000010010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000110010100100100101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111110110100100100100100100100100010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001010010
Module 0:
000000000000000000000000000000000000000000000000000000000000000000001111111111111111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110100100100100100100100010010010010010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010010010100010
DQS initial delay: 101 taps
Write leveling: 65 72 91 85 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011101100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000110110110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000001101111111011011011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000011011111011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 119+-39 110+-47 95+-39 80+-43 done
SDRAM initialized

```

hartytp, 2018-06-12:
Joe, looking at this again, the eye scans during your mem test look really quite bad. 

For example `11101001001001001001001000100100100100100` that's much worse than anything I've seen for a while. Compare that to the eye scans I posted in #1065 for example. 

Can you post "typical" eye scans you see on your boards with + without the RTM, please?

Could also be connected to #1065.

jbqubit, 2018-06-12:
Following is typical.

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1133.g0b086225
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000100000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000110000000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 94 102 128 121 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100111111011100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000100111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 179+-76 168+-85 150+-79 136+-83 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003882s]  INFO(runtime): software version 4.0.dev+1133.g0b086225
[     0.010232s]  INFO(runtime): gateware version 4.0.dev+1133.g0b086225
[     0.016612s]  INFO(runtime): log level set to INFO by default
```

hartytp, 2018-06-12:
Is that with RTM? Any idea what caused the bad eyes then? Did you change gateware between good and bad mem tests?

jbqubit, 2018-06-12:
Otherwise this board has uniformly had nice memtest results for the last two weeks.

sbourdeauducq, 2018-06-16:
#1065

closed 2018-06-16 by sbourdeauducq