m-labs/artiq #1067: Sayma SAWG setting frequency1 causes lock-up

jbqubit, 2018-06-10:
```
[     0.003882s]  INFO(runtime): software version 4.0.dev+1133.g0b086225
[     0.010234s]  INFO(runtime): gateware version 4.0.dev+1133.g0b086225
```

If only setting frequency0 the following can be run many times in succession using artiq_run. 

If frequency1.set is uncommented the liklihood of the core device locking up on the first call of artiq_run is 5/6. In the 1/6 case the core device locked up after calling artiq_run a second time. So this looks reproducible on my end. 

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        freq0 = 210*MHz
        amp1 = 0.5
        for sawg in self.sawgs:
            sawg.reset()
            delay(500*us)
            sawg.frequency0.set(freq0)
            sawg.amplitude1.set(amp1)
            #sawg.frequency1.set(30*MHz)
```

jbqubit, 2018-06-10:
There's nothing printed on the UART. Setting frequency2 also seems to trigger same behavior. 

```
$ artiq_run repository/sines.py
Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/artiq-dev/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/britton/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/britton/artiq-dev/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/core.py", line 126, in run
    self.comm.check_system_info()
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 214, in check_system_info
    self._write_empty(Request.SystemInfo)
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 179, in _write_empty
    self._write_header(ty)
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 171, in _write_header
    self.open()
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 84, in open
    self.socket = initialize_connection(self.host, self.port, **kwargs)
  File "/home/britton/artiq-dev/artiq/artiq/coredevice/comm.py", line 27, in initialize_connection
    sock = socket.create_connection((host, port), 5.0)
  File "/home/britton/miniconda3/envs/artiq-dev/lib/python3.5/socket.py", line 712, in create_connection
    raise err
  File "/home/britton/miniconda3/envs/artiq-dev/lib/python3.5/socket.py", line 703, in create_connection
    sock.connect(sa)
socket.timeout: timed out
```

sbourdeauducq, 2018-06-11:
Can you try by using the startup kernel? Likely there is intermittent memory/gateware corruption (did you rework the HMC7043 reset?) and running the whole TCP/IP stack is more likely to crash than running simpler code. Also disconnect Ethernet during the test so that the CPU doesn't have to process network broadcasts.

sbourdeauducq, 2018-06-11:
It could be that it's longer kernels loaded through Ethernet that cause crashes, not frequency1 itself.

jbqubit, 2018-06-12:
- 4.0.dev+1133.g0b086225
- Sayma_AMC s/n -6 (the one I've been using for two weeks now) 
   - memtests and locks fail in fewer than 1/55 previous boots
   - Ethernet works fine on this board interactively when just working with frequency 0
- HMC7043 reset pullup applied 
- running as startup kernel 

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        # self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        for sawg in self.sawgs:
            sawg.reset()
            delay(500*us)
            sawg.amplitude1.set(100*MHz)
            sawg.frequency1.set(100*MHz)
```

```
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1133.g0b086225
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011001000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100011000000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 92 102 123 118 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101001110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 182+-77 173+-85 152+-78 139+-83 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003882s]  INFO(runtime): software version 4.0.dev+1133.g0b086225
[     0.010232s]  INFO(runtime): gateware version 4.0.dev+1133.g0b086225
[     0.016612s]  INFO(runtime): log level set to INFO by default
[... 78 lines cut ...]
[     9.922865s]  INFO(runtime::kern_hwreq): resetting RTIO
[... cut ...]
- no output on UART, amplitude modulation 100 MHz tone
- no output on UART, amplitude modulation 100 MHz tone

Reconnect SATA0 (Ethernet cable). Change frequency1 to frequency0 in above experiment. 
- no output on UART, amplitude modulation 100 MHz tone, no response to ping
- no output on UART, amplitude modulation 100 MHz tone, no response to ping
- no output on UART, amplitude modulation 100 MHz tone, no response to ping

Reduce @kernel in startup kernel to nothing but ```self.core.reset()```.
- can pin board
- can pin board
- can pin board

jbqubit, 2018-06-12:
Edited previous post to strike ~~Ethernet works fine on this board interactively when just working with frequency 0~~

jbqubit, 2018-06-12:
On this same board I can repeatedly run experiments that don't involve SAWG interactively over Ethernet. 

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.t_list = self.linspace(1*us, 133*us, 100)

    def linspace(self, start, stop, num):
        dx = (stop-start)/(num-1) 
        return [start+n*dx for n in range(num)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        self.ttl_sma_out.output()

        while True:
            self.k3()            

    @kernel
    def k3(self):
        for t in self.t_list:
            delay(20*us)
            self.ttl_sma_out.pulse(t)
```

sbourdeauducq, 2018-06-16:
#1065

closed 2018-06-16 by sbourdeauducq