m-labs/artiq #1071: Sayma SAWG: startup kernel, frequency0 in loop, amplitude modulation

jbqubit, 2018-06-12:
- 4.0.dev+1133.g0b086225
- Sayma_AMC s/n -6 (the one I've been using for two weeks now) 
   - memtests and locks fail in fewer than 1/50 (I'm keeping scorecard)   
- HMC7043 reset pullup applied 
- running as startup kernel 
- kernel continued to run for 5 minutes before I reset board, but can't ping

https://www.youtube.com/watch?v=n625gI3Vlqg

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        for sawg in self.sawgs:
            sawg.reset()
            delay(500*us)
            sawg.amplitude1.set(100*MHz)
        while True:
            self.k1()            

    @kernel
    def k1(self):
        for sawg in self.sawgs:
            sawg.frequency0.set(100*MHz)
            delay(1*ms)
```

```
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1133.g0b086225
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110000100000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 97 105 128 120 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 181+-77 173+-85 152+-78 137+-82 done
SDRAM initialized
[... 63 lines cut ...]
[     9.072222s]  INFO(board_artiq::ad9154):   c2 errors: 0
[     9.077431s]  INFO(board_artiq::ad9154):   c3 errors: 0
[     9.082353s]  INFO(board_artiq::ad9154):   ...passed
[     9.087302s]  INFO(board_artiq::ad9154): AD9154-1 SYSREF scan:
[     9.347073s]  INFO(board_artiq::ad9154):   phase: 24, sync error: 449
[     9.677269s]  INFO(board_artiq::ad9154):   phase: 56, sync error: 450
[     9.682406s]  INFO(board_artiq::ad9154):   phase min: Some(24), phase max: Some(55)
[     9.690047s]  INFO(board_artiq::ad9154): AD9154-1 setting SYSREF phase to 88
[     9.707434s]  INFO(board_artiq::ad9154): AD9154-1 initializing...
[     9.720326s]  INFO(board_artiq::ad9154):   ...done
[     9.794192s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[     9.801437s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[     9.808675s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[     9.815912s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[     9.823149s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[     9.830386s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[     9.837624s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[     9.844861s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[     9.852160s]  INFO(runtime): using MAC address 00-0a-35-03-1e-75
[     9.856956s]  INFO(runtime): using IP address 192.168.1.75
[     9.864001s]  INFO(runtime::mgmt): management interface active
[     9.877306s]  INFO(runtime::session): accepting network sessions
[     9.891565s]  INFO(runtime::session): running startup kernel
[     9.922972s]  INFO(runtime::kern_hwreq): resetting RTIO
```

jbqubit, 2018-06-12:
I've seen this same behavior loading four times.

trxw, 2018-06-12:
For one thing, the amplitude needs to be set between 0-1 and not 100MHz

`for sawg in self.sawgs:
            sawg.reset()
            delay(500*us)
            sawg.amplitude1.set(100*MHz)`

jbqubit, 2018-06-12:
Ah right. With ```sawg.amplitude1.set(0.5)``` this works as startup kernel as in the past. Glad this is a typo and not a regression. :)

closed 2018-06-12 by jbqubit

trxw, 2018-06-12:
@jbqubit  Shouldn't there be a bound checking?

jbqubit, 2018-06-13:
@trxw https://github.com/m-labs/artiq/issues/1045