m-labs/artiq #1076: RPC returning TList(TBytes) corrupts unrelated variables

jonaskeller, 2018-06-20:
Using `3.6py_40+gite8092f6f` on Windows 7 64bit with a KC705, `nist_qc2`.

Calling an RPC which returns a list of `bytes` objects seems to corrupt other variables that should be completely independent:

    from artiq.experiment import *

    class RPCListCorruption(EnvExperiment):
        """RPC list corruption example"""
        
        def build(self):
            self.setattr_device("core")
            self.num_bytes = 9
            
        @kernel
        def run(self):
            self.core.reset()
            test_list = [i for i in range(9)]
            print(test_list)  # everything as it should be here
            # print(test_list)  # uncommenting more `print`s here reduces
                                # the number of corrupted entries below
            # print(test_list)  # with 3 `print`s, the whole list survives
                                # for self.num_bytes = 9
            self.rpc()
            print(test_list)  # corrupted list
            
            
        def rpc(self) -> TList(TBytes):
            return [bytes([0]) for i in range(self.num_bytes)]

This prints

	print:[0, 1, 2, 3, 4, 5, 6, 7, 8]
	print:[1, 1342176868, 1, 3, 4, 5, 6, 7, 8]

The higher `self.num_bytes`, the more list entries (from the left) are corrupted.
Adding more `print`s before the RPC call reduces the number of corrupted entries, e.g. with `self.num_bytes` lines of `print(test_list)`, `test_list` survives unharmed. The values of the bytes in the list returned by the RPC don't seem to make a difference.

This might be a separate issue, but just in case:
In addition, setting `self.num_bytes` to more than `16` breaks the experiment in different ways:
for `17` and `19`, it raises an exception:

	root:Terminating with exception (ConnectionResetError: Connection closed)
	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\master\worker_impl.py", line 279, in main
		exp_inst.run()
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
		return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\coredevice\core.py", line 133, in run
		self.comm.serve(embedding_map, symbolizer, demangler)
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\coredevice\comm_kernel.py", line 599, in serve
		self._read_header()
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\coredevice\comm_kernel.py", line 179, in _read_header
		(sync_byte, ) = struct.unpack("B", self.read(1))
	  File "C:\Anaconda3\envs\artiq3.6\lib\site-packages\artiq\coredevice\comm_kernel.py", line 162, in read
		raise ConnectionResetError("Connection closed")
	ConnectionResetError: Connection closed

with the following UART output:

	[   361.336478s]  INFO(runtime::session): new connection from 132.163.182.162:62549
	[   361.374374s]  INFO(runtime::kern_hwreq): resetting RTIO
	[   361.394937s]  INFO(kernel): panic at /var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/artiq-board_1521703689395/work/artiq/firmware/ksupport/lib.rs:490: exception 6 at PC 0x408403e0, EA 0x1d
	[   361.412675s] ERROR(runtime::session): unexpected request RunAborted from kernel CPU
	[   361.420011s] ERROR(runtime::session): session aborted: protocol error
	[   361.426645s]  INFO(runtime::session): no connection, starting idle kernel
	[   361.433733s]  INFO(runtime::session): no idle kernel found

With `self.num_bytes = 18` and only a single `print` before the RPC, the experiment never returns (shows as `running` in the scheduler indefinitely). Doesn't occur with two `print`s. Didn't test any higher values of `self.num_bytes`.

closed 2018-06-21 by whitequark

jonaskeller, 2018-06-22:
I'd like to test this but can't build the newest `kc705-nist_qc2` 3.6 gateware. The bot is building 4.0.dev despite the argument `-branch=release-3`:
http://buildbot.m-labs.hk/builders/artiq-board/builds/1674

Has the syntax changed?

whitequark, 2018-06-22:
Arguments are passed before builder name, not after.

jonaskeller, 2018-06-22:
Ah, thanks. I've flashed the new gateware and it works now.

commit 095ee28f (2018-06-21): runtime: fix size values for bytes and bytearray RPCs.

Fixes #1076.
commit 3e7cdaa5 (2018-06-21): runtime: fix size values for bytes and bytearray RPCs.

Fixes #1076.