m-labs/artiq #1078: kasli installation

mfe5003, 2018-06-21:
I have a kasli system that I am working through the installation instructions for version 3 [here](https://m-labs.hk/artiq/manual-release-3/in)
I am working with artiq-kasli-master version 4.0.dev (805+gitd0f6123) and artiq 4.0.dev (1191+git8b3c12e6) on Windows 10.
It looks like there have been a couple tool chain changes between 3 and 4 and the manual in the artiq repo has not been updated for 4 yet.
I have managed to figure out the new steps up to writing the compiled idle kernel to the device.

The process fails here, (which was my best attempt to modify the `artiq_coreconfig write -f idle_kernel idle.elf` instruction):
`artiq_coremgmt config write -f idel_kernel idle_kernel.elf`

```
Traceback (most recent call last):
  File "-----\Anaconda2\envs\artiq-dev\Scripts\artiq_coremgmt-script.py", line 9, in <module>
    sys.exit(main())
  File "-----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\frontend\artiq_coremgmt.py", line 164, in main
    mgmt.config_write(key, fi.read())
  File "-----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 168, in config_write
    self._write_header(Request.ConfigWrite)
  File "-----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 81, in _write_header
    self.open()
  File "-----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 65, in open
    self.socket = initialize_connection(self.host, self.port, **kwargs)
  File "-----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm.py", line 27, in initialize_connection
    sock = socket.create_connection((host, port), 5.0)
  File "-----\Anaconda2\envs\artiq-dev\lib\socket.py", line 694, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "-----\Anaconda2\envs\artiq-dev\lib\socket.py", line 733, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno 11001] getaddrinfo failed
```

I am suspecting a couple reasons that could cause this not to work:
- I have not connected the kasli to the network and I am only working over the usb connection (since the error seems to be with a socket)
- I am using the wrong device_db.py file.  I am using the one coming from site-packages/artiq/examples/kasli/device_db_tester.py and the idle_kernel.py also in this directory.  I am not sure what this file is supposed to be doing or if I need to generate/write my own.  If I need to write my own is there a minimum working example, because the files in the example directory have quite a lot of stuff in them and I don't know what any of them are for?

Thanks for any help

marmeladapk, 2018-06-21:
See here, there here is an updated version: https://m-labs.hk/artiq/manual-master/utilities.html#core-device-configuration-toolOn the first line of device.db you are using there is an address of the board you connect with. Please modify it.Also remember to set this address if you haven't already and connect the board: https://m-labs.hk/artiq/manual-master/installing.html#configuring-the-core-device

mfe5003, 2018-06-21:
Thanks for that link.  The updated manual is helpful.  It still seems to be behind since artiq_coreconfig appears to not be a command anymore.

I set up a mac and a static IP for the kasli (and a rule for mac->ip in the router), flashed it, and now I can ping it on my LAN, however it does not appear on my router's DHCP list.
I set the ip address in device_db.py and rerun the write command from before and now I get a ConnectionResetError.

I also get the same error when I attempt a read:

```
(artiq-dev) ----\Anaconda2\envs\artiq-dev\Lib\site-packages\artiq\examples\kasli_basic>artiq_coremgmt config read mac
Traceback (most recent call last):
  File "----\Anaconda2\envs\artiq-dev\Scripts\artiq_coremgmt-script.py", line 9, in <module>
    sys.exit(main())
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\frontend\artiq_coremgmt.py", line 154, in main
    value = mgmt.config_read(args.key)
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 164, in config_read
    self._read_expect(Reply.ConfigData)
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 115, in _read_expect
    if self._read_header() != ty:
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 109, in _read_header
    ty = Reply(*struct.unpack("B", self._read(1)))
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\coredevice\comm_mgmt.py", line 104, in _read
    raise ConnectionResetError("Connection closed")
ConnectionResetError: Connection closed
```

marmeladapk, 2018-06-21:
@mfe5003 Can you run idle kernel you mentioned in your first post? (artiq_run name_of_experiment.py in the same directory as device_db.py).

mfe5003, 2018-06-21:
@marmeladapk I haven't defined and experiment file yet, but I suspect because I haven't been able to write the idle_kernel to the device yet that it would not work.  I will try to make an experiment though and let you know.

mfe5003, 2018-06-21:
@marmeladapk when I run a modified version of the led.py from the manual:
```
from artiq.experiment import *

class LED(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led1")

    @kernel
    def run(self):
        self.core.reset()
        self.led1.off()
'''

I get the following warning but no error:
```
(artiq-dev) ----\Anaconda2\envs\artiq-dev\Lib\site-packages\artiq\examples\kasli_basic>artiq_run led.py
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (4.0.dev+805.gd0f6123f) and software (4.0.dev+1191.g8b3c12e6) versions
```

Neither led0 or led1 seem to turn on or off on the board.  Right now what I think is led1 is on and I don't see an led0.

marmeladapk, 2018-06-21:
> WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (4.0.dev+805.gd0f6123f) and software (4.0.dev+1191.g8b3c12e6) versions

@mfe5003 Did you use prebuilt gateware+firmware (I don't know if there are any TBH) or did you built them yourself and flashed on Kasli?

This warning shouldn't worry us too much, for simple experiments like that it should be fine.

Try to change ` self.setattr_device("led1")` to ` self.setattr_device("led0")` and do something like:
```
while True:
    self.led0.on()
    delay(500*ms)
    self.led0.off()
    delay(500*ms)
```
One of the front panel leds (AFAIR one under SFP1) should be flashing.

mfe5003, 2018-06-21:
@marmeladapk When I add the loop the command hangs and I can't kill it with ^C.
I can kill it by disconnecting from the LAN, which I'll take as a good sign that I am actually talking to the board.
I don't see any flashing lights though for led0 or led1.

The device_db file I am using has them set up like this at the bottom:
```
device_db.update(
    led0={
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 20}
    },
    led1={
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 21}
    },
)
```

marmeladapk, 2018-06-21:
@mfe5003 Ok, if this doesn't work (and no LED is flashing on front panel nor near power supply LEDs) we need to find out if you're using right variant of gateware with right variant of software. If you're flashed different variant of gateware to the board it will have different numbering of channels. 

What command did you use to flash Kasli via USB?

mfe5003, 2018-06-21:
@marmeladapk I was just following the list of commands from the manual installing. The flashing commands I ran were:
- `artiq_flash -t kasli -V master`
- Installed the drivers with Zadig for 'Quad RS232+HS (Interface 0)'
- `artiq_mkfs flash_storage.img -s mac de:ad:be:ef:00:01 -s ip 192.168.1.150`
- `artiq_flash -t kasli -V master -f flash_storage.img storage start`

I am not used to conda (I typically stick to pip and virtualenv) and I had an older version of artiq on my WSL installation which did still have the artiq_coreconfig script, so I went through and updated both artiq and artiq-kasli-master packages on the vanilla windows conda environment at some point, which might be where the gateware/software disconnect originated from.  Maybe I should have only updated one...

marmeladapk, 2018-06-21:
Ok, so you flashed  `master` variant of gateware, and you are using `tester` variant of device_db.

Run `artiq_flash -t kasli -V tester` and try to run experiment with blinking LED.

mfe5003, 2018-06-21:
It looks like there are no binaries for a tester variant.  I get the following error message:
```
(artiq-dev) ----\Anaconda2\envs\artiq-dev\Lib\site-packages\artiq\examples\kasli_basic>artiq_flash -t kasli -V tester
Traceback (most recent call last):
  File "----\Anaconda2\envs\artiq-dev\Scripts\artiq_flash-script.py", line 9, in <module>
    sys.exit(main())
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\frontend\artiq_flash.py", line 333, in main
    artifact_path(variant, "gateware", "top.bit"))
  File "----\Anaconda2\envs\artiq-dev\lib\site-packages\artiq\frontend\artiq_flash.py", line 316, in convert_gateware
    with open(bit_filename, "rb") as bit_file, \
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\ebert\\Anaconda2\\envs\\artiq-dev\\lib\\site-packages\\artiq\\binaries\\kasli-tester\\top.bit'
```

There also doesn't seem to be a master example device_db

marmeladapk, 2018-06-21:
Could you check what variants are present in C:\\Users\\ebert\\Anaconda2\\envs\\artiq-dev\\lib\\site-packages\\artiq\\binaries? Perhaps we could try with opticlock.

mfe5003, 2018-06-21:
There is only kasli-master in the directory, however the actual path is `envs/artiq-dev/**Lib**/site-packages` which might be part of the problem.
I've noticed before that the windows installation use capital `**Lib**` and the WSL installation uses lowercase `lib` and thought it was just a conda thing.

marmeladapk, 2018-06-21:
Perhaps, but it flashed master somehow. On Windows lower/upper case shouldn't matter. 

Ok, so if you only have binaries for master you should modify device_db to be compatible with master. Make a copy of it and change `led0` channel to `0`, `led1` channel to `1` and you can add `led2` with channel # `2`. Comment out everything from [line 51](https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_basic/device_db_tester.py#L51) to [line 172](https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_basic/device_db_tester.py#L172). These are declarations of devices that aren't used in Master variant. 

Also change [line 42](https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_basic/device_db_tester.py#L42) to range from 3 to 3+8=11. That way we won't have any channel number collision, and everything will be as laid out in target gateware file.

mfe5003, 2018-06-21:
Thanks that worked!  I can see L2 flashing as led1 in the code.

I still can't configure the idle kernel though. I get the same error message as before.
`artiq_coremgmt config write -f idle_kernel idle_kernel.elf`
I can't read or write records either.

mfe5003, 2018-06-22:
So it looks like I can communicate with kasli using `artiq_coremgmt` and the log (0x01) and reboot (0x05) commands seem to work fine.  I can change the log level to debug then try to write a key value pair.

When I send a write (0x0D) or a read (0x0C) command the request type is not recognized and the rest of the payload (key_length, key, value_length, value) shows up on the kasli as a network error.

```
--- artiq_coremgmt log set_level DEBUG ---
[  1044.206952s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57282
[  1044.216769s]  INFO(runtime::mgmt): changing log level to DEBUG
[  1051.699629s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57283
--- artiq_coremgmt config write -s macc test ---
[  1106.840312s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57285
[  1106.851560s] ERROR(runtime::mgmt): aborted: unknown request type
[  1106.865400s] DEBUG(runtime): network error: dropped by socket
[  1107.103589s] DEBUG(runtime): network error: dropped by socket
[  1150.045963s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57286
--- added a 1s wait between sending header, key, and value. artiq_coremgmt config write -s macc test ---
[  1277.413412s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57296
[  1277.423535s] ERROR(runtime::mgmt): aborted: unknown request type
[  1278.431249s] DEBUG(runtime): network error: dropped by socket
[  1278.731951s] DEBUG(runtime): network error: dropped by socket
[  1279.333028s] DEBUG(runtime): network error: dropped by socket
[  1280.514811s] DEBUG(runtime): network error: dropped by socket
[  1287.178971s]  INFO(runtime::mgmt): new connection from 192.168.1.120:57297
```

I can't tell for sure, but it looks like commit b81b20caf8 on 5/16/2018 before artiq_coreconfig was rolled into artiq_coremgmt used a different set of command bytes (artiq/artiq/coredevice/comm_kernel.py) where the FLASH_READ_REQUEST was 0x09 instead of 0x12 like it is now.
I tried swapping the byte manually in comm_mgmt.py but I get the same unknown request type error.

Is there an old commit I can roll back to to get the kasli board working?

marmeladapk, 2018-06-22:
@mfe5003 This is a question to @sbourdeauducq or @jordens. But you don't need a idle kernel to work with Kasli and schedule experiments, it's just an experiment that activates when nothing else is happening (for example to toggle diode).

whitequark, 2018-06-22:
> Is there an old commit I can roll back to to get the kasli board working?

Is there a reason you can't use the same version for gateware, runtime and host software?

mfe5003, 2018-06-22:
@whitequark This is my first time trying to use artiq, so I am trying to figure out how it all works.  I did not intend to use different gateware/firmware.  It seems like I need to use version 4 to use kasli, because conda ends up pulling from the version 4 dev branch when I do:
`conda create -n xxx artiq-kasli-master`

I can try wiping my install and seeing if that fixes the mismatch.

whitequark, 2018-06-22:
If you want to stay on the release versions, you can remove the dev channel from conda instead. Alternatively, you could flash the dev channel gateware using the artiq_flash script.

mfe5003, 2018-06-22:
I rolled back the artiq package to match the kasli-master gateware and I can now read/write to the device.
For reference I just had to run:
`conda install artiq=4.0.dev=805+gitd0f6123f`
where 805+gitd0f6123f is the build string for the gateware.

Thanks for the hand holding while I get started. Hopefully with a basic working system I can figure out my own problems in the future.

closed 2018-06-23 by mfe5003