m-labs/artiq #1080: Sayma PRBS errors

hartytp, 2018-06-22:
Since doing the slave FPGA loading rework and upgrading to the latest master, I've started seeing PRBS errors roughly 100% of the time on boot. I never saw that before. Not sure if it's due to the rework or to changes in the code. IIRC @gkasprow saw this as well...

https://pastebin.com/2X0Y17B6

gkasprow, 2018-06-22:
Maybe this has something to to with TXEN pin of DAC?
I observe this on two boards.

hartytp, 2018-06-22:
Maybe. But it happens on both DACs and we only altered the TXEN on DAC2...

gkasprow, 2018-06-22:
that's true. But this was the only modification I did. There is 3.3V -> 1.8V conversion using 200R resistor that injects current to 1.8V port of DAC and FPGA.  Theoretically the FPGA has protection diodes, but DAC may not like voltage peaks of rougly 2.5V (1.8V + 0.7V of diode). I have no idea how this could affect second DAC channel in such bizarre way.
Another question to @sbourdeauducq is if there is any signal present on FPGA_CFG_DIN during normal operation. Is the logic level low after configuration?

hartytp, 2018-06-22:
My guess was that it's due to one of the recent ARTIQ commits rather than the HW changes. But, I might be wrong -- I haven't given it too much thought yet.

gkasprow, 2018-06-22:
the funny thing is that I started seeing PRBS errors on one board a few days ago, another was workin well. And next day second board also got PRBS "sickness ".

sbourdeauducq, 2018-06-23:
Have you tried ``--without-sawg``? I suspect that the corruption runs deeper than just the SDRAM.

hartytp, 2018-06-23:
Okay. I'll try that and your blinker next to see if we can find some issue with a simpler logic block that we can focus on instead of debugging complex jesd/memory issues.

hartytp, 2018-06-23:
One data point here: running with the SAWG held in reset, I don't see the "crash kernel" crash. But, I do see a bunch of errors during init (JESD PRBS, can't determine SYSREF margin at FPGA)

hartytp, 2018-06-23:
Note to self: try this with a no-sawg build. It would be interesting to see if there is a difference between no SAWG and SAWG in reset. If there is, then this seems much more like a vivado issue than a hardware issue.

hartytp, 2018-06-25:
I rebuilt the current master without SAWG and still see this: https://hastebin.com/fafokucoqa.sql

I have never seen this until recently (around the time that slave loading was added), but it's been in 100% of my recent builds.

@sbourdeauducq do you see this issue on your board?

sbourdeauducq, 2018-06-25:
No I don't.

sbourdeauducq, 2018-06-25:
One change that is likely to have exposed this bug is this:
https://github.com/m-labs/artiq/commit/de7d64d482dcd51755be93040562896528670b68
Try reverting this commit, it is not required for anything other than the DRTIO master with local DACs.

hartytp, 2018-06-25:
I'm currently building with: f9910ab2428e79d39061cd58971a19b77523e389

sbourdeauducq, 2018-06-25:
https://github.com/m-labs/artiq/commit/de7d64d482dcd51755be93040562896528670b68 can be easily reverted on top of master. It's a very simple change (disable the other 7043 clock output - not required in theory but let's be paranoid - and change dac_refclk back to 0). The sysref phase doesn't have an impact on PRBS.

hartytp, 2018-06-25:
https://github.com/m-labs/artiq/commit/f9910ab2428e79d39061cd58971a19b77523e389 works: https://hastebin.com/erulazasek.go

hartytp, 2018-06-25:
Running current master with https://github.com/m-labs/artiq/commit/de7d64d482dcd51755be93040562896528670b68 reverted does indeed work.

sbourdeauducq, 2018-06-25:
@gkasprow @enjoy-digital any idea why that happens? The GTH quad imbalance is the same in both cases, and within spec.

enjoy-digital, 2018-06-25:
Do you have the same behaviour if you only keep https://github.com/m-labs/artiq/commit/de7d64d482dcd51755be93040562896528670b68 but disable the GTP_CLK1 output of HMC7043?

Could it be related to the fact that we now have two reference clocks active and still using QPLLXREFCLKSEL=0b001? (Table 2-8 of UG576)

sbourdeauducq, 2018-06-25:
> Could it be related to the fact that we now have two reference clocks active and still using QPLLXREFCLKSEL=0b001?

The ARTIQ code is not using the QPLL. Should it?

enjoy-digital, 2018-06-25:
Ah sorry we are using CPLL. Then maybe check CPLLREFCLKSEL & https://github.com/m-labs/jesd204b/blob/master/jesd204b/phy/gth.py#L260. Should we connect GTREFCLK1 and set  CPLLREFCLKSEL to 2?

sbourdeauducq, 2018-06-25:
> Should we connect GTREFCLK1 and set CPLLREFCLKSEL to 2?

How would that help?

sbourdeauducq, 2018-06-25:
" a single external reference clock with multiple transceivers connected to multiple Quads. The user design connects the IBUDFS_GTE3 output (O) to the ***GTREFCLK0*** ports of the GTHE3/4_COMMON and GTHE3/4_CHANNEL primitives for the GTH transceiver. 
In this case, the Xilinx implementation tools make the necessary adjustments to the  north/south routing as well as pin swapping necessary to route the reference clocks from one Quad to another when required"

hartytp, 2018-06-25:
> Do you have the same behaviour if you only keep de7d64d but disable the GTP_CLK1 output of HMC7043?

> Could it be related to the fact that we now have two reference clocks active and still using QPLLXREFCLKSEL=0b001? (Table 2-8 of UG576)

@sbourdeauducq do you still want me to do this? I'm a bit short on time atm...

sbourdeauducq, 2018-06-25:
No, that's unlikely to help. The other clock is either unrouted or routed to the other quad for DRTIO. And GTREFCLK0 is the correct setting as per the transceiver user guide I quoted above.

hartytp, 2018-06-25:
@gkasprow remind me, did you try turning the HMC7043 GTP_CLK{1,2} outputs back to LVPECL? Did that help the PRBS errors?

marmeladapk, 2018-06-26:
@hartytp I did help with clock amplitude but not with PRBS errors.

hartytp, 2018-06-26:
@marmeladapk that's what I thought, thanks for confirming. It was a long shot, but I wondered if this was some SI issue related to the low clock amplitudes using LVDS outputs in combination with 200R LVPECL bias resistors. If you've tested that then I won't bother looking at it again.

hartytp, 2018-06-28:
I still see this error after fixing the Vccint supply (I measure 0.951V at the 0R power resistors): https://hastebin.com/hevawerodo.sql

I believe that @marmeladapk also found that the Vccint rework did not help the PRBS errors...

marmeladapk, 2018-06-28:
Yes, today I got these errors once.

sbourdeauducq, 2018-06-29:
@gkasprow Can you check SI and jitter on GTP_CLK1 and GTP_CLK2 on a board that exhibits this problem? And generally investigate this?

gkasprow, 2018-06-29:
I have a few more ideas to verify.
The same board exibits DDR memory errors - some chips are not working, I mean during eye scans, two chips do not have any openings. But the same board with Xilinx memory controller and tester works fine. So it is not broken connection between FPGA and memory.
Yet another board has problem with one DDR memory chip. It used to work some time ago...
What's interesting, write works fine, there are problem with reading.

sbourdeauducq, 2018-06-29:
This doesn't sound related to me:
1. transceiver IOs and clocks are very different than the fabric IOs and clocks (they even have separate power supplies).
2. DDR works correctly on Tom's board that exhibits this error.

marmeladapk, 2018-06-29:
@gkasprow These PRBS errors are there no matter which AMC I use.

hartytp, 2018-06-29:
> Yes, today I got these errors once.

They are there every time on my board :( I can send it back to you if that helps?

sbourdeauducq, 2018-07-04:
@gkasprow Any update on the clock measurements? I know Tom measured them after they went through the FPGA, but I'd like to double-check.

hartytp, 2018-07-05:
@gkasprow what's the timeline for looking at the SI for the two GTP_CLK{1,2} inputs on the AMC FPGA, as well as the SYSREF input?

sbourdeauducq, 2018-07-05:
@hartytp Also: it it only the standalone target that is affected by this crazy problem? What happens with masterdac and satellite?

sbourdeauducq, 2018-07-09:
@gkasprow 
1. Have you tested SI on GTP_CLK1 and GTP_CLK2?
2. Are there any significant hardware differences between the two clocks?
3. Is this problem caused by the RTM side or the AMC side?

jbqubit, 2018-07-09:
@hartytp Is your board still showing PRBS errors?

hartytp, 2018-07-09:
@jbqubit I don't think there have been any relevant commits since I last tested, so presumably yes.

So far I've only tested with the "standalone" variant. I plan to test with a different variant when I have time, although I don't expect it to make a difference.

sbourdeauducq, 2018-07-11:
@hartytp Does it make sense that you get a board that doesn't behave crazy like that, for running your experiment?
Right now the situation on my side is:
* 1 set of boards that has extreme 3.3V problems and is not affected by this; on the way to WUT
* 1 set of boards that has more moderate 3.3V problems, also not affected by this, and which I'm keeping for now so that e.g. HMC7043 work is not completely blocked.
@gkasprow What's the situation on your boards? Any progress with https://github.com/m-labs/artiq/issues/1080#issuecomment-403336469?

hartytp, 2018-07-11:
> Does it make sense that you get a board that doesn't behave crazy like that, for running your experiment?

Could do. That way I can play with/test out some of the more advanced features that we haven't looked at yet.

However, I'd generally prefer to prioritize getting the basics rock solid. Relying on "golden" boards that happen to work can be quite fragile; just because they don't exhibit the problems now doesn't mean they won't develop them soon.

jbqubit, 2018-07-12:
@sbourdeauducq do you see this issue on your board with any of variants?

sbourdeauducq, 2018-07-12:
As pointed out earlier, no.

sbourdeauducq, 2018-07-19:
@gkasprow Can you remember to do those tests? https://github.com/m-labs/artiq/issues/1080#issuecomment-403336469

@enjoy-digital The GTH instance in the JESD204 core does not contain all the parameters and ports of the primitive. This would be fine if Xilinx transceivers did not suck, but they do have a track record of idiotic behavior if anything in their plethora of obscure parameters is not set right, and the default values cannot be trusted. Can you run the wizard and copy-paste every single parameter and constant input port value into the JESD code, like we did to get the transceivers to behave for Ethernet and DRTIO, please?

sbourdeauducq, 2018-07-19:
Also @gkasprow were boards assembled with different silicon revisions of the Ultrascale FPGA?

gkasprow, 2018-07-19:
All FPGAs were purchased at the same time, from same batch so should have same revision.

gkasprow, 2018-07-19:
@sbourdeauducq I remember, but @marmeladapk is on holidays till the end of month.
If it is really urgent, I will need bitfile that enables right clocks on HMC that can load from Vivado and observe the signals

sbourdeauducq, 2018-07-19:
Simplest is probably that you flash ARTIQ into the board. Do you need binaries?

enjoy-digital, 2018-07-19:
@sbourdeauducq: i'll try to do that next week.

gkasprow, 2018-07-19:
@sbourdeauducq Stupid question: how to program the FLASH (under Windows), but not only part for FPGA but also firmware? 
I need to create FLASH config file where I have bitstream and also can load data file with certain start address. I assume there are RTM bit and ARTIQ software at certain offsets. Or you can just send me the FLASH image (bit, bin or rbt file) containing all the binaries.
Unfortunately @marmeladapk is far away from civilisation and I don't remember his computer password. On my account the paths are not set correctly.
The ARTIQ version I have loaded to FLASH on my boards is old and tries to set bit ship on HMC7034 indefinitely. Look at the log file here: 
[log.txt](https://github.com/m-labs/artiq/files/2210221/log.txt)

sbourdeauducq, 2018-07-19:
Install the artiq conda package, set up Zadig (http://m-labs.hk/artiq/manual-master/installing.html?highlight=zadig#configuring-openocd), then run ``artiq_flash -t sayma --srcbuild artiq_sayma`` (with ``artiq_sayma`` the name of the folder in the archive I linked).
Old ARTIQ versions do not configure both HMC7043 GTP outputs anyway.

gkasprow, 2018-07-19:
well, i have some problems: 
conda create -n artiq-main artiq-sayma_amc-standalone
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - artiq-sayma_amc-standalone
  - artiq==4.0.dev=py_420+git3eb882b6

Current channels:

  - http://conda.anaconda.org/m-labs/label/main/win-64
  - http://conda.anaconda.org/m-labs/label/main/noarch
  - http://conda.anaconda.org/conda-forge/label/main/win-64
  - http://conda.anaconda.org/conda-forge/label/main/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/win-64
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/pro/win-64
  - https://repo.anaconda.com/pkgs/pro/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

sbourdeauducq, 2018-07-19:
The ``artiq`` package. Not ``artiq-sayma_amc-standalone``.

gkasprow, 2018-07-19:
OK, success. At least from tools point of view.

sbourdeauducq, 2018-07-20:
@enjoy-digital May also make sense to add some transceiver diagnostics, e.g. to check if the QPLL/CPLL locks. Right now AFAICT the JESD code will continue to proceed with an unlocked PLL.
If the CPLL doesn't lock, then we know we don't have to look further in the transceiver settings.

sbourdeauducq, 2018-07-20:
@enjoy-digital Also, are you initializing the transceiver (including CPLL reset) *after* the HMC7043 is fully initialized?

sbourdeauducq, 2018-07-20:
Also, as you can see I am asking Greg to do some hardware tests in parallel. this project has been dragging on for way too long and we cannot wait for things. Please run the wizard, send the XCI file to Xilinx, and update your code asap.

sbourdeauducq, 2018-07-20:
@gkaprow Do you have a problematic board without major power supply or other bugs? are you sending one to HK?

gkasprow, 2018-07-20:
@sbourdeauducq I'm sending you two board-sets + SFP-SATA converter
One is perfectly fine, second has some issues with SDRAM - it does not pass test, but with Xilinx IP core it works fine - no single error in one hour run. I have two boards with same symptoms so please have a look. What is interesting the problem occurs on 3 DDR chips only and only with writes. The levelling window is simply zero width. On second problematic board which I kept the problem occurs on 1 DDR chip but with Xilinx IP it works fine as well.
I applied all the patches to both AMC and RTM boards.

[98 more replies]

closed 2018-08-17 by sbourdeauducq