m-labs/artiq #1081: RTIO sequence errors when setting multiple outputs and collecting inputs

kesht123, 2018-06-26:
I am running the following experiment with gateware and software at commit 60b2221, and looking at the SERDES TTL signals 1092 and ttl_g8 on a scope. Our target is based off the kc705 NIST clock with some additional TTLs
```python
from artiq.experiment import *

class TestSeqErrBug(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_g8")
        self.setattr_device("_1092")
        self.setattr_device("_422u")
        self.setattr_device("_422l")
        self.setattr_device("_1033")
        self.setattr_device("pmt")

    def run(self):
            self.krun()

    @kernel
    def krun(self):
        self.core.reset()
        for _ in range(1000):
            delay(1*ms)
            self.countFluorescence()

    @kernel
    def countFluorescence(self):
        t_pmt = 800*us
        t_pulse = 800*us
        with parallel:
            self.ttl_g8.on()
            self.pmt.gate_rising(t_pmt)
            with sequential:
                self._422u.on()
                self._422l.on()
                self._1092.on()
                self._1033.on()
                delay(t_pulse)
                self._422u.off()
                self._422l.off()
                self._1092.off()
                self._1033.off()
        self.ttl_g8.off()
        return self.pmt.count()
```
The pmt is a TTLInOut, the rest are TTLOut. Running this experiment generates 29 copies of "firmware.runtime.rtio_mgt:RTIO sequence error involving channel 52" (channel 52 is ttl_g8). I see the 1092 TTL pulse as expected, but ttl_g8 does not.

I have tried to reduce the code for easier repro, but small changes cause the problem to go away; no RTIO sequence errors and ttl_g8 pulses as it should. 
Specifically, changing t_pulse by even +- 1us, removing one of the pulsing TTLs, swapping the sequential block for parallel ttl.pulse(t_pulse) calls for each TTL, and removing the pmt.gate_rising all cause the problem to go away.

Adding another TTL in the sequential block (i.e. pulsing 5 TTL's) causes RTIO sequence errors on channel 20 (channel 20 is the 1092 TTL)

The artiq_coreanalyzer trace shows the expected bug-free behaviour, which does not match reality (i.e. what is actually produced on the TTLs as measured by the scope)

sbourdeauducq, 2018-06-26:
What ARTIQ version?

sbourdeauducq, 2018-06-26:
Try setting ``enable_spread`` to ``False`` in gateware/rtio/sed.

kesht123, 2018-06-26:
In gateware/rtio/sed/core.py, line 15, I changed the default ``enable_spread`` argument to ``False`` in the ``__init__`` of the ``SED`` class. I rebuilt the gateware and flashed it to the core device. Running the same script as above I see the same RTIO sequence errors

sbourdeauducq, 2018-07-02:
Here is a minimized repro:
```python
from artiq.experiment import *

class TestSeqErrBug(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")
        self.setattr_device("ttl1")
        self.setattr_device("ttl2")
        self.setattr_device("ttl3")
        self.setattr_device("ttl4")
        self.setattr_device("ttl_sma")

    @kernel
    def run(self):
        self.core.reset()
        with parallel:
            self.ttl0.on()
            self.ttl_sma.pulse(800*us)
            with sequential:
                self.ttl1.on()
                self.ttl2.on()
                self.ttl3.on()
                self.ttl4.on()
                delay(800*us)
                self.ttl1.off()
                self.ttl2.off()
                self.ttl3.off()
                self.ttl4.off()
        self.ttl0.off()
```

Note the most important changes:
* inputs are not required, toggling a TTL instead is sufficient
* removal of the loop, one iteration is sufficient

@kesht123 Can you try to minimize it to this level next time?

sbourdeauducq, 2018-07-02:
This is not a bug, the core is behaving as expected.
Look at how SED lanes get allocated (the core needs to get a new lane every time the timestamp of an event is not strictly increasing):

```python
    @kernel
    def run(self):
        self.core.reset()
        with parallel:
            self.ttl0.on()  # lane0
            self.ttl_sma.pulse(800*us)  # lane1(rising) lane1(falling)
            with sequential:
                self.ttl1.on()  # lane2
                self.ttl2.on()  # lane3
                self.ttl3.on()  # lane4
                self.ttl4.on()  # lane5
                delay(800*us)
                self.ttl1.off() # lane5
                self.ttl2.off() # lane6
                self.ttl3.off() # lane7
                self.ttl4.off() # lane0
        self.ttl0.off()  # lane1 -> clashes with the falling edge of ttl_sma, which is already at +800us
```

You can do two things to fix this problem in your setup:
* increase the number of SED lanes in your gateware to 16, or
* rewrite the kernel to reduce the number of lane changes, e.g.:
```python
    @kernel
    def run(self):
        self.core.reset()
        self.ttl0.on()     # lane0
        self.ttl_sma.on()  # lane1
        self.ttl1.on()     # lane2
        self.ttl2.on()     # lane3
        self.ttl3.on()     # lane4
        self.ttl4.on()     # lane5
        delay(800*us)
        self.ttl1.off()    # lane5
        self.ttl2.off()    # lane6
        self.ttl3.off()    # lane7
        self.ttl4.off()    # lane0  (no clash: new timestamp is higher than last)
        self.ttl_sma.off() # lane1
        self.ttl0.off()    # lane2
```

For the second solution, it is a bit hacky: for inputs, you'd have to use the private ``TTLInOut._set_sensitivity()`` method (look at the source code of ``TTLInOut.gate_rising()``.

closed 2018-07-02 by sbourdeauducq

jordens, 2018-07-02:
A possible extension to SED would be to also have a sorting network at the cpu side that would make smarter lane choices instead of just advancing, to achieve better packing. I.e. it could choose the lane that has the largest timestamp queued that still meets strict monotonicity.

sbourdeauducq, 2018-07-02:
That would increase RTIO latency further.

sbourdeauducq, 2018-07-02:
But we want to optimize the former, don't we?

jordens, 2018-07-02:
The gateware latency? As long as it's orders of magnitude less than the RTIO API and CPU latency then I'd happily get better lane utilization at the expense of a couple cycles of gateware latency.

charlesbaynham, 2020-06-12:
> A possible extension to SED would be to also have a sorting network at the cpu side that would make smarter lane choices instead of just advancing, to achieve better packing. I.e. it could choose the lane that has the largest timestamp queued that still meets strict monotonicity.

With two more years of perspective, do you guys have any further opinions on this idea? I think it'd be really useful. 

For example, we have an experiment which schedules a really long pulse and then does a lot of other stuff while that pulse continues. The way it currently works, we get a sequence error once the other 7 lanes have all been used once. Naively, I assumed that having 1 lane with an event scheduled way into the future wouldn't affect operation with the other 7 lanes so it was a bit surprising to find out that it does.

sbourdeauducq, 2020-06-13:
The timing (as in FPGA timing closure) of the event submission logic is pretty tight. And we need to be able to determine if the destination lane is full in order to block the kernel and avoid losing events, which obviously requires knowledge of which lane is going to be used. Maybe a suitable heuristic is to block the kernel if any lane (not necessarily the one we are writing to) is full.

sbourdeauducq, 2020-06-13:
> A possible extension to SED would be to also have a sorting network at the cpu side

If we do that (which is a larger change) then a sorting network is not necessary: we can have a ``(lane#, last_timestamp)`` array sorted by ``last_timestamp``, and keep it sorted by shuffling elements every time a ``last_timestamp`` entry is updated (i.e. run one step of bubble sort). This has lower latency than a full sorting network.

charlesbaynham, 2020-06-15:
So you'd go through the list starting with the events furthest in the future, then put your event into the first lane it fits into. Do you think that could happen fast enough block the kernel if required? It'd be amazing to avoid the "any lane is full" heuristic too.

sbourdeauducq, 2020-06-16:
> Do you think that could happen fast enough block the kernel if required?

Note that it would also block DMA.

> It'd be amazing to avoid the "any lane is full" heuristic too.

I don't see a way of avoiding it. Remember that FPGAs are slow.