m-labs/artiq #1088: compiler asserts trying to emit attribute writeback for function pointer lists

klickverbot, 2018-06-30:
As per title – workaround for the symptoms:

```diff
--- a/artiq/compiler/transforms/llvm_ir_generator.py
+++ b/artiq/compiler/transforms/llvm_ir_generator.py
@@ -522,12 +522,15 @@ class LLVMIRGenerator:

             def llrpcattr_of_attr(offset, name, typ):
                 def rpc_tag_error(typ):
-                    print(typ)
-                    assert False
+                    raise NotImplementedError("Error determining RPC tag for: ", typ)

                 if not (types.is_function(typ) or types.is_method(typ) or types.is_rpc(typ) or
                         name == "__objectid__"):
-                    rpctag = b"Os" + self._rpc_tag(typ, error_handler=rpc_tag_error) + b":n"
+                    try:
+                        tag = self._rpc_tag(typ, error_handler=rpc_tag_error)
+                    except NotImplementedError:
+                        return None
+                    rpctag = b"Os" + tag + b":n"
                 else:
                     rpctag = b""

@@ -562,7 +565,9 @@ class LLVMIRGenerator:
                     offset += alignment - (offset % alignment)

                 if types.is_instance(typ) and attr not in typ.constant_attributes:
-                    llrpcattrs.append(llrpcattr_of_attr(offset, attr, attrtyp))
+                    attr = llrpcattr_of_attr(offset, attr, attrtyp)
+                    if attr:
+                        llrpcattrs.append(attr)

                 offset += size
```

(Again, apologies for lack of test case.)

klickverbot, 2018-06-30:
(@whitequark)

whitequark, 2018-07-11:
Function pointer lists should be explicitly skipped, just like functions currently are. I'll take a look.

klickverbot, 2018-07-11:
Great, thanks!

commit fab6e5cd (2018-08-10): compiler: skip functional values in attribute writeback.

Fixes #1088.
commit 77d511dc (2018-08-10): compiler: skip functional values in attribute writeback.

Fixes #1088.
closed 2018-08-10 by whitequark