m-labs/artiq #1092: binaries directory does not exist in main artiq package

philipkent, 2018-07-02:
On windows 64 bit, after creating a conda environment using the artiq-kc705-nist_qc2 package via  `conda create -n artiq-main artiq-kc705-nist_qc2` no binaries directory is created which causes  `artiq_flash` commands to fail:

     (artiq-main) C:\Users\pdk1>artiq_flash -t kc705 -m nist_qc2  -f flash_storage.img storage start
     Binaries directory 'c:\users\pdk1\work\artiq\artiq-dev\artiq\artiq\binaries\kc705-nist_qc2' does not exist

sbourdeauducq, 2018-07-03:
Where is it created instead?

philipkent, 2018-07-03:
It looks like it just wasn't created at all.  I only have a single directory containing the kc705 gateware at artiq-main/lib/python3.5/site-packages/artiq/binaries/kc705-nist_qc2/.  Seems like the artiq package and it's dependencies weren't installed.

sbourdeauducq, 2018-07-03:
Ok. Is there a problem other than the artiq package not getting installed? Are things working correctly after installing the artiq package manually?

philipkent, 2018-07-03:
Yes, everything works correctly after installing the artiq package manually.

sbourdeauducq, 2018-07-03:
So, it's #984.

closed 2018-07-03 by sbourdeauducq