m-labs/artiq #1093: seconds_to_mu is not bound to anything error in v3.6

philipkent, 2018-07-02:
On Windows, 64 bit, ARTIQ 3.6, the following experiment fails to compile stating that 'seconds_to_mu' is not bound to anything

```
from artiq.experiment import *

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    @kernel
    def run(self):
        mu = seconds_to_mu(1)
```

The exact error given by ARTIQ is:
```
root:While compiling <repository>\testing\test.py
<repository>\testing\test.py:11:14-11:27: fatal: name 'seconds_to_mu' is not bound to anything
        mu = seconds_to_mu(1)
             ^^^^^^^^^^^^^   
```

seconds_to_mu also appears to not work in a host method.  The experiment below throws the exception 
``` root:Terminating with exception (NameError: name 'seconds_to_mu' is not defined)
Traceback (most recent call last):
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\master\worker_impl.py", line 273, in main
    exp_inst.run()
  File "C:\users\pdk1\work\artiq\repository\experiments\testing\test.py", line 10, in run
    mu = seconds_to_mu(1, self.core)
NameError: name 'seconds_to_mu' is not defined
```

```
from artiq.experiment import *

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    def run(self):
        mu = seconds_to_mu(1, self.core)
```

klickverbot, 2018-07-02:
This is now `self.core.seconds_to_mu`.

philipkent, 2018-07-02:
@klickverbot thank you.

closed 2018-07-02 by philipkent