m-labs/artiq #1094: urukul profile support

mfe5003, 2018-07-03:
I have an Urukul ad9910 and I can set the output for profile 0 only.
Looking at `coredevice/ad9910.py`, I can obviously see where I would need to change the profile register address to populate other profiles (`coredevice/suservo.py` already implements this I assume).

What I don't see though is an API call for changing the profile pins to actually change the profile in either coredevice.

I can change the profile 0 register stability in about 50 us which would work fine for most things, but not everything.

Thanks

jordens, 2018-07-03:
You should be able to change them much faster at the current spi clock speed. The number should be around 2 us.
Refer to the schematics for the wiring of the profile pins. They are controlled through the cpld and accessible there.

closed 2018-07-03 by jordens

mfe5003, 2018-07-04:
I am trying to follow the logic for the RF switch TTL and it looks like for the channel 0 switch the signal just gets processed directly in the CPLD 1:1 based on the input signal.
So I  am guessing that in order to use the profiles I will need to edit the CPLD logic.

It seems like that should be [here](https://github.com/sinara-hw/sinara/tree/master/ARTIQ_ALTIUM/EEMs/Urukul/CPLD_code) but I don't see any vhdl/verilog in that directory.

I see that I may be able to change the profile using `coredive.CPLD.cfg_write()`, is this what you are referring to?

jordens, 2018-07-04:
Yes. The profile lines are bits 8-10. See `urukul_cfg()`

jordens, 2018-07-04:
And the urukul cpld gateware is at https://github.com/quartiq/urukul