m-labs/artiq #1095: Dashboard doesn't get informations from Zotino

LabVIEWfan, 2018-07-04:
Hello,

We use a Kasli-tester setup, so including Kasli and Zotino, with a recent commit (16b917be). 

The dashboard recognizes the Zotino module from the device database and displays the 32 channels, but their values are never updated (everything is fixed at 0.000%). Everything works well when setting the voltages via a EnvExperiment, as well as reading the voltages via the read_reg() function.

sbourdeauducq, 2018-07-04:
Are other monitoring features e.g
 TTL working correctly?

LabVIEWfan, 2018-07-04:
All the TTL are working correctly through both the EnvExperiment and the dashboard.

sbourdeauducq, 2018-07-15:
Did you create a ``artiq.gateware.rtio.phy.ad53xx_monitor.AD53XXMonitor`` in the gateware?

commit b6c70b3c (2018-07-15): eem: add Zotino monitoring. Closes #1095
closed 2018-07-15 by sbourdeauducq