m-labs/artiq #1097: Unable to building target binaries using the artiq-dev package on Linux

philipkent, 2018-07-05:
On Ubuntu, 64 bit, I am trying to compile the nist_qc2 gateware using the artiq-dev package following the instructions here:

https://m-labs.hk/artiq/manual-release-3/developing.html

When I try to build the target binaries using `python3 -m artiq.gateware.targets.kc705_dds -H nist_qc2` I get the following compilation error:

```
(artiq-dev) philip@sophus:~/artiq-dev/artiq$ python3 -m artiq.gateware.targets.kc705_dds -H nist_qc2
make: Entering directory '/home/philip/artiq-dev/artiq/misoc_nist_qc2_kc705/software/ksupport'
 CARGO    cargo/or1k-unknown-none/debug/libksupport.a
   Compiling ksupport v0.0.0 (file:///home/philip/artiq-dev/artiq/artiq/firmware/ksupport)
error[E0633]: malformed `#[unwind]` attribute
   --> ksupport/eh.rs:410:1
    |
410 | #[unwind]
    | ^^^^^^^^^

error[E0633]: malformed `#[unwind]` attribute
   --> ksupport/eh.rs:428:1
    |
428 | #[unwind]
    | ^^^^^^^^^

warning: #[derive] can't be used on a #[repr(packed)] struct with type parameters (error E0133)
  --> ksupport/eh.rs:57:10
   |
57 | #[derive(Clone, Copy)]
   |          ^^^^^
   |
   = note: #[warn(safe_packed_borrows)] on by default
   = warning: this was previously accepted by the compiler but is being phased out; it will become a hard error in a future release!
   = note: for more information, see issue #46043 <https://github.com/rust-lang/rust/issues/46043>

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0633`.
error: Could not compile `ksupport`.

To learn more, run the command again with --verbose.
Makefile:21: recipe for target 'cargo/or1k-unknown-none/debug/libksupport.a' failed
make: *** [cargo/or1k-unknown-none/debug/libksupport.a] Error 101
make: Leaving directory '/home/philip/artiq-dev/artiq/misoc_nist_qc2_kc705/software/ksupport'
Traceback (most recent call last):
  File "/home/philip/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/philip/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/philip/artiq-dev/artiq/artiq/gateware/targets/kc705_dds.py", line 350, in <module>
    main()
  File "/home/philip/artiq-dev/artiq/artiq/gateware/targets/kc705_dds.py", line 346, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/philip/artiq-dev/artiq/artiq/gateware/amp/soc.py", line 52, in build_artiq_soc
    builder.build()
  File "/home/philip/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/misoc/integration/builder.py", line 185, in build
    self._generate_software()
  File "/home/philip/anaconda3/envs/artiq-dev/lib/python3.5/site-packages/misoc/integration/builder.py", line 160, in _generate_software
    subprocess.check_call(cmd)
  File "/home/philip/anaconda3/envs/artiq-dev/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/philip/artiq-dev/artiq/misoc_nist_qc2_kc705/software/ksupport']' returned non-zero exit status 2
(artiq-dev) philip@sophus:~/artiq-dev/artiq$ 

```

The output of rustc --explain E0633 is:

```
The `unwind` attribute was malformed.

Erroneous code example:

```
#[unwind()] // error: expected one argument
pub extern fn something() {}

fn main() {}
```

The `#[unwind]` attribute should be used as follows:

- `#[unwind(aborts)]` -- specifies that if a non-Rust ABI function
  should abort the process if it attempts to unwind. This is the safer
  and preferred option.

- `#[unwind(allowed)]` -- specifies that a non-Rust ABI function
  should be allowed to unwind. This can easily result in Undefined
  Behavior (UB), so be careful.

NB. The default behavior here is "allowed", but this is unspecified
and likely to change in the future.
```

philipkent, 2018-07-05:
If it helps, I am on the release-3 branch of the artiq source from github.  After creating the artiq-dev conda environment using:

```
conda env create -f conda/artiq-dev.yaml
$ source activate artiq-dev
```

I added the artiq source tree to the environment path using

```
pip install -e .
```

philipkent, 2018-07-05:
The ultimate goal of this is to expand the output FIFO depth to 1024 which we need to avoid underflow errors when the FIFO fills up.

sbourdeauducq, 2018-07-06:
Wrong rustc version. Check that conda installs the one specified in the release-3 artiq-dev meta.yaml.

philipkent, 2018-07-06:
We were able to build the target binaries by manually installing the artiq-dev package and specifying the version using:  `conda create -n artiq-dev artiq-dev=3.6`.  It looks like using the conda/artiq-dev.yaml file from the release-3 branch installs the 4.0 version of the artiq-dev package by default.

sbourdeauducq, 2018-07-07:
Yes, it does. The latest release-3 branch is different from the 3.6 release; and you should use matching ARTIQ source code and artiq-dev/rustc version.

closed 2018-07-07 by sbourdeauducq