m-labs/artiq #1099: restore core device network transfer rates

sbourdeauducq, 2018-07-09:
This was reduced between b81b20caf8361c13f091ad85dcee31dd37d11d72 (good) and 3b61b7c30b08d4393ac09bd8f1ea46af1ebf15c8 (bad).

Corresponding builds:
http://buildbot.m-labs.hk/builders/artiq/builds/2352
http://buildbot.m-labs.hk/builders/artiq/builds/2349

Revert 423929a1 once fixed.

jordens, 2018-07-09:
also c.f. #686 for tracking various aspects of coredevice performance

klickverbot, 2018-07-09:
Not to make the issue tracker degrade into a collection of "+1" posts, but let me just stress again I'm still concerned about network transfer rates for medium-term applications – if there is anything in particular I can do to help with making sure things don't regress to even slower speeds than we are currently at, just let me know. (Being able to stream a few MB/s of diagnostics data without having to put Ethernet MACs on the satellites would be great.)

sbourdeauducq, 2018-07-09:
> if there is anything in particular I can do to help with making sure things don't regress to even slower speeds than we are currently at

A simple thing you could do is identify what exactly caused the slowdown between the two commits I posted, and submit a PR.

klickverbot, 2018-07-09:
Oh, sure, I can spend some time debugging this one issue, and might if it gets critical for us at some point (or investigate alternative hardware platforms). What I meant was more in terms of the bigger picture – I'm quite happy leaving this to whitequark else as there are many other things to spend time on in our experiment, but if there is anything in particular you know you won't get to because there isn't enough time-expertise product, etc., I'd probably be able to spend a few weekends with a profiler, building a more elaborate CI setup, or whatever it is.

whitequark, 2018-08-12:
This has been fixed in master. In fact the transfer speed is slightly higher than it was before, 2.5 MB/s to device, 2.1 MB/s to host.

closed 2018-08-12 by whitequark