m-labs/artiq #1102: Proposal: Make verbosity arguments mutually exclusive

drewrisinger, 2018-07-10:
Wanted to discuss, trivial implementation.

[Current implementation of artiq.tools.verbosity_args()](https://github.com/m-labs/artiq/blob/f8ceea20d07776f3c7f3c6b97d6f23bfc5ebd451/artiq/tools.py#L109) allows setting both "--quiet/-q" and "--verbose/-v" options. **argparse** allows creating [mutually exclusive argument groups](https://docs.python.org/3/library/argparse.html#mutual-exclusion), which would only allow setting either -v or -q options.

The main thing you give up is the ability to give a "title" to the group, which is currently set to "verbosity".

Final line if proposal accepted (remainder of function remains same):
`group = parser.add_mutually_exclusive_group(required=False)`

jordens, 2018-07-10:
Thanks for the report.
Why do you want to do this? I.e. is this a feature request or a bug report and what's the feature/bug?
And if you split the single controlled quantity verbosity into two separate top level groups with different titles, that would make this less readable IMO.
Having `-q` and `-v` accumulate with opposite sign has [well defined semantics](https://github.com/m-labs/artiq/blob/f8ceea20d07776f3c7f3c6b97d6f23bfc5ebd451/artiq/tools.py#L160). If there is an issue with the semantics, could you propose a change to the argument description that resolves this?

drewrisinger, 2018-07-10:
I think the advantage of this is it provides cleaner parsing of the input and prevents users from doing anything too funky on command line. Semantics of logging level would still remain the same.

It would still be in a single group (the verbosity group, just wouldn't be named "verbosity") and this prevents users from doing some confusing syntax like `-vv -q` which is functionally identical to `-v`.

**Proposed method:**
```
>>> import argparse
>>> p1 = argparse.ArgumentParser()
>>> meg = p1.add_mutually_exclusive_group(required=False)
>>> meg.add_argument("-v", "--verbose", default=0, action="count",
                       help="increase logging level")
>>> meg.add_argument("-q", "--quiet", default=0, action="count",
                       help="decrease logging level")
>>> p1.print_help()
usage: pydevconsole.py [-h] [-v | -q]
optional arguments:
  -h, --help     show this help message and exit
  -v, --verbose  increase logging level
  -q, --quiet    decrease logging level
>>> p1.print_usage()
usage: pydevconsole.py [-h] [-v | -q]
```
**Existing Method**
```
>>> p2 = argparse.ArgumentParser()
>>> vgroup = p2.add_argument_group('verbosity')
>>> vgroup.add_argument("-v", "--verbose", default=0, action="count",
                       help="increase logging level")
>>> vgroup.add_argument("-q", "--quiet", default=0, action="count",
                       help="decrease logging level")
>>> p2.print_help()
usage: pydevconsole.py [-h] [-v] [-q]
optional arguments:
  -h, --help     show this help message and exit
verbosity:
  -v, --verbose  increase logging level
  -q, --quiet    decrease logging level
```

jordens, 2018-07-10:
You can `ArgumentParser().add_argument_group("verbosity").add_mutually_exclusive_group()`. That makes me happy. Want to spin a PR?

sbourdeauducq, 2018-07-11:
One advantage of the current behavior is, if you're running an artiq tool from a script, you can tune the verbosity level again by adding -v/-q to any of those flags that the script may already have used.

jordens, 2018-07-11:
That's also why I like it the way it is. I'd only make it mutually exclusive if it really causes confusion.

sbourdeauducq, 2018-07-15:
@drewrisinger Unless you insist, we'll keep the current behavior.

closed 2018-07-15 by sbourdeauducq