m-labs/artiq #1108: Assigning pointer to entry in 2D-array gives a compilation error

philipkent, 2018-07-17:
In artiq 3.6, the following gives a compilation error due to the line `j = i` where the local pointer i is being copied to j.  

```
from artiq.experiment import *

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')
        self.array = [[0, 0], [0, 1]]

    @kernel
    def run(self):
        i = self.array[0]
        print(i)
        j = i
        print(j)
```

The above gives an AssertionError error: 
```
root:Terminating with exception (AssertionError)
Traceback (most recent call last):
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\master\worker_impl.py", line 273, in main
    exp_inst.run()
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\coredevice\core.py", line 124, in run
    self.compile(function, args, kwargs, set_result)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\coredevice\core.py", line 104, in compile
    attribute_writeback=attribute_writeback)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\module.py", line 71, in __init__
    escape_validator.visit(src.typedtree)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 273, in visit_ModuleT
    self.visit_in_region(node, None, node.typing_env)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 266, in visit_in_region
    self.generic_visit(node)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 278, in visit_FunctionDefT
    args={ arg.arg: Argument(arg.loc) for arg in node.args.args })
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 266, in visit_in_region
    self.generic_visit(node)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 330, in visit_Assign
    self.visit_assignment(target, node.value)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 318, in visit_assignment
    if not Region.outlives(value_region, target_region):
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 64, in outlives
    assert not lhs.intersects(rhs)
  File "C:\Users\pdk1\AppData\Local\Continuum\anaconda3\envs\artiq-3.6\lib\site-packages\artiq\compiler\validators\escape.py", line 46, in intersects
    assert self.range
AssertionError
```

Compilation succeeds if the `j=i` section is removed.  For example, the below code compiles just fine.

```
from artiq.experiment import *

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')
        self.array = [[0, 0], [0, 1]]

    @kernel
    def run(self):
        i = self.array[0]
        print(i)
```

philipkent, 2018-07-17:
ok, tx.

sbourdeauducq, 2021-12-13:
Not a bug in NAC3.
```python
from numpy import int32
from min_artiq import *

@nac3
class Demo:
    core: KernelInvariant[Core]
    array: list[list[int32]]

    def __init__(self):
        self.core = Core()
        self.array = [[23, 15], [93, 10]]

    @kernel
    def run(self):
        i = self.array[0]
        print_int32(i[0])
        print_int32(i[1])
        j = i
        print_int32(j[0])
        print_int32(j[1])

if __name__ == "__main__":
    Demo().run()
```

```
> python demo.py
> ../../target/release/runkernel ./module.elf
print_int32: 23
print_int32: 15
print_int32: 23
print_int32: 15

```

closed 2021-12-13 by sbourdeauducq