m-labs/artiq #1110: Irregular delays when handling Novogorny sample output

vmsch, 2018-07-19:
Processing of output values of the Novogorny sample() function requires long delays, that considerably slow down the effective sample rate.
For example, saving the sample() output in a class attribute requires 10x as long as saving it in a local variable.
However, passing a local array of dimension >1 filled with sample() outputs as an argument to an asynchronous rpc function also causes RTIO underflows and takes considerably longer than passing a local array of the same size with fixed values as argument.
My questions are
(i) why is there a difference in the duration of writing the output to a class attribute or a local variable?
(ii) why is it so slow to pass the local array as an argument?
(iii) is there a way to readout data faster than case 2 in the below example, where data is written to a class attribute with long delays, and if yes, how?
Issues (i) and (ii) might also be unrelated to each other.

Here are two examples to demonstrate and reproduce the two problems:
(The code was run on a Kasli v1.0 controller with the opticlock gateware (commit 46fb5ad) connected to a Novogorny rev1.1)

1. Different behaviour between saving to class attribute and local variable:

```
class Test_array_saving(EnvExperiment):
    """Test array saving"""
    def build(self):
        self.setattr_device("core")
        self.setattr_device("novogorny0")

        self.n_channels = 2
        self.voltages = [0.0]*self.n_channels
        self.voltage = 0.0
        self.number = 3.05

    def run(self):
        print("Test Novo")
        dt = self.runScan()
        print(dt)

    @kernel
    def runScan(self):
        self.core.break_realtime()
        voltages = [0.0]*self.n_channels
        delay(5*ms)
        dt = 0

        for jj in range(self.n_channels):

            ## gives RTIO underflow error, needs 90us delay to work, duration: 50856mu
            t0 = self.core.get_rtio_counter_mu()
            self.voltages[jj] = self.novogorny0.sample()
            dt = self.core.get_rtio_counter_mu() - t0
            delay(90*us)

            # ## gives RTIO underflow error, needs 90us delay to work, duration: 50296
            # t0 = self.core.get_rtio_counter_mu()
            # self.voltage = self.novogorny0.sample()
            # dt = self.core.get_rtio_counter_mu() - t0
            # delay(8*us)

            # ## does run, duration: 6888
            # t0 = self.core.get_rtio_counter_mu()
            # voltages[jj] = self.novogorny0.sample()
            # dt = self.core.get_rtio_counter_mu() - t0
            # delay(8*us)

            # ## does run, duration: 448
            # t0 = self.core.get_rtio_counter_mu()
            # self.voltages[jj] = self.number
            # dt = self.core.get_rtio_counter_mu() - t0

        return dt
```
2. Problems when further processing the local variable array

```
class Test_async_saving(EnvExperiment):
    """Test async saving"""
    def build(self):
        self.setattr_device("core")
        self.setattr_device("novogorny0")

        self.n_channels = 10
        self.voltages = [0.0]*self.n_channels
        self.number = 3.05239487562039

    def run(self):
        print("Test Novo Async")
        dt = self.runScan()
        print(dt)

    @kernel
    def runScan(self):
        self.core.break_realtime()
        voltages = [0.0]*self.n_channels
        voltages_v2 = [0.0]*self.n_channels
        dt = 0

        for jj in range(self.n_channels):

            ## case 1
            ## gives rtio underflow for self.n_channels > 1, when call self.save_value(voltages)
            voltages[jj] = self.novogorny0.sample()
            delay(8*us)

            # ## case 2
            # ## does run, duration: 52904
            # self.voltages[jj] = self.novogorny0.sample()
            # delay(90*us)

            # case 3
            ## does run
            # voltages[jj] = self.number # a
            # self.voltages[jj] = self.number # b

        ## use with case 1, or case 3a
        ## case 1:  duration with n_channels=1:  122784
        ## case 1:  duration with n_channels=10: ???, long
        ## case 3a: duration with n_channels=10: 196432
        t0 = self.core.get_rtio_counter_mu()
        self.save_value(voltages)
        dt = self.core.get_rtio_counter_mu() - t0

        # ## case 1 comparison with pre-set numbers
        # ## duration with n_channels=1 : 116000
        # ## duration with n_channels=10: 195640
        # t0 = self.core.get_rtio_counter_mu()
        # self.save_value(voltages_v2)
        # dt = self.core.get_rtio_counter_mu() - t0

        # ## use with case 2, or case 3b
        # ## duration with n_channels=10: 80720
        # t0 = self.core.get_rtio_counter_mu()
        # self.save_value_v2()
        # dt = self.core.get_rtio_counter_mu() - t0

        return dt

    @rpc(flags={"async"})
    def save_value(self, voltages):
        for channel,voltage in enumerate(voltages):
            self.set_dataset("test.channel{:.0f}".format(channel),voltage,broadcast= True, persist = True)

    @rpc(flags={"async"})
    def save_value_v2(self):
        for channel,voltage in enumerate(self.voltages):
            self.set_dataset("test.channel{:.0f}".format(channel),voltage,broadcast= True, persist = True)

```

jordens, 2018-08-01:
Thanks for the report. To triage:

* Does this also occur with Sampler?
* Your measurement strategy (that `dt`) is broken. You want to do `t0 = now_mu()`. Otherwise you are wrongly counting whatever the slack was before calling `sample()`.
* What's the problem with your second code snippet? As documented, `sample()` eats all slack. On the second invocation you are bound to get an underflow because 8 µs is not enough.
* Your `save_value_v2()` doesn't work. We don't do attribute writeback before calling an RPC, only when returning from a kernel.

* If after all this there is still a difference between an attribute and a local variable the suspect would be attribute writeback.

jordens, 2018-08-14:
Ping @vmsch

vmsch, 2018-08-15:
Sorry for the slow reply.
I tested it with the Sampler, and there is no difference between saving to local or member variables with the sampler.
It still requires a very long delay of 110us when trying to sample repeatedly using self.sampler0.sample(voltages) however.
This delay is only 5*us when using sample_mu. Maybe some people would find it useful to have a comment in the code or documentation, that the difference in speed between the two functions is this large, as this seriously affects the possible sampling rate?

To your other comments:
(2): I think it still gives a roughly accurate value, as I adjusted the delays to just not give an RTIO underflow error, and it runs the loop repeatedly? When using your method I get similar values.
(3): Ok. This was just a hack trying to increase the sampling rate, but since it works with the sampler it is no longer necessary.
(4): Yes

I guess since the Novogorny has been replaced by the sampler and the problem doesn't occur with the sampler this shouldn't be an issue then.

ChenGhua1, 2018-09-05:
I also met the same question.
The delay is very long(~100us) when using sample(). Is there any way to improve the sampling rate?

jordens, 2018-10-24:
The actual sample rate is much faster (see the `sample_mu()` speed). Sample scaling is limited by floating point speed #535 #1007

closed 2018-10-24 by jordens