m-labs/artiq #1113: Setting TTL input gates from DMA doesn't seem to work

klickverbot, 2018-07-19:
(Failing) test case here: https://github.com/m-labs/artiq/pull/1112

This is the loopback test, but with the output events (including the TTL counter sensitivity settings) generated via DMA. `count()` after the sequence returns incorrect values.

Another test case with debug output, where exactly the same sequence without DMA works without issues: https://gist.github.com/klickverbot/b274840ce8e2372d406767a41e20140f.

klickverbot, 2018-07-20:
The timeline cursor seems to be advanced correctly: https://gist.github.com/klickverbot/c50e6e89a0752ae809ba5dc1caa4436c

sbourdeauducq, 2018-07-20:
What about the analyser, anything suspicious there? The raw records should contain the gate events.

klickverbot, 2018-07-21:
Oh, the problem here is much more pedestrian: DMA of course doesn't advance `TTLInOut.i_previous_timestamp`, so `count()` still uses the timeout from when the sequence was recorded (i.e. `gate_*` was actually called). Same would go for output events and `.sync()`, although that is much less frequently used.

The fact that three experienced devs looked at this and failed to notice the issue, obvious as it may be in retrospect, probably indicates that we have an usability issue here. However, I'm not sure what to do about this – "attribute writeback" for DMA would be rather nontrivial to implement, and a questionable design in the first place (especially across kernels).

An example illustrating this in the DMA docs will certainly help to some extent, but any other ideas on how to address this?

jordens, 2018-07-21:
Maybe invalidate `i_previous_timestamp` in `count()` (and elsewhere) to prevent it being used wrong in the DMA case and describe how to manually compute it and set it when doing `count()` ing a DMA gate. Or change the TTL api to enforce returning the gate closure time from `gate()` and require using that in `count()` (and elsewhere).

klickverbot, 2018-07-21:
Requiring a timestamp argument to `count` is perhaps the cleanest way to solve this. I was going to add this anyway to be able to write correct user code without fiddling with `i_previous_timestamp` or directly using `rtio_input_timestamp`, but we could remove the no-timestamp version with the next major release.

We should probably ditch `sync()` entirely as well.

jordens, 2018-07-21:
I'd like to have `sync_mu(t_mu)` as a method of the core device though. It's a frequent pattern.

klickverbot, 2018-07-21:
Indeed, and currently a rather clunky one (manually writing out the loop), unless I've missed something.

jordens, 2018-11-02:
Analogous to this and to #1003 the following states are also potential sources for problems as a side effect of recording DMA (tracked state becoming inconsistent with implied hardware or model state):

* `spi2`: `xfer_period_mu`
* `ad9914`: `continuous_phase_comp` and `phase_mode`
* `ad9910`: `phase_mode`.

(The two `phase_mode` instances somewhat less so since they don't mirror hardware state.)
I am ok with leaving them as is for now and add warnings.

sbourdeauducq, 2018-11-04:
``xfer_period_mu``? DId you mean ``xfer_duration_mu``?

jordens, 2018-11-04:
I did and I do.

sbourdeauducq, 2018-11-15:
> I am ok with leaving them as is for now and add warnings.

Add warnings in the documentation?

jordens, 2018-11-15:
Yes.

commit a3e0b1c5 (2018-11-17): ad9914,spi2: add warnings about driver state and DMA. Closes #1113
closed 2018-11-17 by sbourdeauducq