m-labs/artiq #1114: mutate_dataset() fails to find broadcasted datasets

jonaskeller, 2018-07-20:
Using 3.6 py_42+git3e7cdaa5 on Windows 7.

Despite the fact that the dataset `test` exists and is broadcasted, the experiment below triggers this exception:
https://github.com/m-labs/artiq/blob/17665c7271fcb16482f5bdf999a9e8e149042c20/artiq/master/worker_db.py#L215

    from artiq.experiment import *

    class mutateDatasetTest(EnvExperiment):
        '''mutate dataset test'''
        def build(self):
            pass
            
        def prepare(self):
            # self.set_dataset('test', [0., 0.], persist=True)
            self.mutate_dataset('test', 0, 1.)

        def run(self):
            pass

Local datasets are found, i.e. the exception is not raised if the commented line is added.

jordens, 2018-08-01:
I think the point is that you can't mutate a dataset that you have not created yourself. Allowing that would make the semantics of the dataset in the archive ambiguous. If that's not clear from the documentation it should be explained there.

jonaskeller, 2018-08-01:
Ok, thanks for clarifying. If there's a reason for this behavior, I guess I will work it around with something like 

    def mutate_broadcasted_dataset(self, key, index, value, persist=True, save=True):
        try:
            self.mutate_dataset(key, index, value)
        except KeyError:
            temp = self.get_dataset(key)
            self.set_dataset(key, temp, broadcast=True, persist=persist, save=save)
            self.mutate_dataset(key, index, value)
            
Not pretty, but it'll work.

I agree that it would be helpful to point out that `mutate_dataset` only works on datasets created within the same experiment, e.g. at http://m-labs.hk/artiq/manual-release-3/core_language_reference.html?highlight=mutate_dataset#artiq.language.environment.HasEnvironment.mutate_dataset .

jordens, 2018-08-14:
Does that really work? It should suffer from the same semantics problem and I remember seeing a proper error for that. You can't both get and set a dataset in the same experiment.

jonaskeller, 2018-08-14:
I tested that code before posting it (and again just now to be safe) by doing the following:

Generate a dataset using

        self.set_dataset('test', [0.,0.], persist=True)

then run

	from artiq.experiment import *

	class mutateDatasetTest(EnvExperiment):
		'''mutate dataset test'''
		def mutate_broadcasted_dataset(self, key, index, value, persist=True, save=True):
			try:
				self.mutate_dataset(key, index, value)
			except KeyError:
				temp = self.get_dataset(key)
				self.set_dataset(key, temp, broadcast=True, persist=persist, save=save)
				self.mutate_dataset(key, index, value)
			
		def prepare(self):
			self.mutate_broadcasted_dataset('test',0,1.)

		def run(self):
			pass

The first entry of `test` will now show up as `1.` in subsequent experiments.
The method also changes the `persist` and `save` attributes of the dataset as expected when those arguments are given. If the dataset doesn't exist, a `KeyError` exception is thrown.

That is exactly the behavior I wanted to achieve. Am I missing something?

jordens, 2018-08-14:
```python
		temp = self.get_dataset(key)
		self.set_dataset(key, ...)
```
should hit

https://github.com/m-labs/artiq/blob/2463e5667d1b226fff8c59b2afcd2aa9a6aae5c6/artiq/master/worker_db.py#L191-L193

jonaskeller, 2018-08-14:
It does, and the archived value is the one from before the `mutate`.
I see your ambiguity concern now, but while I don't know a good way around it, it's not worth giving up the ability to mutate broadcasted datasets at all (at least in my use case). And since the warning is triggered, the behavior is at least not unexpected.

jordens, 2018-08-14:
Reproducibility and data provenience. The typical way to implement this is to have one experiment that evolves data with `x=get_dataset("x_ref"); ...; {set,mutate}_dataset("x", ...)` and another that "commits" by doing `set_dataset("x_ref", get_dataset("x")`.

sbourdeauducq, 2018-10-22:
Can we close this?

jonaskeller, 2018-10-22:
Sure. Sorry - forgot that this was still open.

closed 2018-10-22 by jonaskeller