m-labs/artiq #1117: Artiq 3.6, using an 80,000 element array in a kernel method never returns

philipkent, 2018-07-24:
Is there a limit to the size of an array that can be used in a kernel method?  I have an experiment that is running in Artiq 3.6, on Windows 64 bit that creates an array, set's it's first element to zero on the kernel, but then never returns:

The experiment I'm running is:
```
from artiq.experiment import *

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    def prepare(self):
        size = 80000
        self.data = [0 for _ in range(size)]

    def run(self):
        self.run_core()
        print('done')

    @kernel
    def run_core(self):
        self.data[0] = 1

```
Which never prints 'done' and the experiment stays running in the schedule window of the dashboard.  If I set the size to 70,000 the kernel method does return though, 'done' is printed and the experiment completes.

In either case, the last two debug messages in the logs are:
```
artiq.coredevice.comm_kernel:rpc service: [1]<function Core.run.<locals>.set_result at 0x0000000000EAE730> (async) [None] {} -> b'n'
artiq.coredevice.comm_kernel:receiving message: type=<_D2HMsgType.RPC_REQUEST: 10>
```

philipkent, 2018-07-24:
If I change `self.data` to a numpy array, that fixes it and the kernel method does return.

```from artiq.experiment import *
import numpy as np

class Test(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    def prepare(self):
        size = 80000
        self.data = np.array([0 for _ in range(size)])

    def run(self):
        self.run_core()
        print('done')

    @kernel
    def run_core(self):
        self.data[0] = 1

```
Should numpy arrays always be used instead of vanilla python arrays?

jordens, 2018-08-01:
I would generally use arrays. But it shouldn't hang like that. Is there anything on the UART?

whitequark, 2018-08-01:
It almost certainly hangs in the type inferencer, which has to typecheck the entire 80000 elements. And yes, numpy arrays are preferred in kernels for this reason.

dhslichter, 2018-10-01:
@whitequark @jordens @sbourdeauducq it would be very helpful if these sorts of distinctions (subtle issues that can dramatically impact efficiency of operation) could have some brief mentions in the compiler section of the manual.  I feel like a number of ARTIQ adopters might get very frustrated by issues like this and #709 and #804, and if it's in the manual as "known bug, fix in progress" that would help.

pca006132, 2021-07-02:
It seems that this is not an issue anymore with the current master.

closed 2021-07-02 by pca006132