m-labs/artiq #1124: Sayma-BugⓇ: SDRAM breaks with some bitstreams on some boards

sbourdeauducq, 2018-08-03:
Symptom when that happens: ``Memory test failed (294860/4458496 words incorrect)``

VCCINT rework applied and voltage at regulator output measured at 0.96V. Voltage measured by XADC 0.948V.

Some boards never have this problem.

Does not appear to be triggered by SAWG (building with SAWG can trigger the problem, but sometimes the problematic board runs fine with another bitstream build that has the SAWG).

gkasprow, 2018-08-10:
Is it on same board that has problem with read/write leveling?

hartytp, 2018-11-28:
@sbourdeauducq what's the status of this? Is there anything else we should do before finalising plans for hardware v2.0? Do you think it's likely this could be a gw/fw bug, or do you think it's almost certainly a hw issue?

sbourdeauducq, 2018-11-28:
No idea.

hartytp, 2018-11-28:
@sbourdeauducq so, are you happy for us to press on with v2.0 as things are now? Or, is there anything else you want us to do first?

sbourdeauducq, 2018-11-28:
Yeah, let's make more boards and see what happens. The current lack of usable boards is making trench warfare against bugs like this particularly difficult anyway.

hartytp, 2018-11-28:
Sounds like a good plan.

jbqubit, 2018-12-13:
Agreed. Postpone investigation until Sayma v2 hardware is available.

sbourdeauducq, 2019-01-09:
This bug seems not to happen anymore with Vivado 2018.3; I loaded three different bitstreams into the problematic board and the SDRAM worked every time.

@gkasprow / @marmeladapk Can you test other boards?

This is good since the board I was using before has developed a power supply bug...

sbourdeauducq, 2020-01-19:
I have not seen this at all on Sayma v2. @marmeladapk if you experienced any problem like this on your boards, please reopen.

closed 2020-01-19 by sbourdeauducq