m-labs/artiq #1126: pmt readout slow

jbqubit, 2018-08-05:
Something blocks for 135 ms ever 50 ms when PMT is readout. Removing ```self.pmt.count()``` is commented out the blocking stops. 

```
from artiq.experiment import *
import numpy as np

class MasterIsSlow(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        self.setattr_device("led0")
        self.setattr_device("pmt")
        self.nreps = 50

    @kernel
    def mykernel(self):
        self.core.break_realtime()
        self.led0.on()
        c = 0 
        for i in range(self.nreps):
            self.pmt.gate_rising(1000*ms/self.nreps)
            delay(1000*ms/self.nreps)
            c = self.pmt.count()
        self.led0.off()
        delay(10*us)

    def run(self):
        self.core.reset()
        while True:
            self.mykernel()
```

sbourdeauducq, 2018-08-05:
How do you measure the "block"?

jbqubit, 2018-08-07:
>> How do you measure the "block"?

``artiq_coreanalyzer`` and watching LED flashing.

 If ``c = self.pmt.count()`` is replaced by `` c = i `` the long delays go away.

sbourdeauducq, 2018-08-07:
> watching LED flashing.

Really? Isn't the LED a RTIO device?
Is there anything in the log? What ARTIQ version?

jbqubit, 2018-08-07:
Nothing unusual in the log. 4.0.dev+1298.ge4d48a78;opticlock

sbourdeauducq, 2018-08-07:
Could be the compiler getting slow for some reason; try using the usual Python profiling tools, and minimizing it by removing the core device interaction (e.g. like ``artiq_compile``).

jbqubit, 2018-08-07:
See attached .vcd. 
[rtio.vcd.zip](https://github.com/m-labs/artiq/files/2264734/rtio.vcd.zip)

jbqubit, 2018-08-07:
Compiler only runs once per call to ``artiq_run``. How can that be the cause?

sbourdeauducq, 2018-08-07:
No, it runs every time you call a kernel from outside a kernel.

klickverbot, 2018-08-07:
(And removing `count()` will cause the kernel to return early, whereas with it in, it needs to wait for the RTIO timeline to reach the current cursor.)

jbqubit, 2018-08-08:
> And removing count() will cause the kernel to return early

That clarifies how in the absence of count() the slowdown isn't observed. 

Still not understood is why this happens in the first place. It seems like only sometimes that there is the slowdown.

jbqubit, 2018-08-15:
No network problems that I've seen. Running Ubuntu with Linux kernel 4.15.0-30. 

> Have you reproduced it with artiq_compile?

I tried running the same program as idle kernel and don't see the pauses. 

I'm uncertain about the execution model for idle and startup kernels that involve methods not marked by @kernel (or engage in RPC). I suspect they're only compiled once and RPCs are silently dropped since they're designed to run when Master is unavailable.

sbourdeauducq, 2018-08-15:
Likely the "pauses" are in the compiler, or the rest of the Python program on the PC, or in your network. Try profiling artiq_run plus checking wireshark traces for problems.

jbqubit, 2018-08-15:
@sbourdeauducq Have you tried my code? If you can't reproduce I'll look to a problem that's unique to my setup.

jordens, 2018-10-09:
This will generally not work. `cound(); ...; self.led0.off()` and `count(); ...; gate_rising()` are guaranteed to see an `RTIOUndeflow` the way it's written. The 135 ms are probably exactly the compiler overhead.

sbourdeauducq, 2018-10-22:
@jbqubit Is this still an issue?

closed 2019-03-15 by sbourdeauducq