m-labs/artiq #1129: HMC7043 cycle slip breaks DAC sync

sbourdeauducq, 2018-08-07:
Using the HMC7043 cycle slip feature intermittently causes breakage of the DAC synchronization, which is manifested by a 53.33ns slip between the two DAC chip outputs.

The magnitude of the slip is always 53.33ns and can be in either direction.

53.33ns corresponds to 8 cycles of the 150MHz clock.

Those phase slips are used for synchronizing DACs to the RTIO timestamp counter. If only 8 DAC channels on one Sayma are required, a workaround is to disable this synchronization; then the DACs are always properly synchronized with each other.

I am pretty sure that the problem comes from this obnoxious HMC7043 acting up again (i.e. it produces wrong SYSREF phases) as the DAC is programmed to ignore SYSREF while the RTIO/SYSREF alignment is running.

Maybe someone with a fast scope, e.g. @hartytp or @gkasprow can confirm? It should be sufficient to measure the time difference of SYSREF edges between DAC0 and DAC1, and confirm that it is incorrectly varying between board reboots. (Those two SYSREF should actually occur at the same time, i.e. the time difference should be ~0, since everything is delay-matched)

Replacing ``sysref_rtio_align`` with this  code is sufficient to reproduce the problem:
```
fn sysref_rtio_align(phase_offset: u16) -> Result<(), &'static str> {
    info!("aligning SYSREF with RTIO...");

    // meet setup/hold (assuming FPGA timing margins are OK)
    hmc7043::sysref_offset_fpga(phase_offset);
    // get to the edge of the 0->1 transition (our final setpoint)
    let mut count = 0;
    //while !sysref_sample() {
    for _ in 0..14 {
        hmc7043::sysref_slip();
        count += 1;
    }
    info!("  ...done, count={}", count);

    Ok(())
}
```

hartytp, 2018-08-09:
> Maybe someone with a fast scope, e.g. @hartytp or @gkasprow can confirm? It should be sufficient to measure the time difference of SYSREF edges between DAC0 and DAC1, and confirm that it is [...]

@marmeladapk are you okay to do this, as you have boards up and running? If not, I'll do it when I get my board back from you...

hartytp, 2018-08-09:
Can you align SYSREF onto the RTIO clock before initializing the DACs? Maybe mute the HMC7043 outputs that go to the DACs?

sbourdeauducq, 2018-08-09:
You can try that, it's straightforward on the standalone target - but I doubt that'll make a change. The DACs are in one-shot sync mode now, and the code does not trigger any syncs until SYSREF/RTIO alignment is done. As I wrote, I suspect the HMC7043 sometimes permanently outputs broken SYSREFs after you use the cycle slip.

hartytp, 2018-08-09:
okay, well it's easy to look at the two sysrefs and verify that their phase is constant w.r.t. each other and to the DAC clock. That's a good starting point.

hartytp, 2018-08-10:
@gkasprow /@marmeladapk are you okay to look at this?

marmeladapk, 2018-08-10:
>  As I wrote, I suspect the HMC7043 sometimes permanently outputs broken SYSREFs after you use the cycle slip.

@sbourdeauducq 
Does sysref have 300 MHz frequency?  That's what I see on one DAC.

@hartytp 

Yes, after we finish testing boards for Duke.

sbourdeauducq, 2018-08-10:
> Does sysref have 300 MHz frequency?

Hahaha no. Mayba that's why things don't work...

sbourdeauducq, 2018-08-10:
https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard_artiq/hmc830_7043.rs#L165

Are they all 300MHz? Any idea why?

hartytp, 2018-08-11:
> Does sysref have 300 MHz frequency? That's what I see on one DAC.

wft? Are you sure?

marmeladapk, 2018-08-11:
@hartytp @sbourdeauducq Nah, I misread @sbourdeauducq's comment. I've got 300 MHz coming out of one DAC (to AFE) no matter which frequency I set.

marmeladapk, 2018-08-11:
@hartytp 

https://github.com/m-labs/artiq/blob/master/artiq/examples/sayma_standalone/repository/sines.py

sbourdeauducq, 2018-08-11:
That's probably another board-specific bug, can you locate a board that doesn't do that and then focus on SYSREF?

hartytp, 2018-08-11:
@marmeladapk odd. How are you loading that? Are you flashing it as a startup kernel?

sbourdeauducq, 2018-08-12:
sines.py is working correctly for me, both as a startup kernel and over Ethernet. Let's keep this issue on DAC sync.

sbourdeauducq, 2018-08-17:
@marmeladapk Any progress with measuring SYSREF?
@hartytp Did you get your board back and are you able to do this measurement as well?

hartytp, 2018-08-17:
@sbourdeauducq nope, I don't have it yet. FYI: I'm away next week, so the earliest I can look at this now is the 27th, assuming I get the hardware back by then.

sbourdeauducq, 2018-08-19:
@gkasprow Where are the best points to measure SYSREF (DAC0, DAC1 and FPGA_DAC_SYSREF)?

sbourdeauducq, 2018-08-21:
Also, is there an easy-enough way we can disconnect SYSREF for both DACs from the HMC7043, and connect them to the FPGA directly? That certainly will work for 600MHz and perhaps 2000MHz since the FPGA IOs are advertised to support 2133Mbps DDR4. Note that DDR4 uses source-synchronous clocking with DQS, and they probably have a lower-noise clocking path inside the "native" I/O block to support that rate. If necessary (we'd need to experiment...) that can be handled by using an external clock fanout on the PCB that drives DQS pins, but I'm not sure which is the biggest trash fire - HMC chips or Ultrascale I/O native mode.

hartytp, 2018-08-21:
@sbourdeauducq that might well work, but I’m still a bit concerned with how far the gateware design from ARTIQ is diverging from the way that *everyone* else does this and turning into a NIH reinvention of the wheel.

As I said, look in the JESD app notes from ADI etc and you’ll see only one way this is done: using the 7044 as the SYNC source and feeding the SYNC/TRIGGER pulse that’s sampled from the 100MHz clock. Then let it take care of all the synchronisation issues internally.

My feeling is still that part of the reason we’ve hit so many bugs in the 7043 (which is a widely used chip) is that we’re using it in a way that wasn’t intended or properly tested. Going further down the path of reinventing the wheel doesn’t feel like it’s going to end well.

Conceptually what you’re saying sounds right, but pratcically I think we’ll be in much better shape if we do what everyone else does to solve this problem...

sbourdeauducq, 2018-08-21:
> Then let it take care of all the synchronisation issues internally.

Hmm, how does it know the proper skew to apply to SYSREF so it meets s/h at the DACs?

How do we know if the RFSYNCIN pulse meets s/h at the 7044?

Why is it "NIH" to scan a delay that drives the DAC, and not NIH to scan a delay that drives the 7044?

sbourdeauducq, 2018-08-21:
The only drawback that I see with the FPGA technique is PVT skew/jitter issues that can become problematic at high sample rates and may require the use of that trashy "native" I/O block to mitigate. Not "NIH".

sbourdeauducq, 2018-08-21:
> The 7044 does take care of all of that with just a simple trigger that has to meet s/h on the 100MHz clock, which is easy.

Even if the window is larger, how do you know that you actually hit it? It's not really "simple" and "easy".

hartytp, 2018-08-21:
Let’s discuss this more next week when I’m back in the office, but I’m confident we can find a good solution with the 7044 if we want to. I can also use an eval board to test as much of this as possible to make sure it will work properly.

I don’t think that hitting the (>5ns IIRC) S/H window for the 7044 will be a problem — If we need to we can route the HMC830 clock to the AMC and then clock the IOB FF from that clock. Then we can just specify an appropriate timing constraint on the SYSREF signal to ensure that S/H is met. We can then check this easily with a scope. We can even do something like add an extra FF on the RTM, whose output is looped back to the AMC to allow us to verify the timing window...

sbourdeauducq, 2018-08-21:
Sure, but what should the timing relationship be between RFSYNCIN and the input clock at the 7044 pins? Do you trust what HMC tells you, with their datasheets that show AC-coupled RFSYNCIN and tell you to read registers that return nonsense such as 0x007d?
I'd rather double-check, scan the delay and look at the DAC outputs to determine that window myself, but that's not simple or easy.

sbourdeauducq, 2018-08-21:
I don't necessarily want to do it that way, as I pointed out the native Ultrascale I/O is also full of garbage (and so is the GTH transceiver which would be a third solution for driving SYSREF) - I'm just pointing out that the 7044 should be treated with caution. Basically there is no good option unless silicon vendors get their act together.

sbourdeauducq, 2018-08-21:
Also, SYSREF has to be aligned to the RTIO clock, which is not 100MHz.

marmeladapk, 2018-08-22:
[image]
[image]

Measured on R63 and R68 (one is _P one is _N of diff pair, that's why they're inverted).

marmeladapk, 2018-08-22:
Right, I reenabled sysref_auto_rtio_align and margins are proper everytime now, when I'm trying to catch bad DAC sync. Sigh...

marmeladapk, 2018-08-22:
Another board:

Calibration:

```
[    13.782813s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...
[    13.809001s]  INFO(board_artiq::jesd204sync):   ...done (64/64 slips)
[    13.817360s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -18 +16
[    13.822923s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase at DAC-0...
[    13.931766s]  INFO(board_artiq::jesd204sync):   ...done, min=27, max=95, result=61
[    13.942436s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0...
[    14.039058s]  INFO(board_artiq::jesd204sync):   margins: -35 +34
[    14.045004s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-1...
[    14.141649s]  INFO(board_artiq::jesd204sync):   margins: -34 +35
```

Then, after power cycle:

```
[    13.652703s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...<\n>
[    13.672069s]  INFO(board_artiq::jesd204sync):   ...done (22/64 slips)<\n>
[    13.680241s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -17 +17<\n>
[    13.685784s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0...<\n>
[    13.783335s]  INFO(board_artiq::jesd204sync):   margins: -1 +68<\n>
[    13.787956s] ERROR(runtime): failed to align SYSREF at DAC: SYSREF margins at DAC are too small, board needs recalibration
```

[image]
[image]

sbourdeauducq, 2018-08-22:
The SYSREF offset after a failed margin test is normal; the only visible anomaly here is the test failure.

sbourdeauducq, 2018-08-22:
Maybe the code should keep going after a failed test and program SYSREF delays, in order to expose any 7043 shenanigans...

sbourdeauducq, 2018-08-25:
Or more simply, the margin scans can be disabled (if that doesn't make the bug disappear).

sbourdeauducq, 2018-08-26:
@marmeladapk 

0. Calibrate the board once and keep the same calibration values for all further tests (and do not recompile the bitstream).
1. Is DAC-to-DAC sync truly working reliably on your board as well with RTIO/SYSREF alignment disabled?
2. Do bugs appear on your boad as well when RTIO/SYSREF alignment is re-enabled?
3. Simplify the code by disabling the margin scans (patch below). Recompile and reflash the runtime __*only*__. Verify points 1 and 2 again to make sure no shenanigans are going on. What do SYSREF signals look like when DAC synchronization breaks?
```
diff --git a/artiq/firmware/libboard_artiq/jesd204sync.rs b/artiq/firmware/libboard_artiq/jesd204sync.rs
index c4cc1881..9ec5ceb4 100644
--- a/artiq/firmware/libboard_artiq/jesd204sync.rs
+++ b/artiq/firmware/libboard_artiq/jesd204sync.rs
@@ -97,7 +97,7 @@ fn sysref_rtio_align(phase_offset: u16) -> Result<(), &'static str> {
     }
     info!("  ...done ({}/{} slips)", slips0, slips1);
 
-    let mut margin_minus = None;
+    /*let mut margin_minus = None;
     for d in 0..phase_offset {
         hmc7043::sysref_offset_fpga(phase_offset - d);
         if !sysref_sample() {
@@ -119,7 +119,7 @@ fn sysref_rtio_align(phase_offset: u16) -> Result<(), &'static str> {
         }
     } else {
         return Err("unable to determine SYSREF margin at FPGA");
-    }
+    }*/
 
     Ok(())
 }
@@ -187,7 +187,7 @@ fn sysref_cal_dac(dacno: u8) -> Result<u16, &'static str> {
 }
 
 fn sysref_dac_align(dacno: u8, phase: u16) -> Result<(), &'static str> {
-    let mut margin_minus = None;
+    /*let mut margin_minus = None;
     let mut margin_plus = None;
 
     info!("verifying SYSREF margins at DAC-{}...", dacno);
@@ -223,7 +223,7 @@ fn sysref_dac_align(dacno: u8, phase: u16) -> Result<(), &'static str> {
         }
     } else {
         return Err("Unable to determine SYSREF margins at DAC");
-    }
+    }*/
 
     // Put SYSREF at the correct phase and sync DAC
     hmc7043::sysref_offset_dac(dacno, phase);
```

marmeladapk, 2018-08-28:
@sbourdeauducq 

1. Yes*

\* If I do power cycle then yes. If I do `artiq_flash -t sayma start` then no.

2. The same as in 1.
```
[    16.033970s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase offset at FPGA...
[    16.070866s]  INFO(board_artiq::jesd204sync):   ...done, phase offset: 130
[    16.081699s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...
[    16.107721s]  INFO(board_artiq::jesd204sync):   ...done (64/64 slips)
[    16.115865s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -17 +17
[    16.121484s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0...
[    16.218943s]  INFO(board_artiq::jesd204sync):   margins: -35 +34
[    16.224867s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-1...
[    16.322552s]  INFO(board_artiq::jesd204sync):   margins: -36 +34
```
This is the worst case I caught when alignment worked.
[image]

Usually I get something like this when alignment works:
```
[    16.044651s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...
[    16.066540s]  INFO(board_artiq::jesd204sync):   ...done (38/64 slips)
[    16.074526s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -16 +18
[    16.080083s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0...
[    16.177609s]  INFO(board_artiq::jesd204sync):   margins: -35 +34
[    16.183532s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-1...
[    16.279981s]  INFO(board_artiq::jesd204sync):   margins: -35 +34
```
[image]

> What do SYSREF signals look like when DAC synchronization breaks?

Broken:
[image]

I didn't have time for 3. yet. So far I tried few times with your patch and sysref alignment enabled, but nothing conclusive. Sometimes signals where out of phase in a similar way that is shown in the first picture.

sbourdeauducq, 2018-08-28:
> Yes*
> 
> * If I do power cycle then yes. If I do artiq_flash -t sayma start then no.

Interesting. My board behaves differently. Can you check that the HMC7043 receives a reset pulse at some point after doing ``artiq_flash -t sayma start``?

>     The same as in 1.

What I meant by "DAC to DAC sync" in all the questions above refers to what happens at the DAC outputs, not what is in the log. Did you look at the DAC outputs?

> This is the worst case I caught when alignment worked.

What does that mean?

sbourdeauducq, 2018-09-19:
@marmeladapk From your comment https://github.com/sinara-hw/sinara/issues/557#issuecomment-422415161

do I understand correctly that you do not have a single board on which the HMC7043 garbage behaves itself even with RTIO/SYSREF alignment disabled, and so you are unable to debug this?

gkasprow, 2018-09-19:
one working board was shipped to you, another to Tom.
We have 2 sets that sometimes do not align.

hartytp, 2018-09-19:
@gkasprow well, since we are most likely going to scrap the HMC7043 in the next revision, is it worth tracking this down further? I'm implementing the new synchronization scheme right now...

sbourdeauducq, 2018-10-09:
@jbqubit ping

closed 2018-12-26 by hartytp