m-labs/artiq #1133: Phase difference between DACs that varies when changing frequency

marmeladapk, 2018-08-14:
```
[    11.066016s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[    11.077903s]  INFO(board_artiq::ad9154):   ...done
[    11.151790s]  INFO(board_artiq::ad9154): AD9154-0 running PRBS test...
[    12.158169s]  INFO(board_artiq::ad9154):   ...passed
[    12.161832s]  INFO(board_artiq::ad9154): AD9154-0 running STPL test...
[    12.168620s]  INFO(board_artiq::ad9154):   c0 errors: 0
[    12.173825s]  INFO(board_artiq::ad9154):   c1 errors: 0
[    12.179035s]  INFO(board_artiq::ad9154):   c2 errors: 0
[    12.184244s]  INFO(board_artiq::ad9154):   c3 errors: 0
[    12.189168s]  INFO(board_artiq::ad9154):   ...passed
[    12.204409s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[    12.216256s]  INFO(board_artiq::ad9154):   ...done
[    12.300760s]  INFO(board_artiq::ad9154): AD9154-1 initializing...
[    12.312499s]  INFO(board_artiq::ad9154):   ...done
[    12.386377s]  INFO(board_artiq::ad9154): AD9154-1 running PRBS test...
[    13.392746s]  INFO(board_artiq::ad9154):   ...passed
[    13.396407s]  INFO(board_artiq::ad9154): AD9154-1 running STPL test...
[    13.403193s]  INFO(board_artiq::ad9154):   c0 errors: 0
[    13.408402s]  INFO(board_artiq::ad9154):   c1 errors: 0
[    13.413611s]  INFO(board_artiq::ad9154):   c2 errors: 0
[    13.418821s]  INFO(board_artiq::ad9154):   c3 errors: 0
[    13.423743s]  INFO(board_artiq::ad9154):   ...passed
[    13.438986s]  INFO(board_artiq::ad9154): AD9154-1 initializing...
[    13.450724s]  INFO(board_artiq::ad9154):   ...done
[    13.524662s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase offset at FPGA...
[    13.542312s]  INFO(board_artiq::jesd204sync):   ...done, phase offset: 132
[    13.552550s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...
[    13.578576s]  INFO(board_artiq::jesd204sync):   ...done (64/64 slips)
[    13.586748s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -17 +17
[    13.592230s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase at DAC-0...
[    13.741603s] ERROR(runtime): failed to align SYSREF at DAC: no sync errors found when scanning delay
[    13.749450s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[    13.757967s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[    13.765210s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[    13.772454s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[    13.779697s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[    13.786940s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[    13.794183s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[    13.801426s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[    13.808735s]  INFO(runtime): using MAC address 34-45-32-12-ff-22
[    13.813545s]  INFO(runtime): using IP address 192.168.95.180
[    13.820672s]  INFO(runtime::mgmt): management interface active
[    13.834159s]  INFO(runtime::session): accepting network sessions
[    13.848528s]  INFO(runtime::session): running startup kernel
[    13.853040s]  INFO(runtime::session): no startup kernel found
[    13.858737s]  INFO(runtime::session): no connection, starting idle kernel
[    13.865653s]  INFO(runtime::session): no idle kernel found
[   331.919712s]  INFO(runtime::session): new connection from 192.168.95.122:43822
[   331.958462s]  INFO(runtime::kern_hwreq): resetting RTIO
[   331.965747s]  INFO(runtime::session): no connection, starting idle kernel
[   331.971716s]  INFO(runtime::session): no idle kernel found
[   456.236325s]  INFO(runtime::session): new connection from 192.168.95.122:43824
[   456.274929s]  INFO(runtime::kern_hwreq): resetting RTIO
[   456.282365s]  INFO(runtime::session): no connection, starting idle kernel
[   456.288103s]  INFO(runtime::session): no idle kernel found
[   556.754603s]  INFO(runtime::session): new connection from 192.168.95.122:43826
[   556.793305s]  INFO(runtime::kern_hwreq): resetting RTIO
[   556.800748s]  INFO(runtime::session): no connection, starting idle kernel
[   556.806490s]  INFO(runtime::session): no idle kernel found

```

Kernel run with artiq_run:

```
from artiq.experiment import *
class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        for sawg in self.sawgs:
            delay(2*ms)
            sawg.amplitude1.set(.4)
            # Do not use a sub-multiple of oscilloscope sample rates.
            sawg.frequency0.set(7*MHz)
```

On next images sawg.frequency was set to 4 MHz, 5,5MHz and 3 MHz. Two signals are from different DACs, cables are matched. I tried changing delay, but it didn't affect the phase.

[image]
[image]

[image]

hartytp, 2018-08-14:
@marmeladapk Isn't this expected, since you don't update all the frequencies simultaneously?

marmeladapk, 2018-08-14:
@hartytp Changing delay doesn't affect phase difference (I measured phase, changed delay and frequency, changed frequency once again and measured phase to make sure that frequency was changed).

hartytp, 2018-08-14:
> I tried changing delay, but it didn't affect the phase.

Sorry, I missed that.

hartytp, 2018-08-14:
> I tried changing delay, but it didn't affect the phase.

But, it should do, right?

> (I measured phase, changed delay and frequency, changed frequency once again and measured phase to make sure that frequency was changed)

So, your measurement sequence was:
1. run the above experiment with frequency0 set to `4*MHz` and measure the phase difference between channels
2. change both the delay and frequency, and repeat the above measurement
3. same as (1) but with only the frequency changed

and you found that (2) and (3) gave the same results. Is that correct?

marmeladapk, 2018-08-14:
@hartytp (1) and (3) gave the same results, in (2) there was different phase.

hartytp, 2018-08-14:
what frequency and delay did you use in each measurement?

jordens, 2018-08-14:
@marmeladapk I don't see the correlation between your description and the screenshots. The phases in the screenshots are all different (see the measurements window). If you look at 
https://m-labs.hk/artiq/manual-master/core_drivers_reference.html#artiq.coredevice.sawg.Config.set_clr
and the reset condition, does that clear up things?

marmeladapk, 2018-08-14:
@hartytp I didn't write them down, sorry. 

@jordens Screenshots show three different frequencies with different phases. I don't have screenshots from (1) to (3) procedure.

> set_clr

But I'm fairly sure, that phase is different is only between DACs. Signals on same DAC are in sync with each other (modulo 180 deg shift caused by swaped _P and _N pairs). I'll make more measurements on Thursday.

jordens, 2018-08-14:
I still don't understand your interpretation. What are you doing, what do you expect to measure, and what do you measure instead?

sbourdeauducq, 2018-08-15:
Note that the DACs are not properly synchronized with SC1:

> [ 13.741603s] ERROR(runtime): failed to align SYSREF at DAC: no sync errors found when scanning delay

@marmeladapk To work around this cool new Sayma-Bug, can you try to set ``sysref_phase_dac`` manually in the flash configuration? The value that works on my board is 61, then you can check in the log if the margins are ok. Maybe this problem is also explained by SYSREF generation issues and I encourage you to look at it on the scope...

jordens, 2018-08-15:
@marmeladapk referring to your experiment above and assuming you are running this on a fresh boot (with the default values of accumulator auto clear), the phase of SAWG channel `n` should be `2*pi*f*(t + n*dt)` up to a common constant. With `f=7*MHz` and `dt=2*ms` and for all the frequency values you give above that phase is zero modulo `2*pi`. I.e. you for those values you should not expect an offset (apart from the non-SC1 offset between DACs that @sbourdeauducq correctly points out). Just convince yourself by actually computing the phase offset you expect.

marmeladapk, 2018-08-22:
I think I can close it until I find a case where software reports alignment and it doesn't match measurements.

closed 2018-08-22 by marmeladapk