m-labs/artiq #1135: Text missing on right click context menu in linux

prasath-d, 2018-08-17:
I have tried to run dashboard from Ubuntu 18.04 and noticed that the right-click context menu is not visible. for example 
I tried to plot the datasets, but the applet menu that shows templates etc was missing.

similarly, all the right-click menu fields are empty.
is there any quick fix for that?

sbourdeauducq, 2018-08-18:
Can you post screenshots?

prasath-d, 2018-08-18:
[image]

prasath-d, 2018-08-18:
I found a fix.

`export QT_STYLE_OVERRIDE=gtk2`

Adding these lines to .bashrc fixes the problem.

sbourdeauducq, 2018-08-18:
Where did you find this? Looks like a Qt or Ubuntu bug?

prasath-d, 2018-08-18:
[](https://wiki.archlinux.org/index.php/Uniform_look_for_Qt_and_GTK_applications
)
[](https://unix.stackexchange.com/questions/176696/qt5-applications-dont-use-gtk-style-on-liux-mint
)
These two links helped me.

sbourdeauducq, 2018-08-18:
OK, well this doesn't look like a problem with ARTIQ. Good you found a workaround.

closed 2018-08-18 by sbourdeauducq