m-labs/artiq #1138: Conda >= 4.3 Support requires "noarch/repodata.json"

drewrisinger, 2018-08-20:
When installing Artiq with Conda >= 4.3, conda throws error about missing "noarch/repodata.json" file.

## Error Message (shown in Conda 4.5.10)
```
CondaHTTPError: HTTP 404 NOT FOUND for url <https://conda.anaconda.org/mlabs/noarch/repodata.json>
Elapsed: 00:00.125801
CF-RAY: 44d83ce0ced89592-IAD

The remote server could not find the noarch directory for the
requested channel with url: https://conda.anaconda.org/mlabs

As of conda 4.3, a valid channel must contain a `noarch/repodata.json` and
associated `noarch/repodata.json.bz2` file, even if `noarch/repodata.json` is
empty. please request that the channel administrator create
`noarch/repodata.json` and associated `noarch/repodata.json.bz2` files.
$ mkdir noarch
$ echo '{}' > noarch/repodata.json
$ bzip2 -k noarch/repodata.json

You will need to adjust your conda configuration to proceed.
Use `conda config --show channels` to view your configuration's current state.
Further configuration help can be found at <https://conda.io/docs/config.html>.
```

## Remedy
M-Lab Conda channel admin must run following commands in the conda directory (on the buildbot directory, probably), as shown above.
```
$ mkdir noarch
$ echo '{}' > noarch/repodata.json
$ bzip2 -k noarch/repodata.json
```

sbourdeauducq, 2018-08-21:
> conda create -n "any_env_name" -c mlabs artiq

> The remote server could not find the noarch directory for the requested channel with url: https://conda.anaconda.org/mlabs

It's m-labs / https://conda.anaconda.org/m-labs and not mlabs / https://conda.anaconda.org/mlabs. Could that be the actual problem?

https://conda.anaconda.org/m-labs/noarch/repodata.json exists.

sbourdeauducq, 2018-08-26:
@drewrisinger ping

drewrisinger, 2018-08-29:
Confirmed. Misspelled `m-labs` as `mlabs`. Fixed the issue.

closed 2018-08-29 by drewrisinger

xuChenSJTU, 2024-07-10:
have the same issue, anyone can help?