m-labs/artiq #1139: remove device_db alias support

sbourdeauducq, 2018-08-28:
http://m-labs.hk/artiq/manual-master/environment.html?highlight=aliases#aliases

Aliases add complexity and bugs. Does anyone use them anyway?

The CI system uses them but could do without.

Note: Replacing them with multiple copies of the same description should not be done because:
* multiple moninj widgets would appear
* multiple copies of the same device driver could be created by requesting the different names, and they would conflict
* this cannot be solved with a simple description → existing driver dictionary, as the description contains some elements (e.g. the comment field) that do not affect the device driver.

jordens, 2018-08-28:
They are used quite a bit. What complexity and bugs are you referring to?

cjbe, 2018-08-28:
@sbourdeauducq we also find aliases handy, and use them quite a lot.

k-beck, 2018-08-28:
I agree with @cjbe -- we use aliases within the device database to abstract physical channels (ex: digital channel 3) with their functional names (ex: line trigger input), while maintaining the ability to use either.

sbourdeauducq, 2018-08-29:
> What complexity and bugs are you referring to?

* ``artiq_coremgmt`` breaks if ``core`` is an alias
* requesting a device and its alias creates two copies of the same driver
* probably others, this is only what I found when looking briefly at that code again

Nothing unfixable, but generally the alias special-case is something that is easily forgotten during development and poorly tested. If it weren't used, it would have been better to just scrap that feature.

jordens, 2018-08-29:
ACK. Let's file a new issue for those.

closed 2018-08-29 by jordens