m-labs/artiq #1142: Urukul monitoring/injection

jordens, 2018-08-30:
* monitoring is doable (tricky corners are the hidden and mutable state of the ref clock frequency, pll multiplier etc)
* injection is more difficult due to the shared resources (SPI bus and shared IO_UPDATE), exact requirements and options need to be discussed

sbourdeauducq, 2018-11-21:
One option for DDS "injection": the dashboard submits an experiment into the scheduler that simply sets the frequency.
That would require implementing support for submission-by-content to the master; this feature would play nice with a "sketchpad" feature that lets you write and test quick-and-dirty experiment code in the dashboard.

dhslichter, 2018-11-26:
Honestly I am not sure how necessary this is as an ARTIQ feature; we have already written simple "set DDS"-type experiments that achieve the desired injection, and I think that this is a cleaner way of doing it than having implicit experiment submissions coming from the DDS panel, especially given the potential for latency (e.g. if another experiment is already running or in the queue, if priorities are not properly set, etc.) and the need to hit a "submit new frequency" button anyway so that you're not firing off a new experiment each time a digit is changed.

sbourdeauducq, 2018-11-27:
It's just a nicer UI.
The experiment submissions are not so "implicit", they would appear in the scheduler panel, and pipeline/priority/etc. could be set in the options of the DDS panel. It would become pretty clear to the user what is happening.
There is no need for a "submit new frequency" button with this design: double click on the frequency to pop up a small text editor, then press Enter after editing the frequency to close the pop-up editor and submit the experiment.

klickverbot, 2018-11-27:
In terms of injection, what would be higher priority in my opinion than a nice UI would be working override functionality similar to what we have for TTLs, where one can pin the channel to a fixed frequency/phase setting regardless of what the experiment is doing. (This is motivated by diagnosing issues with both DDS hardware and the experiment attached to it over the last few months.)

There are a few tricks one could play in that regard, but I don't have a good design for this yet which doesn't involve shifting management of a considerable portion of state into the FPGA.

dhslichter, 2018-11-27:
@klickverbot yes this would be nice, but definitely requires more mucking around with the guts on the FPGA in a serious way.  @sbourdeauducq is there a reasonable way to maintain the DDS state in gateware or is this a nightmare?

sbourdeauducq, 2018-11-28:
Supporting injection everywhere is quite complicated with all the details and different ways to set frequency (different DDS chips, SPI/parallel, su-servo, upcoming AD9910 RAM mode, AD9914 fractional FTW, etc)

dhslichter, 2018-11-29:
But can one wrap the physical layer specifications for each in some sensible way, and just make a DDS state machine that will be suitably track a simplified set of parameters (frequency with 32-bit FTW only, amplitude scale factor) and enable those to be overridden by injection?  AFAICT the only thing one would want to do with injection would be to set frequency or amplitude to a fixed value, manually.  Setting phase is irrelevant because the timing is not deterministic, and finer resolution on a FTW (or RAM mode for an AD9910) are sufficiently unusual cases that I think users can/should write their own code for testing/debugging those things.  

Basically if every DDS had a little frequency/amplitude state machine that was updated every time the DDS is programmed, and could be injected to that DDS (with a flag then set such that all commands to that DDS are ignored while the injection is on), would that run afoul of these various corner cases?  There would have to be a gateware variant for each type of DDS (serial/parallel, different chip, etc) but that is already happening, right?

sbourdeauducq, 2021-09-12:
Funded by Duke

sbourdeauducq, 2021-09-13:
WIP implementation by @b-bondurant - maybe some code can be reused from there:
https://github.com/b-bondurant/artiq/commit/5e03a102ca15e7ef54c73a412cbce432aa76ef61

PR #1921 (2022-07-07): Urukul monitoring
# ARTIQ Pull Request

## Description of Changes

Added support for Urukul monitoring in the dashboard.

Code mostly lifted from #1850 - with few changes, mostly on the dashboard side not to put everything together.

Tested with AD9910 and AD9912. 

[image]

Since it monitors what is sent to the DDS themselves but has no control over the CPLD or rf_sw, it won't know if the channel is turned off or on, however. But the frequencies will be displayed.

### Related Issue

Closes #1142

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :sparkles: New feature [... cut]
closed 2022-07-07 by sbourdeauducq