m-labs/artiq #1143: Urukul: Kasli-Urukul  synchronization using SYNC IN driven by Kasli

jordens, 2018-08-30:
* the is some proof of principle code in https://github.com/cjbe/artiq/ for driving SYNC from Kasli
* delay calibration in the DDS
* IO_UPDATE alignment calibration (cabling dependent)
* calibration persistence (EEPROM or elsewhere)
* absolute phase, coherent phase modes (like AD9914)

klickverbot, 2018-09-17:
We have code for this in production use (minus automatic phase tracking); I'll try and push this somewhere once I have a chance to catch my breath. Anyone interested in this, please feel free to contact me in the meantime.

closed 2018-11-05 by jordens

jordens, 2018-11-13:
Some example code to show the functionality:

```python
from artiq.language import *

from artiq.coredevice.ad9910 import PHASE_MODE_TRACKING

class UrukulSync(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.d0 = self.get_device("urukul0_ch0")
        self.d1 = self.get_device("urukul0_ch1")
        self.d2 = self.get_device("urukul0_ch2")
        self.d3 = self.get_device("urukul0_ch3")
        self.t = self.get_device("ttl4")

    @kernel
    def run(self):
        self.core.break_realtime()
        self.d0.cpld.init()
        self.d0.init()
        self.d1.init()
        self.d2.init()
        self.d3.init()

        # This calibration needs to be done only once to find good values.
        # The rest is happening at each future init() of the DDS.
        if self.d0.sync_delay_seed == -1:
            delay(100*us)
            d0, w0 = self.d0.tune_sync_delay()
            t0 = self.d0.tune_io_update_delay()
            d1, w1 = self.d1.tune_sync_delay()
            t1 = self.d0.tune_io_update_delay()
            d2, w2 = self.d2.tune_sync_delay()
            t2 = self.d0.tune_io_update_delay()
            d3, w3 = self.d3.tune_sync_delay()
            t3 = self.d0.tune_io_update_delay()
            # Add the values found to each of the four channels in your
            # device_db.py so that e.g. for urukul0_ch0 it looks like:
            #    "urukul0_ch0": {
            #       ...
            #        "class": "AD9910",
            #        "arguments": {
            #            "pll_n": 32,
            #            "chip_select": 4,
            #           "sync_delay_seed": D,
            #           "io_update_delay": T,
            #           "cpld_device": "urukul0_cpld",
            #           ...
            #       }
            # where T is the io_update_delay of the channel and
            # D is the sync_delay_seed of the channel below:
            print("sync_delay_seed", [d0, d1, d2, d3])
            print("io_update_delay", [t0, t1, t2, t3])
            # As long as the values don't differ too much between the channels,
            # using the mean for them is also fine.
            # This one is for information purposes only:
            # print("validation delays", [w0, w1, w2, w3])
            #
            # then run this script again
            return

        self.d0.set_phase_mode(PHASE_MODE_TRACKING)
        self.d1.set_phase_mode(PHASE_MODE_TRACKING)
        self.d2.set_phase_mode(PHASE_MODE_TRACKING)
        self.d3.set_phase_mode(PHASE_MODE_TRACKING)

        self.d0.set_att(10*dB)
        self.d1.set_att(10*dB)
        self.d2.set_att(10*dB)
        self.d3.set_att(10*dB)

        t = now_mu()
        self.d0.set(80*MHz, phase=0., ref_time=t)
        self.d1.set(80*MHz, phase=.5, ref_time=t)
        self.d2.set(80*MHz, phase=.25, ref_time=t)
        self.d3.set(80*MHz, phase=.75, ref_time=t)

        self.t.on()
        self.d0.sw.on()
        self.d1.sw.on()
        self.d2.sw.on()
        self.d3.sw.on()
```

Or:

```python
    @kernel
    def multi(self):
        """more complicated phase gymnastics"""
        self.core.break_realtime()

        self.d0.cpld.init()
        self.d0.init()
        self.d1.init()

        delay(1*ms)

        self.d0.set_att(10.)
        self.d1.set_att(10.)

        dt = 1*us
        f = 1*MHz

        t0 = now_mu()
        self.d0.set(f, phase=0., ref_time=t0, phase_mode=1)
        self.d1.set(f, phase=0., ref_time=t0, phase_mode=1)
        delay(dt)
        self.d0.sw.on()
        self.d1.sw.on()
        delay(1.18*us)

        self.d0.sw.off()
        self.d1.sw.off()
        delay(dt)
        self.d0.sw.on()
        self.d1.sw.on()

        p0 = self.d1.set(f, phase=.5, ref_time=t0, phase_mode=1)
        delay(dt)

        self.d1.set(.6*f, phase=p0)
        delay(dt)

        self.d1.set(f, ref_time=t0, phase_mode=1)
        delay(dt)

        self.d0.sw.off()
        self.d1.sw.off()
```

PR #1190 (2018-11-05): Urukul sync
# ARTIQ Pull Request

## Description of Changes

### Related Issue

Closes #1143 
See roadmap and milestones there.

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |
| ✓  | :sparkles: New feature |
| ✓  | :hammer: Refactoring  |
| ✓  | :scroll: Docs |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELEASE_NOTES.md](../RELEASE_NOTES.md) if there are noteworthy changes, especially if there are changes to existing APIs: nothing w.r.t. release-3
- [x [... cut]