m-labs/artiq #1144: Sayma: SAWG at 1GBPS

hartytp, 2018-08-31:
I had a go at building Sayma with the SAWG running at 1GBPs line rate using https://github.com/hartytp/artiq/tree/fast_sawg to get a better idea of resource utilization with the current design

I believe I've made all the gateware adjustments required to do this, but I haven't touched the firmware yet. I don't have hardware atm so can't test in any case.

Build files are here: https://drive.google.com/open?id=1TjC1ldEbLZj6Le0T68QqvPX6pN1Rxy3y

Design meets timing.

Resource utilization:
```
Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.
------------------------------------------------------------------------------------
| Tool Version : Vivado v.2018.1 (lin64) Build 2188600 Wed Apr  4 18:39:19 MDT 2018
| Date         : Thu Aug 30 22:03:30 2018
| Host         : buildsvr running 64-bit Ubuntu 17.10
| Command      : report_utilization -file top_utilization_place.rpt
| Design       : top
| Device       : xcku040ffva1156-1
| Design State : Fully Placed
------------------------------------------------------------------------------------

Utilization Design Information

Table of Contents
-----------------
1. CLB Logic
1.1 Summary of Registers by Type
2. CLB Logic Distribution
3. BLOCKRAM
4. ARITHMETIC
5. I/O
6. CLOCK
7. ADVANCED
8. CONFIGURATION
9. Primitives
10. Black Boxes
11. Instantiated Netlists

1. CLB Logic
------------

+----------------------------+--------+-------+-----------+-------+
|          Site Type         |  Used  | Fixed | Available | Util% |
+----------------------------+--------+-------+-----------+-------+
| CLB LUTs                   | 137678 |     0 |    242400 | 56.80 |
|   LUT as Logic             | 136670 |     0 |    242400 | 56.38 |
|   LUT as Memory            |   1008 |     0 |    112800 |  0.89 |
|     LUT as Distributed RAM |    776 |     0 |           |       |
|     LUT as Shift Register  |    232 |     0 |           |       |
| CLB Registers              | 107787 |     0 |    484800 | 22.23 |
|   Register as Flip Flop    | 107787 |     0 |    484800 | 22.23 |
|   Register as Latch        |      0 |     0 |    484800 |  0.00 |
| CARRY8                     |  10424 |     0 |     30300 | 34.40 |
| F7 Muxes                   |    651 |     0 |    121200 |  0.54 |
| F8 Muxes                   |     65 |     0 |     60600 |  0.11 |
| F9 Muxes                   |      0 |     0 |     30300 |  0.00 |
+----------------------------+--------+-------+-----------+-------+

1.1 Summary of Registers by Type
--------------------------------

+--------+--------------+-------------+--------------+
|  Total | Clock Enable | Synchronous | Asynchronous |
+--------+--------------+-------------+--------------+
| 0      |            _ |           - |            - |
| 0      |            _ |           - |          Set |
| 0      |            _ |           - |        Reset |
| 0      |            _ |         Set |            - |
| 0      |            _ |       Reset |            - |
| 0      |          Yes |           - |            - |
[... 175 lines cut ...]
| MMCME3_ADV    |      3 |               Clock |
| IBUFDS_GTE3   |      2 |            Advanced |
| STARTUPE3     |      1 |       Configuration |
| BUFGCE_DIV    |      1 |               Clock |
+---------------+--------+---------------------+

10. Black Boxes
---------------

+----------+------+
| Ref Name | Used |
+----------+------+

11. Instantiated Netlists
-------------------------

+----------+------+
| Ref Name | Used |
+----------+------+
```

For comparison, here is a build with the current ARTIQ master (600MSPs DAC data rate):
```
still building...
```

hartytp, 2018-08-31:
So, unless I've completely messed that up, the 1GSPs SAWG doesn't seem that much worse than the 600MSPs...

bradbqc, 2018-10-25:
What's your hardware setup? I see some references in sayma_rtm.py to clk_mez, but we have no clock mezzanine board here. Is it possible to do this without it? I built and flashed the current revision of your code (without any modifications, despite my lack of mezzanine, just to see what would happen), and get the following output on startup: 
https://paste.ofcode.org/vdwbAkhgiTR5pDV7R2aT3B

hartytp, 2018-10-25:
Hi @bradbqc. Short answer, this branch never worked and never will.

I believe that I configured the SAWG correctly for the higher data rate (although I never tested on hw so I might be wrong about that) and the resource utilisation wasn't an issue. However, the JESD204b core can't be used for the higher line rate without modification. That's not something I'm going to have time to do, so I have no plans to take this further (there was a discussion about this on IIRC some time back, but I never got round to updating the issue).

The references to the clock mezzanine are because this branch morphed into my synchronisation tests at some point (I was using the clock mezzanine IO to control a delay line). Basically, poor git hygiene on my part.

bradbqc, 2018-10-26:
Ah, I see. Well, thank you for trying!

jordens, 2018-10-26:
To use Sayma with 1 GHz data rate, the parameters of SAWG, the JESD core as well as the clocking setup and DAC configuration code need to be adjusted. This is likely achievable as most of the parts are designed for and were individually already tested at 1 GHz. But it's not straight-forward. Based on input and requirements from the funding groups and users, with the robustness of the system in mind, and considering the pros and cons and the risk involved, the consensus was 600 MHz.

bradbqc, 2018-10-26:
Seems reasonable. Thanks for the explanation.

jordens, 2018-10-26:
Depends on what you want to do. Tell us about yourself.
There is a Sayma manual somewhere. It summarizes the information from the schematics and layout and talks a bit about the external connectivity. If you want to develop on Sayma—your original question seems to indicate that—then the schematics and layouts and the discussions in the issues and wiki pages are everything. Plus the datasheets and literature on JESD, DSP algorithms etc.
If you want to develop gateware/software then you'll have to learn python/migen/misoc and read lots of code. Development and documentation and tutorials on migen (and/or the funding of that) would be very welcome.

sbourdeauducq, 2020-08-29:
@jbqubit @hartytp @jordens Is there still interest in the old 600MHz clocking scheme, or can we remove it eventually?

hartytp, 2020-08-29:
I've no interest in it.

jordens, 2020-08-29:
Doesn't matter to me.

sbourdeauducq, 2020-08-29:
Other than that, clocking and DDMTD appear to work with 1GSPS, but I'm having problems with litejesd204...

sbourdeauducq, 2020-08-29:
Also SAWG timing is still met.

jbqubit, 2020-08-30:
I don't have interest in maintaining support for 600 MSPS if 1000 MSPS works out.

sbourdeauducq, 2022-01-04:
Done

closed 2022-01-04 by sbourdeauducq