m-labs/artiq #1147: Binaries end up in wrong directory in Windows

jonaskeller, 2018-09-04:
Using windows 7:
I've been manually fixing this for a while, it occurred for every version at least since 3.0, including 4.0.dev.
Since it is confusing to new users, I thought I should bring it up:

After installing artiq in a new environment, e.g. using
`conda create -n <envname> artiq artiq-kc705-nist_qc2 ...`,
instead of
`<anaconda dir>/envs/<envname>/lib/site_packages/artiq/binaries/...`,
the binary files end up in
`<anaconda dir>/envs/<envname>/lib/python3.5/site_packages/artiq/binaries/...`.

This makes `artiq_flash` unhappy:

	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiqdev\Scripts\artiq_flash-script.py", line 9, in <module>
		sys.exit(main())
	  File "C:\Anaconda3\envs\artiqdev\lib\site-packages\artiq\frontend\artiq_flash.py", line 333, in main
		artifact_path(variant, "gateware", "top.bit"))
	  File "C:\Anaconda3\envs\artiqdev\lib\site-packages\artiq\frontend\artiq_flash.py", line 316, in convert_gateware
		with open(bit_filename, "rb") as bit_file, \
	FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Anaconda3\\envs\\artiqdev\\lib\\site-packages\\artiq\\binaries\\kc705-nist_qc2\\top.bit'

 Manually copying them over fixes it for all 3.x versions I tried (which is most of them), but for 4.0.dev, it's still missing (at least) `bootloader.img`. If you want me to make that a separate issue, let me know.

3.5.5 is the only python version installed on the system, conda is version 4.5.11. Let me know if you need any additional information.

sbourdeauducq, 2018-09-04:
Is it actually happening with the latest .dev versions currently? I believe we fixed several conda bugs of the sort already.

jonaskeller, 2018-09-05:
Sorry, I just realized that conda seems to have defaulted to the 3.6 version of `artiq-kc705-nist_qc2`. That probably explains the missing file, and I can't say whether the problem persists for the dev version. It's definitely still an issue in 3.6 though.

It seems that the latest 4.0.dev build of `artiq-kc705-nist_qc2`, 1199+git51a5d8df, has a dependency on the `artiq` package with the exact same version number, which doesn't seem to be in the repository anymore.

Since guests apparently don't have speaking privileges in your IRC channel anymore, is there another way to trigger a new build of `artiq-kc705-nist_qc2`?

sbourdeauducq, 2018-09-05:
> Since guests apparently don't have speaking privileges in your IRC channel anymore,

You can register a freenode account, it also works with the web client.

jordens, 2018-09-05:
The requirement to register with freenode is a response to vandalism.

jonaskeller, 2018-09-05:
> You can register a freenode account, it also works with the web client.

Thanks for letting me know.

> Is it actually happening with the latest .dev versions currently? I believe we fixed several conda bugs of the sort already.

You're right, the files end up in the correct path for the dev version now that I've installed the correct version of all packages.

klickverbot, 2018-09-05:
This can be closed, then?

jonaskeller, 2018-09-05:
My problem is solved - however, this is still occurring in the current release version and potentially confusing for new users.

I don't really think it's up to me to decide whether or not that resolves this issue.

sbourdeauducq, 2018-09-06:
We're due for a new release, but this is currently blocked on #1007. Help with the latter would be much appreciated.

closed 2018-09-06 by sbourdeauducq