m-labs/artiq #1148: channel numbering for DIO_BNC EEM is wrong

sbourdeauducq, 2018-09-06:
TTL numbers are permuted within banks.

klickverbot, 2018-09-06:
Huh? With the new front panels, the indicated order should be correct (if unintuitive).

sbourdeauducq, 2018-09-06:
Then I guess I have an old front panel - its numbering is usual.
Wouldn't it be better to change the code than introduce a counter-intuitive numbering on the front panel?

klickverbot, 2018-09-06:
It's also the electrical signal ordering. But yes, the current state is not ideal, especially as that makes it easy to miss that the panel is in fact zero-indexed as IO1 is top left (a favourite gotcha for new users here). [Still on the way back to Oxford, I'd look up the current revision to confirm if I could.]

hartytp, 2018-09-06:
@klickverbot All the DIO boards we have in Oxford are from rev 1.1.

The current release (rev1.3) has channels numbered as:
0 1
2 3
4 5
6 7

Which matches other boards like Sampler.

So, I think that @sbourdeauducq is correct. However, fixing this will break the code for the v1.1 boards (which is fine, since we can fix that in our device db generation tooling).

gkasprow, 2018-09-06:
This was fixed in newest release.

closed 2018-10-09 by jordens