m-labs/artiq #1150: ConnectionResetError on TCP Keepalive

cjbe, 2018-09-13:
On certain experiments I see a significant rate of ConnectionResetErrors (up to one every 10 minutes for some use-cases).

Here is a [packet dump](https://github.com/m-labs/artiq/files/2378883/crash_3.zip) and a [trace core log](https://hastebin.com/igicojokax.php) of an typical error - the error occurs at core log timestamp [449493.322821s] == packet dump timestamp 23:01:29.962904 (packet 98). The core device is 10.255.6.210 and the master is running on 10.255.6.191. The core device connection on 1381 is idle for 3s so a keep-alive is sent. The core device responds to the keepalive, but also records a timeout and kills the connection. It looks like this is because 2.9s before the error (449490.397252) the master ARP entry (.191) is evicted from the ARP cache.

I have several more packet dumps (but without core trace logs). All the errors occur after a keepalive when the Kasli sends a RST/ACK.

I am using a Kasli with gateware and firmware build off c9d8bd15669139b97b42ee830081d2d8ca3eee02
This may be related to #1125

cjbe, 2018-09-13:
It looks like the ARP cache is statically allocated at 8 entries long - is there any reason not to allocate this on the heap?

whitequark, 2018-09-13:
No, we can easily enlarge the ARP cache by a large factor.

klickverbot, 2018-09-14:
Even with the ARP cache being limited, shouldn't the mapping just be re-learnt?

(These are the network issues I mentioned on IRC a while ago, by the way; I also saw a keep-alive/RST in all the traces I captured.)

klickverbot, 2018-09-14:
Looking at the log, that does seem to happen. However, instead of the keepalive ACK, a RST/ACK is sent, presumably due to "timeout exceeded":

```
[449492.966147s] TRACE(smoltcp::iface::neighbor): filled 10.255.6.191 => 4c-cc-6a-ca-58-ad (evicted 10.255.6.138 => 04-a3-16-b5-e6-57)
```

This is the master host.

```
[449492.966482s] DEBUG(smoltcp::socket::tcp): #0:10.255.6.210:1381:10.255.6.191:59650: received a keep-alive or window probe packet, will send an ACK
```

The master host indeed sent a keepalive – so far, so good.

```
[449492.966865s] TRACE(smoltcp::socket::meta): #0: neighbor 10.255.6.191 discovered, unsilencing
[449492.967028s] DEBUG(smoltcp::socket::tcp): #0:10.255.6.210:1381:10.255.6.191:59650: timeout exceeded
```

This, however, is not.

```
[449492.967227s] TRACE(smoltcp::socket::tcp): #0:10.255.6.210:1381:10.255.6.191:59650: state=ESTABLISHED=>CLOSED
[449492.967440s] TRACE(smoltcp::socket::tcp): #0:10.255.6.210:1381:10.255.6.191:59650: outgoing segment will send data or flags
[449492.967661s] TRACE(smoltcp::socket::tcp): #0:10.255.6.210:1381:10.255.6.191:59650: sending RST|ACK
```

This would make sense again if there indeed was a timeout.

klickverbot, 2018-09-14:
Ah, right, there is this:

```
[449490.461576s] TRACE(smoltcp::socket::meta): #0: neighbor 10.255.6.191 missing, silencing until t+3.0s
```

The timeout on the Kasli side is only 1000 ms (keepalive: 500 ms), which is shorter than the Linux defaults. Thus, by the time the master host sends the keepalive (and hence the ARP entry is added again), the timeout has already elapsed.

A workaround could be to simply increase the timeouts beyond what the OS defaults for the hosts interacting with the core device use. This is obviously not a proper solution, though.

klickverbot, 2018-09-14:
I've locally changed our Kasli masters to use a BTreeMap-backed neighbor cache (with default GC threshold of 1024), which should work around the issue by avoiding ARP table entries being evicted in the first place. If I don't succeed in provoking a crash, I'll post a PR. (It would still be good to have a more principled fix.)

sbourdeauducq, 2018-09-17:
What would such a "principled" fix involve?

klickverbot, 2018-09-17:
Figure out how the timeouts are supposed to be handled, and either fix smoltcp if it's a bug, or increase the timeout setting on the core device.

PR #1153 (2018-09-18): firmware: Use larger ARP cache
This works around a problematic interaction between ARP cache
expiry in smoltcp (with its 3 seconds timeout before a discovery
request is sent) and our TCP keepalive settings, where the timeout
is reached before the keepalive had a chance to be sent.

GitHub: Closes #1150.
commit b482f5fe (2018-09-18): firmware: Use larger ARP cache

This works around a problematic interaction between ARP cache
expiry in smoltcp (with its 3 seconds timeout before a discovery
request is sent) and our TCP keepalive settings, where the timeout
is reached before the keepalive had a chance to be sent.

GitHub: Closes #1150.
closed 2018-09-18 by whitequark