m-labs/artiq #1155: DRTIO master-satellite alignment (kasli-kasli) on serdes TTL outs

jordens, 2018-09-18:
```python
from artiq.language import *

class Hammertime(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl4")  # on master
        self.setattr_device("ttl28")  # on satellite

    @kernel
    def run(self):
        self.core.break_realtime()
        delay(10*ms)

        while True:
            self.ttl28.pulse(100*ns)
            delay((272 - 100)*ns)
            self.ttl4.pulse(100*ns)
            delay(3*us)
```

Master reboots (through `artiq_flash start`):

[image]

Satellite reboots:

[image]

c.f. #958

cjbe, 2018-09-18:
Does the si5324 output maintain phase alignment after restarting (measured via the one of the Kasli clock-out SMPs)? 

If so, it is a manifestation of #958 (incorrect resets of IOSERDES). Otherwise it is properly the phase alignment module clocking.

jordens, 2018-09-18:
The master outputs? Aligned to what?

cjbe, 2018-09-18:
The master si5324 output and the slave si5324 output should have a fixed phase relationship between resets. Equivalently non-serdes TTL outputs should have a fixed timing relationship over resets.

sbourdeauducq, 2018-09-18:
Is the problem présent at 150MHz?

jordens, 2018-09-18:
Was deterministic sync in the original code with 150 MHz not tested?

sbourdeauducq, 2018-09-18:
Chris tested it (that was during siphaser developments).

cjbe, 2018-09-18:
I tested deterministic sync on SERDES and non-SERDES outputs at 150 MHz on build shortly after #958 was fixed, and both had a deterministic time alignment (over ~100 resets).

sbourdeauducq, 2018-09-18:
CI for this shouldn't be very complicated to do (using RTIO inputs and a TTL connection between master and satellite), if someone wants to fund it.

jordens, 2018-09-18:
Same with the current, original 150 MHz `Master/Satellite` variants. There was really no reason to suspect otherwise, was there? I'm going to pause here. It's probably best if you test code yourself. @sbourdeauducq

jordens, 2018-09-18:
@cjbe I don't follow your analysis. It can't really be SiPhaser (which what I assume is what you mean by "phase alignment module clocking") since that's only in Satellite and since that synchronizes fine each time. Looking at the SI5324 alignment between the master and the satellite is not not really the best bet. I did it anyway since you and @sbourdeauducq seem to think that's the problem: but the master-satellite SI5324 phase is deterministic as expected.

@cjbe Do you have notes on what was tested back then? You said you may not have tested all possible combinations. Could you re-test it? You mentioned you have been using it this way for quite some time.

The fact that this is at t_rtio_coarse/4 and only if the master is reset to me suggests the SERDES on the master. I'll give `ttl_simple` a run. @sbourdeauducq any chance you could document the design of the clocking code and the (D)RTIO logic?

sbourdeauducq, 2018-09-18:
It's likely the SERDES then; but if not, it could be GTP shenanigans around the built-in clocking and elastic buffer (which is supposed to be disabled).

jordens, 2018-09-18:
`ttl_simple` on master against `ttl_serdes_7series` are deterministic.

jordens, 2018-09-18:
Hmm. Maybe it would be worked around by `core.reset()`. If it is, then I wonder why we can't get the master into a deterministic state without the manual reset.

cjbe, 2018-09-18:
@jordens my test involved restarting the master and slave simultaneously (via JTAG), running a startup kernel (which waits for the DRTIO link, then runs core.reset()), and running an experiment that generates simultaneous pulses on master and satellite.

The way we are using the 125 MHz DRTIO we are not sensitive to the SERDES output alignment, so I have no idea if the alignment works.

sbourdeauducq, 2018-09-19:
> If it is, then I wonder why we can't get the master into a deterministic state without the manual reset.

Because the Xilinx SERDES doesn't get into a deterministic state without an explicit reset. The only solution is to inject a reset in the firmware or somewhere else once the clocks have stabilized.

sbourdeauducq, 2018-09-19:
Hmm, the reset should already be there:
https://github.com/m-labs/artiq/blob/master/artiq/firmware/runtime/rtio_mgt.rs#L300
I don't know what the issue is then. Maybe the SERDES needs it asserted for more than one cycle?

``core.reset()`` is not supposed to help as it does not reset the ``rio_phy`` domain.

hartytp, 2018-09-25:
- using the current standalone build
- using one serdes ttl and one simple ttl, both on the core device (Kasli)
- experiment below
- looking at the time of the serdes pulse on a scope triggered from the simple TTL. The delay between the two pulses is constant to <<8ns over half a dozen power cycles of Kasli

```python
from artiq.language import *

class Hammertime(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")  # SERDES
        self.setattr_device("ttl4")  # simple

    @kernel
    def run(self):
        self.core.break_realtime()
        self.ttl0.output()
        self.ttl4.output()
        self.core.break_realtime()
        delay(10*ms)

        at_mu((now_mu() >> 3) << 3)

        while True:
            self.ttl0.pulse_mu(8*10)
            delay_mu(8*10)
            self.ttl4.pulse_mu(8*10)
            delay_mu(8*500)
```

hartytp, 2018-09-25:
Running the same thing on the master build (125MHz f_rtio) I do see variation in the latency across power cycles.

hartytp, 2018-09-25:
@sbourdeauducq maybe a silly thought, but AFAICT, the startup sequence for the master is:

- drtio::startup sets `drtio_transceiver.stable_clkin.storage`, which enables the clock input from the Si5324 and releases the rtio clock multiplier PLL (https://github.com/m-labs/artiq/blob/9584c30a1f327574fc1da1cf4032700d260cf055/artiq/gateware/targets/kasli.py#L594)
- we then immediately (few us) reset the SERDESs, without waiting for that PLL to settle

https://github.com/m-labs/artiq/blob/e4d48a78ebac60b14ec6a6f3a538e533d6ccbccb/artiq/firmware/runtime/rtio_mgt.rs#L299-L300

Do you think the issue could be related to not giving the PLL enough time to settle before resetting the SERDES PLL?

sbourdeauducq, 2018-09-25:
Quite possibly. Try adding some delay (spin_us) and see what happens, though the proper fix will be to wait for that pll to lock.

hartytp, 2018-09-25:
already building that :)

hartytp, 2018-09-25:
I need to run now, but provisionally it looks like a 1s delay there fixes this issue. I'll check more carefully later on. If it works, i'll submit a PR for a proper fix.

sbourdeauducq, 2018-09-26:
b92350b0 please test.

sbourdeauducq, 2018-10-02:
@jordens ping

jordens, 2018-10-02:
Waiting for another kasli.

jordens, 2018-10-02:
If you have two, you can test it yourself.

jordens, 2018-10-08:
Still the same behavior as of `5de319d7`.

jordens, 2018-10-08:
cherry-picked for master, tested, works now.

closed 2018-10-08 by jordens