m-labs/artiq #1160: Request: --version command-line arg for bug reporting

drewrisinger, 2018-09-26:
When bug reporting, it's useful to have an idea what ARTIQ version is causing the issue. I suggest adding a `--version` flag to `artiq_run` or `artiq_master` to print the version number. 

It could be added elsewhere, but this seems one of the easiest places.

Implementation (neglecting adding flag to argparse):
```
import artiq

print("Current ARTIQ version: {}".format(artiq.__version__))
```

sbourdeauducq, 2018-09-27:
That can be obtained with conda list or running that code in the python repl. The issue is to which tool the flag should be added. Why not dashboard, browser, or compile?

jordens, 2018-09-27:
`artiq_run` and `_master` are fine by me. Why not just add it to `verbosity_args()`? That covers a lot.

sbourdeauducq, 2018-10-22:
Maybe rename ``verbosity_args`` to e.g. ``common_args`` then?

drewrisinger, 2018-10-22:
@sbourdeauducq That would require a fair amount of refactoring, though I agree with the idea. A quick scan of the repo shows ~15 files where `verbosity_args` is used, not counting code built on top of it.

drewrisinger, 2018-10-22:
Also, processing the `--version` flag could be abstracted to some common function, maybe something like (in `artiq.tools`):

```python
def process_common_args(cmd_arguments: argparse.Namespace):
    """Process common ARTIQ arguments."""
    if cmd_arguments.version:
        # print version
    init_logger(cmd_arguments)
    if hasattr(cmd_arguments, "bind"):
        bind_address_from_args(cmd_arguments)
```

drewrisinger, 2018-10-22:
See #1181 for related PR.

drewrisinger, 2019-01-09:
Pending approval on #1181

PR #1181 (2019-01-12): [tools]: Add --version flag to cmd line args
# ARTIQ Pull Request

## Description of Changes

Rename `verbosity_args()` to `add_common_args()`. Add option for `--version`.
Use `process_common_args()` instead of `init_logger()`. Rename `init_logger()` to `_init_logger()` because deprecated.

### Related Issue

Closes #1160

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :sparkles: New feature |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELEASE_NOTES.md](../RELEASE_NOTE [... cut]
closed 2019-01-12 by sbourdeauducq