m-labs/artiq #1162: Compiler doesn't parse @PREFIX.kernel (kernel decorator with path)

drewrisinger, 2018-09-27:
# Description
ARTIQ compiler doesn't recognize `@PREFIX.kernel` decorator.

## Replicating
Instead of using `from artiq.experiments import *`, use `import artiq.language.core`. Then decorate function with `@artiq.language.core.kernel`.

### Sample Code

I would like to use following, because per [PEP20](https://www.python.org/dev/peps/pep-0020/), **Explicit is better than implicit**. Also, because it avoids cluttering the namespace and speeds imports.

Sample based on LED tutorial:
```python
import artiq.language.environment as artiq_env
import artiq.language.core as artiq_lang_core

class LED(artiq_env.EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @artiq_lang_core.kernel
    def run(self):
        self.core.reset()
        self.led.on()

```

## Behavior
### Expected Behavior
LLVM compiler should compile the function for the core device as it would any other function with `@kernel`

### Actual Behavior
ARTIQ throws CompileError (when run in separate `pytest` config), or standard error with above sample & `artiq_run`. Seems to be issue with https://github.com/m-labs/artiq/blob/0fef2afccbf7460dacac839caaab86ff376f3e03/artiq/compiler/transforms/inferencer.py#L1292-L1302

Tested also with `import artiq.language.core;  @artiq.language.core.kernel`, and produced same result.

## Log Message/Traceback

With artiq_run: `error: decorators are not supported @artiq_lang_core.kernel`.

### Full traceback
```
$ artiq_run ./led_kernel_explicit.py --vv
DEBUG:artiq.compiler.import_cache: hook installed
DEBUG:artiq.compiler.import_cache: added 'PATH/artiq/coredevice/ttl.py' to cache
./led_kernel_explicit.py:9:5-9:6: error: decorators are not supported
    @artiq_lang_core.kernel
    ^
```

jordens, 2018-09-27:
I agree. The star imports are bad. In terms of recommendations I would go for `import artiq.language as aq; ...; @aq.kernel`. Along the lines of numpy and similar.

drewrisinger, 2018-09-27:
@jordens the point is that the compiler can't recognize any sort of prefix, so even with your suggestion, the ARTIQ compiler would throw an error. However, I haven't tested it with that specific combination.

drewrisinger, 2018-09-27:
Side note: when this is fixed, should change documentation to remove all examples of `from artiq.experiment import *` because it's bad practice.

jordens, 2018-09-27:
Yes. None of this works right now because of the compiler. My point is not about the compiler but about the recommended style. I didn't want to sidetrack this issue.

whitequark, 2018-09-27:
> the point is that the compiler can't recognize any sort of prefix, so even with your suggestion, the ARTIQ compiler would throw an error

That can be fixed quite easily, I just forgot about `import as` when implementing decorator support in the compiler.

drewrisinger, 2019-01-04:
ping. out of my expertise.

jbqubit, 2019-01-08:
>  it feels better to know exactly where other code is coming from instead of just "magic."

I agree with @drewrisinger. Explicit dependencies make ARTIQ Python code easier to understand. It would be nice to support those who don't want to use ```import *```.

drewrisinger, 2019-01-09:
ping @whitequark

whitequark, 2019-01-09:
For now, import `kernel` explicitly.

sbourdeauducq, 2019-01-15:
@whitequark What is holding this?

sbourdeauducq, 2019-01-29:
@whitequark Any update?

pca006132, 2021-07-02:
A hacky solution, not sure if it works for all circumstances:

```diff
--- a/artiq/compiler/transforms/inferencer.py
+++ b/artiq/compiler/transforms/inferencer.py
@@ -6,6 +6,7 @@ from collections import OrderedDict
 from pythonparser import algorithm, diagnostic, ast
 from .. import asttyped, types, builtins
 from .typedtree_printer import TypedtreePrinter
+from artiq.experiment import kernel
 
 
 def is_nested_empty_list(node):
@@ -1662,7 +1663,13 @@ class Inferencer(algorithm.Visitor):
 
     def visit_FunctionDefT(self, node):
         for index, decorator in enumerate(node.decorator_list):
-            if types.is_builtin(decorator.type, "kernel") or \
+            def eval_attr(attr):
+                if isinstance(attr.value, asttyped.QuoteT):
+                    return getattr(attr.value.value, attr.attr)
+                return getattr(eval_attr(attr.value), attr.attr)
+            if isinstance(decorator, asttyped.AttributeT):
+                decorator = eval_attr(decorator)
+            if id(decorator) == id(kernel) or types.is_builtin(decorator.type, "kernel") or \
                     isinstance(decorator, asttyped.CallT) and \
                     types.is_builtin(decorator.func.type, "kernel"):
                 continue
```

At least works for the following cases:
```python
import artiq.experiment as exp

class Test(exp.EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @exp.kernel
    def run(self):
        pass
```

and

```python
import artiq

class Test(artiq.experiment.EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @artiq.experiment.kernel
    def run(self):
        pass
```

drewrisinger, 2021-07-02:
Thanks @pca006132, I honestly got to the point where I forgot about this b/c I was just doing ``from artiq.language.core import kernel``, which feels like a reasonable compromise.

sbourdeauducq, 2021-11-11:
Supporting special decorators not simply called as ``kernel``, ``portable``, etc. is difficult to do correctly because of https://github.com/m-labs/artiq/issues/416. We need to know which modules to cache...

PR #1710 (2021-07-02): compiler: supports kernel decorator with path
# ARTIQ Pull Request

## Description of Changes

Supports kernel decorator with path. Example:
```python
import artiq.experiment as exp

class Test(exp.EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @exp.kernel
    def run(self):
        pass
```

### Related Issue
Closes #1162.

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELEASE [... cut]