m-labs/artiq #1164: sawg: amplitude1.set()

jbqubit, 2018-09-27:
Running ```4.0.dev+1401.g20cddb6a``` on a Sayma that produces nice sinusoids if amplitude is not modified. The following produces glitchy output instead of amplitude-stepped output. 

```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawg0 = self.get_device("sawg0")

    @kernel
    def run(self):
        self.core.reset()

        delay(2*ms)
        self.sawg0.reset()
        self.sawg0.frequency0.set(100*MHz)
        delay(2*ms)   

        while True:
            delay(2*ms)
            self.ttl_sma_out.pulse(100*ns)

            self.sawg0.amplitude1.set(1.0)
            delay(100*ns)
            self.sawg0.amplitude1.set(0.5)
            delay(100*ns)
            self.sawg0.amplitude1.set(0.0)
```

[image]

sbourdeauducq, 2018-10-02:
@jordens Can you look into this?

sbourdeauducq, 2018-10-02:
@jbqubit Wrong ref_period?

jbqubit, 2018-10-02:
Agreed pulses were too short. Works fine with longer pulses.

closed 2018-10-02 by jbqubit