m-labs/artiq #1165: sayma sayma_masterdac pulse length

jbqubit, 2018-09-27:
Using 4.0.dev+1401.g20cddb6a

I'm running sayma_masterdac on internal clock with
    ```"ref_period": 1/(8*150e6)```
and see 
- ttl_sma_out.pulse(100*ns) pulse scope duration is 86.5 ns. 
- ttl_sma_out.pulse(1000*ns) pulse scope duration is 1000 ns. 

Looking at master examples/sayma_masterdac:
 ```   "ref_period": 1/(8*150e6)```

Looking at master examples/sayma_master:
```    "ref_period": 1/150e6, "ref_multiplier": 1```

What should ref_period be? Why is pulse length sometimes wrong?

jbqubit, 2018-09-28:
I'm trying to reconcile that some variants use "ref_period": 1/(8*150e6) and others use "ref_period": 1/150e6.

> Note that there is no SERDES TTL clocking on Ultrascale yet

That's news to me. Is there an Issue on this? 

> Are you sure it's 100ns->86.5ns?

Yes.

sbourdeauducq, 2018-09-28:
https://github.com/m-labs/artiq/issues/792
I don't know how it can be news after the many emails to Xilinx tech support, on which you were copied, about the mess they made of Ultrascale I/O clocking.

sbourdeauducq, 2018-09-28:
This prints the expected result 120:
```python
from artiq.experiment import *

class Blah(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        print(self.core.seconds_to_mu(100*ns))

```

What happens if you do ``pulse_mu(120)``?

closed 2022-01-04 by sbourdeauducq