m-labs/artiq #1166: sayma ramp gen issues

hartytp, 2018-10-03:
Building in a fresh artiq-dev=4.0.dev conda environment with the latest master. Only change is to set the HMC830 reference frequency to 150MHz in the Sayma AMC target. Clocking at 150MHz from a synth.

Building without sawg, I see glitches on the ramp generator's output

[image]

Some channels look better than others, but all have some level of glitch. This could well be related to the synchronisation issues I see...

sbourdeauducq, 2018-10-03:
@marmeladapk / @jbqubit are you able to reproduce this?

hartytp, 2018-10-04:
Reproduced in a fresh conda environment with a current master, now clocking at 100MHz. Binaries are [here](https://drive.google.com/open?id=1_h-E_fdQIOmMpUdifZFKY9qnXg2y_Mep), timing is met.

```
conda list
# packages in environment at /home/tph/anaconda3/envs/artiq-dev:
#
# Name                    Version                   Build  Channel
aiohttp                   3.1.3                    py35_0    conda-forge/label/main
alabaster                 0.7.12                     py_0    conda-forge/label/main
artiq                     4.0.dev0+1408.gd0ee2c29           <pip>
artiq-dev                 4.0.dev         1408+gitd0ee2c29    m-labs/label/dev
asn1crypto                0.24.0                   py35_3    conda-forge/label/main
async-timeout             2.0.1                    py35_0    conda-forge/label/main
asyncserial               0.1             py_13+git340e430    m-labs/label/main
attrs                     18.2.0                     py_0    conda-forge/label/main
babel                     2.6.0                      py_1    conda-forge/label/main
binutils-or1k-linux       2.30                          7    m-labs/label/main
blas                      1.0                         mkl  
bscan-spi-bitstreams      0.10.0                        2    m-labs/label/main
bzip2                     1.0.6                h470a237_2    conda-forge/label/main
ca-certificates           2018.8.24            ha4d7672_0    conda-forge/label/main
certifi                   2018.8.24             py35_1001    conda-forge/label/main
cffi                      1.11.5           py35h5e8e0c9_1    conda-forge/label/main
chardet                   3.0.4                    py35_3    conda-forge/label/main
colorama                  0.3.9                      py_1    conda-forge/label/main
coverage                  4.5.1            py35h470a237_1    conda-forge/label/main
cryptography              2.3.1            py35hdffb7b8_0    conda-forge/label/main
cryptography-vectors      2.3.1                    py35_0    conda-forge/label/main
dbus                      1.13.0               h3a4f0e9_0    conda-forge/label/main
docutils                  0.14                     py35_1    conda-forge/label/main
expat                     2.2.5                hfc679d8_2    conda-forge/label/main
fontconfig                2.13.1               h65d0f4c_0    conda-forge/label/main
freetype                  2.9.1                h6debe1e_4    conda-forge/label/main
gettext                   0.19.8.1             h5e8e0c9_1    conda-forge/label/main
glib                      2.55.0               h464dc38_2    conda-forge/label/main
gst-plugins-base          1.12.5               hde13a9d_0    conda-forge/label/main
gstreamer                 1.12.5               h61a6719_0    conda-forge/label/main
h5py                      2.8.0            py35h7eb728f_2    conda-forge/label/main
hdf5                      1.10.2               hc401514_2    conda-forge/label/main
icu                       58.2                 hfc679d8_0    conda-forge/label/main
idna                      2.7                      py35_2    conda-forge/label/main
idna_ssl                  1.0.0                         0    conda-forge/label/main
imagesize                 1.1.0                      py_0    conda-forge/label/main
intel-openmp              2019.0                      118  
jesd204b                  0.10                       py_1    m-labs/label/main
jinja2                    2.10                       py_1    conda-forge/label/main
jpeg                      9c                   h470a237_1    conda-forge/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libcurl                   7.61.0               h1ad7b7a_0  
libffi                    3.2.1                hfc679d8_5    conda-forge/label/main
libgcc-ng                 7.2.0                hdf63c60_3    conda-forge/label/main
libgfortran               3.0.0                         1    conda-forge/label/main
libgfortran-ng            7.2.0                hdf63c60_3    conda-forge/label/main
libgit2                   0.24.1                        7    m-labs/label/main
libiconv                  1.15                 h470a237_3    conda-forge/label/main
libpng                    1.6.35               ha92aebf_2    conda-forge/label/main
libssh2                   1.8.0                h5b517e9_2    conda-forge/label/main
libstdcxx-ng              7.2.0                hdf63c60_3    conda-forge/label/main
libusb                    1.0.20                        0    m-labs/label/main
libuuid                   2.32.1               h470a237_2    conda-forge/label/main
libxcb                    1.13                 h470a237_2    conda-forge/label/main
libxml2                   2.9.8                h422b904_5    conda-forge/label/main
lit                       0.4.1                      py_9    m-labs/label/main
llvm-or1k                 6.0.0                        25    m-labs/label/main
llvmlite-artiq            0.23.0.dev               py35_4    m-labs/label/main
markupsafe                1.0              py35h470a237_1    conda-forge/label/main
microscope                1.3                        py_1    m-labs/label/main
migen                     0.7             py35_73+gitbef9dea    m-labs/label/dev
misoc                     0.11            py35_31+git5ce139dd    m-labs/label/dev
mkl                       2019.0                      118  
[... 148 lines cut ...]
[... cut ...]
[    16.206694s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[    16.213940s]  WARN(runtime): using default MAC address 02-00-00-00-00-11; consider changing it
[    16.221314s]  INFO(runtime): using default IP address 192.168.1.60
[    16.228970s]  INFO(runtime::mgmt): management interface active
[    16.242120s]  INFO(runtime::session): accepting network sessions
[    16.256063s]  INFO(runtime::session): running startup kernel
[    16.282968s]  INFO(runtime::kern_hwreq): resetting RTIO
```

hartytp, 2018-10-04:
Looking at all channels on a fast scope. Using an ac-coupled TCM2-43X+ and a 50Ohm scope. Channel numbers refer to the physical ordering of the SMPs, with the ones nearest the SATA connectors being 7 (I recall the mapping between SMPs and DAC channels being somewhat screwy).

0
[image]

1
[image]

2
[image]

3
[image]

4
[image]

5
[image]

6
[image]

7
[image]

hartytp, 2018-10-04:
So we have these odd glitches and channels 3/7 show some odd curvature...

@jbqubit @marmeladapk @gkasprow can one of you look at your board with the binaries I posted and see if you reproduce this...

jordens, 2018-10-04:
The "odd curvature" (assuming you are talking about the low frequency deviation from the sawtooth) on 1,3,5,7 is your balun AFAICT. Do they worry you?
There are glitches on 0,1,4,5: periodic on 1,5 and random on 0,4.

jbqubit, 2018-10-05:
Using @hartytp build I also see problems with the sawtooth pattern. When errors occur they don't occur with the same period as the sawtooth so I've done a single-trigger capture. sawg0...sawg7

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sbourdeauducq, 2018-10-06:
Is there an older version where that did not happen?

hartytp, 2018-10-06:
> Is there an older version where that did not happen?

Not sure. Are you able to test this on your hardware? Or, is there some other issue blocking that?

We could go back and check with the commit that we though fixed this. If you post binaries then maybe @jbqubit can test them (I won't be able to do that this week)? Since this issue only affects some channels, it's quite possible that we thought it was fixed when it wasn't. Or, it could be one of those PITA issues like an incorrect reset sequence or CDC or something that only causes issues with some builds.

gkasprow, 2018-10-06:
@sbourdeauducq Is there a way to get access to your MTCA crate?
 I mean MCH Ethernet, NAT MCH console, Sayma USB + MMC programmer + some mean to switch the power of the crate ?

jordens, 2018-10-09:
Given that the behavior differs between channels (where the ramp generator does not differ), this is likely something downstream and not the ramp generator.

hartytp, 2018-10-09:
> Is it known whether or not this happens with SAWG as well?

@jbqubit 

I haven't got enough data at the moment to give a definitive answer to that, although I can retest later if necessary. 

I only looked at one channel with SAWG and didn't see any evidence of this (see the shots I posted on the synchronisation issue). That same channel displayed glitches with the ramp generator.

However, without knowing the cause of this issue, it's possible that the presence/absence of glitches depends on exactly what RF waveform is produced on the various channels, so my one observation doesn't necessarily tell us much.

> To frame this, when do you (or anyone) remember checking the outputs the last time? Before the synchronization/sysref work?

This is not related to the SYSREF work. @jbqubit sees this on an unmodified Sayma using the current ARTIQ master. I see it both with master and with my sync branch.

I did some tests looking at glitches shortly after the current JESD204B release (see the IRC logs). It's been a while now, so I can't recall if I verified that the glitches were absent on all channels, or if I just confirmed that after that release the channel I had been looking at no longer had glitches (i.e. I'm not sure if I can confirm whether the glitches disappeared after that release or if they just moved channels).

Once the power supply issue is fixed, it would be great to know if @sbourdeauducq can reproduce this.

marmeladapk, 2018-10-09:
@hartytp Did you try to supply other reference frequencies (100, 125 MHz?). Just as a sanity check. With @jbqubit we had very similar symptoms, which were caused by wrong reference frequency (even though hmc830 locked). Perhaps hmc830 freq isn't set properly?

It's a wild guess, but it should be quick to check it.

hartytp, 2018-10-09:
> @hartytp Did you try to supply other reference frequencies (100, 125 MHz?). Just as a sanity check. With @jbqubit  we had very similar symptoms, which were caused by wrong reference  frequency (even though hmc830 locked). Perhaps hmc830 freq isn't set  properly?

Good thought, but I was quite careful about this. I was also able to reproduce this issue with a variety of clocking options and reference frequencies. 

Note also, that @jbqubit reproduced my measurements, so he would have to be making the same mistake as me.

sbourdeauducq, 2018-12-22:
The picture here looks suspicious, I don't see large glitches but that's perhaps just because of the limited scope bandwidth.
[image]

I don't know what is causing this.

hartytp, 2018-12-22:
Did you look at all DAC channels?

sbourdeauducq, 2018-12-22:
That's the next thing I'm planning to do, but I ran out of time for today.
BTW, this was taken with the board inside the µTCA crate and Ethernet also works (through the crate).

hartytp, 2018-12-22:
> BTW, this was taken with the board inside the µTCA crate and Ethernet also works (through the crate).

🎆

jbqubit, 2018-12-26:
> perhaps just because of the limited scope bandwidth.

Your scope has enough vertical and time resolution to see the glitches that I saw. Looks like both @hartytp and I saw that only a subset of DAC channels had the glitches. 

https://github.com/m-labs/artiq/issues/1166#issuecomment-427388926

hartytp, 2018-12-27:
Digging through the issue trackers to remind myself the history of this issue:
- Issue posted 30 May 2018 https://github.com/m-labs/artiq/issues/1022
  - moved to https://github.com/sinara-hw/sinara/issues/551 on 1 June
  - also related https://github.com/sinara-hw/sinara/issues/557 @jbqubit still saw some small glitches near the ends of the ramp gen range, but I interpreted that as SI issues (these glitches only occur *after* the large step in the ramp gen output)
  - Sinusoids from SAWG give garbage on certain channels at certain amplitudes
  - I observed glitches on the ramp generator https://pasteboard.co/HnT4Gf6.jpg
  - I reported the ramp generator glitches fixed after https://github.com/m-labs/misoc/commit/2436a68dcb24ef6624a6c9f86bdccb2f2632c158 but there were also some migen and JESD core fixes around the same time
- that misoc patch also appeared to fix https://github.com/m-labs/artiq/issues/1040 https://github.com/m-labs/artiq/issues/1039 and https://github.com/sinara-hw/sinara/issues/536 (odd SAWG behaviour with certain settings)

sbourdeauducq, 2019-01-01:
Okay, now it won't boot anymore, ``panic at src/libcore/result.rs:945:5cannot load RTM FPGA gateware: "Did not assert INIT in reaction to PROGRAM"``
Maybe one of the reworks broke when I added more Allaki to look at other channels...

jordens, 2019-01-01:
Cleaning this up and categorizing the historic and current issues observed/fixed, we have the following:

1. The glitches we are looking at in this issue here. Earliest report (and distinct from the others) is in this issue up top.
2. https://github.com/m-labs/artiq/issues/1022#issuecomment-393905927 and https://github.com/sinara-hw/sinara/issues/551#issuecomment-395099535 (https://pasteboard.co/HnT4Gf6.jpg) is from 2018-06-01 and before the JESD changes and before the Cat() changes. Those glitches were also qualitatively different and hat a specific, deterministic pattern (when they occurred) that was addresses in the commits there. They were the dominant issue we isolated, and fixed in those issues.
3. https://github.com/m-labs/artiq/issues/1022#issuecomment-395895632 may or may not be a false positive since the preceding fix required a misoc update which was frequently overlooked. 
4. The glitches after the ramp jump (https://github.com/sinara-hw/sinara/issues/557#issuecomment-394902405 and https://github.com/sinara-hw/sinara/issues/551#issue-328542083) may or may not be the same as (1). Should be verified by cable length changes.

AFAIR since there were also significant changes to the test pattern generator (without-sawg) afterwards that could have and were intended to converted rare and low visibility glitches (like 4) into more dramatic glitches like (1).

Ergo, let's focus on (1) first and ignore the rest for now.

Since we do not suspect the ramp generator, we should look at JESD.

hartytp, 2019-01-01:
Thanks for looking into this @sbourdeauducq and @jordens. Glad to hear you can reproduce the issue. If there are any specific tests you'd like me to do then let me know.

sbourdeauducq, 2019-01-02:
The problem is **not** present with the DRTIO satellite target (@hartytp please confirm). So this looks like a clocking gateware/firmware bug, or more HMC7043 shenanigans (with DRTIO satellite, the JESD transceivers are clocked by the Si5324 and not the HMC).

To test, compile the DRTIO satellite with the ``--without-sawg`` flag and then provide a DRTIO uplink on the SFP (with 150MHz RTIO clock) which can come from a Kasli or another Sayma acting as DRTIO master. Just establishing the DRTIO link is sufficient, no command has to be sent.

jordens, 2019-01-02:
Ha. The glitches are every 40 ns. That's an even 6 coarse RTIO periods and 5 periods of 125 MHz. That's why they look random on the ram and periodic on the sine. There isn't that much (anything?) driven at 125 MHz in DRTIO satellites, but quite a bit in master bitstreams. This could be some fabric or external crosstalk due to the beat and thus PI/SI issues. Or a gateware/firmware bug that's always there and only exposed by the beat.

sbourdeauducq, 2019-01-02:
The satellite has the CPU with satman + SDRAM at 125MHz.

sbourdeauducq, 2019-01-02:
We can test this theory by changing the system frequency in the master bitstream. I'll try that tomorrow.

jordens, 2019-01-02:
Ack. The difference could still be due to the reduced amount of 125 MHz logic (kernel cpu and associated logic). But this is definitely at the beat. If this is PI, one might naively expect a significant "intermodulation" product on the power supplies at 25 MHz. If it is SI, then we should look at how the 125 MHz clock is getting into the 150 MHz/JESD domain. Could also be some 125 MHz hitting the DAC since its PClock is also at 150 MHz AFAIK (line rate/40).
OTOH from the way the sine data is corrupted and from the almost clean 600/4/16/3 MHz periodicity of the ramp corruption it looks like it is predominantly taking "clean" data from some other/wrong group of four samples. I don't immediately see what that means but maybe EB/CDC or JESD framing issue?

sbourdeauducq, 2019-01-03:
Changed the system clock frequency using a patch similar to https://hastebin.com/nepilataje.rb, but with 21 instead of 19 (19 tickles a SDRAM bug), which yields 131.25MHz.
Observed glitches (all on the same channel):
[image]
[image]
[image]

enjoy-digital, 2019-01-07:
I just did some tests with KCU105 / AD9154 + simple test design @10Gbps linerate + pattern from Artiq, here is what i have:

Channel0:
[image]

Channel1:
[image]

Channel2:
[image]

Channel3:
[image]

Channel 1 and 3 seem fine (83*600/1000 = 50MHz), but i was expeting to have the same pattern than with Artiq on Channel 0 and 2 but it does not seem to be the case. Otherwise, i don't seem to see the glitches.

Please tell me if you want me to do more tests.

sbourdeauducq, 2019-01-08:
@enjoy-digital thanks!

sbourdeauducq, 2019-02-02:
Seems resolved with Vivado 2018.3. @hartytp please confirm.

closed 2020-04-14 by sbourdeauducq