m-labs/artiq #1171: API inconsistency: set_dataset(..., save=True) vs get_dataset(..., archive=True)

jordens, 2018-10-09:
It would be clearer leave fewer chances for errors to call them both `archive` since the functionality is the same. Or am I missing something?

sbourdeauducq, 2018-10-10:
Agreed. If no one objects, I'll add ``archive`` to ``set_dataset`` while keeping ``save`` (for backward compatibility and with a warning; we can e.g. do that in 4.0 and remove it completely in 5.0).

closed 2018-10-21 by sbourdeauducq

jordens, 2018-10-22:
The slightly confusing wrinkle (and probably the reason it was not like this before) is the following behavior:
Something that has been set with `set(..., archive=True)` does **not** appear in the `archive` group (but in the `datasets` group) in the HDF5 but something from `get(..., archive=True)` does appear in the `archive` group.
Since we treat the datasets (and the "archive" stuff) as a single namespace anyway we might also merge the two groups.

commit 97936322 (2018-10-21): enviromnment: rename 'save' in set_dataset to 'archive'. Closes #1171