m-labs/artiq #1172: Kasli ignores rtio_clock e

connorgoham, 2018-10-09:
No SMA input supplied to Kasli. Why does Si5324 report ```locked```?

```
$ artiq_coremgmt config write -s rtio_clock e
(artiq-dev) rabi@master0:~/artiq-lab/hucul-west$ artiq_coremgmt config read rtio_clock
e
(artiq-dev) rabi@master0:~/artiq-lab/hucul-west$ artiq_flash -t kasli start
Open On-Chip Debugger 0.10.0-00013-gbb7beda (2018-02-13-15:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : gdb server disabled
TEMP 90.70 C
VCCINT 0.991 V
VCCAUX 1.785 V
VCCBRAM 0.997 V
VPVN 0.000 V
VREFP 0.000 V
VREFN 0.000 V
VCCPINT 0.000 V
VCCPAUX 0.000 V
VCCODDR 0.000 V
(artiq-dev) rabi@master0:~/artiq-lab/hucul-west$ artiq_coremgmt log
[     0.000010s]  INFO(runtime): ARTIQ runtime starting...
[     0.003945s]  INFO(runtime): software ident 4.0.dev+1298.ge4d48a78;opticlock
[     0.011007s]  INFO(runtime): gateware ident 4.0.dev+1298.ge4d48a78;opticlock
[     0.018114s]  INFO(runtime): log level set to INFO by default
[     0.023805s]  INFO(runtime): UART log level set to INFO by default
[     0.029956s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.030653s]  INFO(runtime): continuing boot
[     1.311238s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     5.321584s]  INFO(board_artiq::si5324):   ...locked
[     5.325328s]  INFO(runtime): using MAC address 00-0a-35-03-1e-75
[     5.331298s]  INFO(runtime): using IP address 192.168.1.76
[     5.338734s]  INFO(runtime::mgmt): management interface active
[     5.352674s]  INFO(runtime::session): accepting network sessions
[     5.367759s]  INFO(runtime::session): running startup kernel
[     5.372335s]  INFO(runtime::session): no startup kernel found
[     5.378053s]  INFO(runtime::session): no connection, starting idle kernel
[     5.412542s]  INFO(runtime::kern_hwreq): resetting RTIO
[    10.420209s]  INFO(runtime::mgmt): new connection from 192.168.1.80:49884
```

jordens, 2018-10-09:
It uses its internal oscillator. Kasli clocking is set in the target definition. The `rtio_clock` config entry is only used on kc705 (see the manual).

jbqubit, 2018-10-09:
OK. Current doc is confusing. Please move ```rtio_clock``` config instructions from [here](http://m-labs.hk/artiq/manual-master/installing.html#configuring-the-core-device) to [here](http://m-labs.hk/artiq/manual-master/core_device.html?highlight=kc705#clocking). 

Kasli Opticlock variant [doc](http://m-labs.hk/artiq/manual-master/core_device.htm) says "Kasli is supplied with 100 MHz reference at its SMA input" but [target definition](https://github.com/m-labs/artiq/blob/469a66db61da4c40d72b8627053821df169a8dbe/artiq/gateware/targets/kasli.py#L156) is ```self.config["SI5324_EXT_REF"] = None```. This looks inconsistent.

hartytp, 2018-10-09:
> `self.config["SI5324_EXT_REF"] = None`

I don't think that does what you think it does...

jbqubit, 2018-10-09:
I gather this is creating a key with empty value. Presence or absence of key as a way of toggling a bool is opaque. The only reference I can find to ```SI5324_EXT_REF``` in source relates to Sayma.

hartytp, 2018-10-09:
To explain more (and @jordens should correct me if I'm wrong since it's been a while since I thought about this):
- adding lines like `self.config["SI5324_EXT_REF"] = None` add config entries which cause rust `#[cfg()]' blocks to evaluate true. So, the code taken by OptiClock is this https://github.com/m-labs/artiq/blob/b482f5feaedde7bd10903fd9e7cad249e596f88b/artiq/firmware/runtime/main.rs#L138-L150
- as robert says, Kasli does not use do any runtime configuration of the clocking, e.g. via `artiq_coremgmt config read rtio_clock` although that could be added
- so, the OptiClock code path does use the Si5324 with the external clock. However, it still works with no clock present since the Si5324 has an on-board Xtal which just free-runs when no input clock is present. i.e. if you give it a clock then it will lock to it, otherwise it will free-run. 

Is that about right @jordens or am I missing something?

hartytp, 2018-10-09:
> I gather this is creating a key with empty value. Presence or absence of key as a way of toggling a bool is opaque. The only reference I can find to SI5324_EXT_REF in source relates to Sayma.

@jbqubit I think we either need to fund a major documentation effort from migen->misoc->artiq or live with having to do a bit of digging through sources to understand these things.

When you build Kasli, look in the rust sources that are generated. You'll find that each entry in self.config leads to a config option in the rust sources. 

> The only reference I can find to SI5324_EXT_REF in source relates to Sayma.

https://github.com/m-labs/artiq/search?q=SI5324_EXT_REF&unscoped_q=SI5324_EXT_REF

It's also in the runtime

jbqubit, 2018-10-09:
>> each entry in self.config leads to a config option in the rust sources

Thanks for the explanation @hartytp. Yes. I was thrown off by the odd logic ```not(si5324_sayma_ref)```. 

> if you give it a clock then it will lock to it, otherwise it will free-run.

In the case of no external clock, boot diagnostics imply locking when it's really free-running. 

```
[     1.311238s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     5.321584s]  INFO(board_artiq::si5324):   ...locked
```

jordens, 2018-10-09:
The si5324 fallback mode also won't work because that needs the other input. But a one shot runtime fallback could be done.

hartytp, 2018-10-09:
> @hartytp almost. IIRC there is no fallback. It either does one or the other. 

Ack. So, with OptiClock, what happens when one doesn't supply a reference via the SMA? Why does it report lock? Just an un-driven input buffer free oscillating?

jordens, 2018-10-10:
It doesn't report lock.

hartytp, 2018-10-10:
@jordens ok well then I don't understand the uart trace posted above

jordens, 2018-10-10:
That's old code.

hartytp, 2018-10-10:
Aah .:) right that changed recently. Should have spotted that.

hartytp, 2018-10-10:
@jbqubit specifically, see https://github.com/m-labs/artiq/commit/d0ee2c2955e23c835e122b04b915e03c3cecfa46

sbourdeauducq, 2020-10-15:
a94786788749b1541c926b137f8e0fe897fd3c74 implements the legacy kc705 behavior.
More advanced configuration of the Si5324 without firmware recompilation would be a separate issue.

closed 2020-10-15 by sbourdeauducq