m-labs/artiq #1174: with parallel urukul.set() broken

jbqubit, 2018-10-09:
Expectation is that frequencies are left at ```2*self.f``` by end of experiment but observe ```self.f```.

```
import sys
import select

from artiq.experiment import *

class UrukulSet(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch0")
        self.setattr_device("urukul0_ch1")

        self.f = 23.4*MHz 

    @kernel
    def run(self):
        self.core.reset()

        self.core.break_realtime()
        self.urukul0_cpld.init()

        self.urukul0_ch0.init()
        self.urukul0_ch1.init()

        self.urukul0_ch0.sw.on()
        self.urukul0_ch0.set_att(0.)
        self.urukul0_ch1.sw.on()
        self.urukul0_ch1.set_att(0.)

        self.urukul0_ch0.set(self.f, 0.)
        self.urukul0_ch1.set(self.f, 0.)

        with parallel: 
            self.urukul0_ch0.set(2*self.f, 0.)
            self.urukul0_ch1.set(2*self.f, 0.)
```

sbourdeauducq, 2018-10-10:
There is only one SPI bus, so setting several DDS on the same card at the same time is not possible. Isn't there a collision error in the log?

jbqubit, 2018-10-10:
When operating AD9910 in Single Tone Mode there are 8 independent Profiles. Set PROFILE[2:0] to select the desired profile. A change in the state of the profile pins with the next rising edge on SYNC_CLK updates the DDS with the parameters specified by the selected profile.

AFAICT it is straightforward to update several AD9910 DDSs in parallel. Use a pair of PROFILEs on each DDS and switch between them round robin. 
0) load FTW_i into profile_0 on both DDSs
1) at time t_i, set PROFILE[2:0] to select profile_0 in advance of of SYNC_CLK 
2) load FTW_i+1 into profile_1 on both DDSs
3) at time t_i+1, set PROFILE[2:0] to select profile_1 in advance of of SYNC_CLK 
4) continue at step 0

Non-parallel DDS updates follow naturally. CPLD keeps track of which profile is currently selected by PROFILE[2:0].

jordens, 2018-10-10:
It would be great if you try your suggestion, report back, and then prepare a high quality pull request.

If this was user error and there is nothing more, we can close it.

closed 2018-10-11 by jordens