m-labs/artiq #1175: urukul: inter-channel phase between runs

jbqubit, 2018-10-10:
Expectation: Phase relationship between DDS's on single Urukul board are constant from one run of an experiment file to the next. Observation: For program below, random phase between ch0 and ch1 from one run of the experiment to the next. 

```
import sys
import select

from artiq.experiment import *

class UrukulSet(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("scheduler")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch0")
        self.setattr_device("urukul0_ch1")

        self.f = 23.4*MHz 

    @kernel
    def run(self):
        self.core.reset()

        self.core.break_realtime()
        self.urukul0_cpld.init()

        self.urukul0_ch0.init()
        self.urukul0_ch1.init()

        self.urukul0_ch0.sw.on()
        self.urukul0_ch0.set_att(0.)
        self.urukul0_ch1.sw.on()
        self.urukul0_ch1.set_att(0.)

        self.urukul0_ch0.set(self.f, 0.)
        self.urukul0_ch1.set(self.f, 0.)
```

Note: The phase relationship between ch0 and ch1 remains well defined within a single experiment run even with multiple calls to set(). For example, if I were to make a loop stepping the phase of ch1 within a kernel.

jordens, 2018-10-10:
Don't reset the DDS each time.

closed 2018-10-10 by jordens

jbqubit, 2018-10-10:
> Don't reset the DDS each time.

Thanks. Skipping ```.init()``` fixes the problem I saw. Looks like the following needs to be called once after powering on the crate. 
```
        self.urukul0_cpld.init()
        self.urukul0_ch0.init()
        self.urukul0_ch1.init()
```

Consider noting this in the docs so users don't run into the same problem.