m-labs/artiq #1184: Question: Difference between Controllers and ControllerDB

drewrisinger, 2018-10-23:
In [ctlmgr.py](./artiq/devices/ctlmgr.py), there are two nearly-identical classes ([Controllers](https://github.com/m-labs/artiq/blob/0b43ec4719dd67aa0d839a9e9c69e26783481bdf/artiq/devices/ctlmgr.py#L157) and [ControllerDB](https://github.com/m-labs/artiq/blob/0b43ec4719dd67aa0d839a9e9c69e26783481bdf/artiq/devices/ctlmgr.py#L204)). What is the point of having both, and the usage difference between them? Is there any reason that `Controllers` and `ControllerDB` can't be merged?

I'm trying to write docstrings for this module right now (https://github.com/drewrisinger/artiq/commit/4a387bed63bb2c282aa8449f633ed1f699a1d2bc), and got confused about these.

sbourdeauducq, 2018-10-23:
``ControllerDB`` is just a small adaptation layer on top of ``Controllers``, that provides ``sync_struct_init()`` to use ``Subscriber``.

sbourdeauducq, 2018-10-23:
``ControllerDB`` doesn't really need to be a class though, ``set_host_filter`` can be inlined and ``sync_struct_init`` implemented as a bare function.

closed 2018-10-23 by sbourdeauducq