m-labs/artiq #1195: Connection fails: unknown packet 0x3

hermitdemschoenenleben, 2018-11-15:
# Bug Report

## One-Line Summary
Connection to artiq using the `artiq_session` fails with `lost connection to core device moninj` and `moninj aborted: unknown packet 0x3`.

## Issue Details
### Packet trace
https://berrsche.selfhost.eu/d/e051ee3cc8/files/?p=/artiq.pcapng&dl=1 

### Steps to Reproduce
* ` git clone https://github.com/quartiq/artiq-setup.git --branch hub-krutzik --single-branch`
* `conda create -n artiq-1483 artiq-kasli-hub=4.0.dev=1483+git0b2661a3`
* `conda activate artiq-1483`
* Now I'm linking the folder with the bitstreams to /usr/share in order to make flashing work: `sudo ln -s ~/.conda/envs/artiq-1483/share/bscan-spi-bitstreams /usr/share`
* `artiq_mkfs -s ip 192.168.0.200 -s mac 54:10:ec:a9:d2:7b storage.img`
* `sudo env PATH=$PATH ~/.conda/envs/artiq-1483/bin/artiq_flash -t kasli -V hub -f storage.img storage start`
* `artiq_session`

### Expected Behavior
Successful connection.

### Actual (undesired) Behavior
* Connection fails:
* Log message:
```
(artiq-1483) ➜  artiq-setup git:(hub-krutzik) ✗ artiq_session
ARTIQ master is now ready.
INFO:worker(scan,demo.py):print:/home/ben/.conda/envs/artiq-1483/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
INFO:worker(scan,demo.py):print:  return f(*args, **kwds)
/home/ben/.conda/envs/artiq-1483/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  return f(*args, **kwds)
Gtk-Message: 12:47:23.311: Failed to load module "pk-gtk-module"
Gtk-Message: 12:47:23.311: Failed to load module "canberra-gtk-module"
INFO:controller(core_log):firmware.runtime:ARTIQ runtime starting...
INFO:controller(core_log):firmware.runtime:software version 4.0.dev+1236.g701c93d4
INFO:controller(core_log):firmware.runtime:ARTIQ runtime starting...
INFO:controller(core_log):firmware.runtime:gateware version 4.0.dev+1236.g701c93d4
INFO:controller(core_log):firmware.runtime:software version 4.0.dev+1236.g701c93d4
INFO:controller(core_log):firmware.runtime:log level set to INFO by default
INFO:controller(core_log):firmware.runtime:gateware version 4.0.dev+1236.g701c93d4
INFO:controller(core_log):firmware.runtime:UART log level set to INFO by default
INFO:controller(core_log):firmware.runtime:log level set to INFO by default
INFO:controller(core_log):firmware.runtime:press 'e' to erase startup and idle kernels...
INFO:controller(core_log):firmware.runtime:continuing boot
INFO:controller(core_log):firmware.board_artiq.si5324:waiting for Si5324 lock...
INFO:controller(core_log):firmware.board_artiq.si5324:  ...locked
INFO:controller(core_log):firmware.runtime:using MAC address 54-10-ec-a9-d2-7b
INFO:controller(core_log):firmware.runtime:using IP address 192.168.0.200
INFO:dashboard:root:ARTIQ dashboard 4.0.dev+1483.g0b2661a3 connected to ::1
INFO:controller(core_log):firmware.runtime.mgmt:management interface active
INFO:controller(core_log):firmware.runtime.session:accepting network sessions
INFO:controller(core_log):firmware.runtime.session:running startup kernel
INFO:controller(core_log):firmware.runtime.session:no startup kernel found
INFO:controller(core_log):firmware.runtime.session:no connection, starting idle kernel
INFO:controller(core_log):firmware.runtime.session:no idle kernel found
INFO:controller(core_log):firmware.runtime.mgmt:new connection from 192.168.0.9:47616
INFO:controller(core_log):firmware.runtime:UART log level set to INFO by default
INFO:controller(core_log):firmware.runtime:press 'e' to erase startup and idle kernels...
INFO:controller(core_log):firmware.runtime:continuing boot
INFO:controller(core_log):firmware.board_artiq.si5324:waiting for Si5324 lock...
INFO:controller(core_log):firmware.board_artiq.si5324:  ...locked
INFO:controller(core_log):firmware.runtime:using MAC address 54-10-ec-a9-d2-7b
INFO:controller(core_log):firmware.runtime:using IP address 192.168.0.200
INFO:controller(core_log):firmware.runtime.mgmt:management interface active
INFO:controller(core_log):firmware.runtime.session:accepting network sessions
INFO:controller(core_log):firmware.runtime.session:running startup kernel
INFO:controller(core_log):firmware.runtime.session:no startup kernel found
INFO:controller(core_log):firmware.runtime.session:no connection, starting idle kernel
INFO:controller(core_log):firmware.runtime.session:no idle kernel found
INFO:controller(core_log):firmware.runtime.mgmt:new connection from 192.168.0.9:47616
[... 98 lines cut ...]
regex                     2015.11.22               py35_1    m-labs/label/main
scipy                     1.1.0            py35hfc37229_0  
setuptools                40.4.3                   py35_0    conda-forge/label/main
sip                       4.18.1           py35hfc679d8_0    conda-forge/label/main
six                       1.11.0                   py35_1    conda-forge/label/main
sqlite                    3.25.3               hb1c47c0_0    conda-forge/label/main
system                    5.8                           2  
tk                        8.6.8                ha92aebf_0    conda-forge/label/main
wheel                     0.32.0                py35_1000    conda-forge/label/main
xorg-kbproto              1.0.7                h470a237_2    conda-forge/label/main
xorg-libice               1.0.9                h470a237_4    conda-forge/label/main
xorg-libsm                1.2.3                h8c8a85c_0    conda-forge/label/main
xorg-libx11               1.6.6                h470a237_0    conda-forge/label/main
xorg-libxau               1.0.8                h470a237_6    conda-forge/label/main
xorg-libxdmcp             1.1.2                h470a237_7    conda-forge/label/main
xorg-libxext              1.3.3                h470a237_4    conda-forge/label/main
xorg-libxrender           0.9.10               h470a237_2    conda-forge/label/main
xorg-libxt                1.1.5                h470a237_2    conda-forge/label/main
xorg-renderproto          0.11.1               h470a237_2    conda-forge/label/main
xorg-xextproto            7.3.0                h470a237_2    conda-forge/label/main
xorg-xproto               7.0.31               h470a237_7    conda-forge/label/main
xz                        5.2.4                h470a237_1    conda-forge/label/main
yarl                      1.2.6            py35h470a237_0    conda-forge/label/main
zlib                      1.2.11               h470a237_3    conda-forge/label/main
```

sbourdeauducq, 2018-11-17:
You are using old firmware (4.0.dev+1236.g701c93d4) with a newer dashboard (4.0.dev=1483+git0b2661a3).
Please reflash the board completely (not just the storage area) and this problem should go away.
You shouldn't have to mess with /usr/share "in order to make flashing work". What was the problem exactly?

hermitdemschoenenleben, 2018-11-21:
thank you, now it works!

>  You shouldn't have to mess with /usr/share "in order to make flashing work". What was the problem exactly?
I had a problem with artiq_flash not finding the bitstreams in an older revision, but I can't reproduce it anymore; now it works fine without the link, thank you!

closed 2018-11-21 by hermitdemschoenenleben