m-labs/artiq #1196: Different Server/Client Limits

drewrisinger, 2018-11-19:
# Bug Report

## One-Line Summary

Async co-routines have different reader limits. Related: #285 

## Issue Details

### Steps to Reproduce

AsyncioClient: 
https://github.com/m-labs/artiq/blob/8f9858be4c93f86774316ef406c40f4be6c28467/artiq/protocols/pc_rpc.py#L203-L204

AsyncioServer:
https://github.com/m-labs/artiq/blob/8f9858be4c93f86774316ef406c40f4be6c28467/artiq/protocols/asyncio_server.py#L27-L29

### Expected Behavior

I would **expect these limits to be the same**. Is there any reason why they aren't? If I'm transferring this much data, _is PYON not the right tool and I should use something else_ (like a client as mentioned in [developing an ndsp.rst](https://github.com/m-labs/artiq/blob/master/doc/manual/developing_a_ndsp.rst#developing-a-network-device-support-package-ndsp))

### Actual (undesired) Behavior

This caused me some errors when sending large numpy arrays (~2 MB of raw data) via PYON RPC. Actual code is too complex, but changing the AsyncioServer limit to ~16 MB fixed it for me.

sbourdeauducq, 2018-12-01:
@drewrisinger Did you figure out better numbers and can you send a PR?

drewrisinger, 2019-01-04:
I used 16 MB for asyncio_server and that worked for my data. Think this is highly data-dependent, and I don't have a good handle on when it's worth increasing the limit vs moving to a different non-PYON protocol. The raw data that the numpy array was constructed from was ~2-4 MB of text, so it makes sense that 4 MB was too tight.

closed 2019-03-15 by sbourdeauducq