m-labs/artiq #1198: malformed device_db entries break artiq_ctlmgr

hartytp, 2018-11-28:
device_db containing an entry that is a dictionary without a "type" field leads to an unhanded exception here https://github.com/m-labs/artiq/blob/57caa7b149685585dbc777528d7c86514bcb4874/artiq/devices/ctlmgr.py#L181

This breaks the synchronisation between the controller manager's local copy of the device db and the master. After this, no further entries in the master's device db are processed.

Because the synchronisation is handled in a coroutine, this doesn't result in an error message, the device manager just silently stops parsing the device db. Since dictionary iteration is non-deterministic, the result is the device manager only creating devices for a random subset of the ones in the device_db, but without any information in the log as to why.

Any preferences as to how we fix this? Obvious options would be to put https://github.com/m-labs/artiq/blob/57caa7b149685585dbc777528d7c86514bcb4874/artiq/devices/ctlmgr.py#L215 in a try-catch block or by replacing `v["type"] == "controller"` with `v.get("type", "none") == "controller"`.

sbourdeauducq, 2018-12-01:
Putting that one in the try-except block is not sufficient, as ``sync_struct`` will also call ``__setitem__`` directly.

hartytp, 2018-12-01:
hmm...yes...so is the preferred solution to add some exception handling to `sync_struct`, which places informative info in the log when exceptions occur?

sbourdeauducq, 2018-12-01:
The methods that sync_struct calls should not raise exceptions; when they do it is because of internal program errors (which should be debugged the same way as other asyncio problems).
Handling improper user input in a graceful way should be the task of ctlmgr.

commit 6fdfd610 (2018-12-01): ctlmgr: do not raise exceptions in Controllers.__setitem__. Closes #1198
commit fd00021a (2018-12-01): ctlmgr: do not raise exceptions in Controllers.__setitem__. Closes #1198
commit c894fe02 (2018-12-01): ctlmgr: do not raise exceptions in Controllers.__setitem__. Closes #1198
closed 2018-12-01 by sbourdeauducq